istoreos/istoreos #1861: 经常无故死机

slqx, 2024-10-22:
## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：经常性死机，长则几周，断则半天

### (2) 路由器型号和固件版本

A：ikuai虚拟机安装，固件版本：iStoreOS 22.03.7 2024080210

### (3) 详细日志和/或截图

A：如日志文件（一次完整的启动到死机的死机）：

【14:06】前一次死机，重启服务
【...】       网络正常
【21:03】 发现断网，ping不通，管理后台页面无法打开，ikuai虚拟机vnc可以进入但键盘无反应，日志无滚动

重启虚拟机，保留日志

[system.log](https://github.com/user-attachments/files/17476779/system.log)

WuJieOnce, 2024-11-28:
x86主机也出现相同的问题，啥时候能有解方法哦。。

ron159, 2025-01-14:
我这边也是，准备换了

lilmik, 2025-02-14:
最近几个月我也碰到了这个问题，稳定性不知道出了什么问题。

ron159, 2025-02-14:
> 我这边也是，准备换了

节前抽空换成了immortalwrt，已经稳定运行半个月了

lizzzhh, 2025-08-15:
相同的问题，很奇怪