istoreos/istoreos #1863: 本人小白经常路由有陌生ip有时候会莫名出来7.8个，这些ip都是一个mac地址，这个mac是我的手机，ip段也不是我内网ip段，经常有10开头的100开头的

daoyelihuowang, 2024-10-22:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述
经常会出现陌生ip跟我设置的内网网段不是一个网段
### (1) 具体问题

A：
cpu占用正常、上传正常、就是经常莫名出现陌生ip都是一个mac地址多的时候这个mac地址能有7，8个连接ip，重启就消失了不定时出现，跟我设置的内网ip段不一样我设置的是192.168.31.1但那些ip有的10开头的有的100开头，困扰我很久请教大佬们。另外这个连接数正常吗重启完立刻就这么多了
### (2) 路由器型号和固件版本

A：x86 最新固件

### (3) 详细日志和/或截图

A：
[image]

[image]

daoyelihuowang, 2024-10-22:
出现的陌生ip都是(br-lan)这个接口

daoyelihuowang, 2024-10-22:
#1740 这个情况跟我一样一个mac地址但有个很奇怪的ip

daoyelihuowang, 2024-10-22:
> #1740 这个情况跟我一样一个mac地址但有个很奇怪的ip

都是(br-lan)这个接口出现的

jjm2473, 2024-10-24:
是不是有VPN流量泄漏出来了

daoyelihuowang, 2024-10-24:
> 是不是有VPN流量泄漏出来了

应该不是vpn问题之前没开也经常这样子

jjm2473, 2024-10-24:
那你看看自己的手机是不是有这个ip地址

daoyelihuowang, 2024-10-24:
> 那你看看自己的手机是不是有这个ip地址

只有192.168.31.xx那是我内网网段ip，这些莫名出来的ip，10开头，100开头的都有

jjm2473, 2024-10-24:
> 另外这个连接数正常吗重启完立刻就这么多了

哪张图有显示连接数？你发的监控图是一段时间的统计

daoyelihuowang, 2024-10-24:
> > 另外这个连接数正常吗重启完立刻就这么多了
> 
> 哪张图有显示连接数？你发的监控图是一段时间的统计

那就是我搞错了，不太懂

jjm2473, 2024-10-24:
> 只有192.168.31.xx那是我内网网段ip，这些莫名出来的ip，10开头，100开头的都有

我是说看看自己手机是不是真有这些ip地址，如果有，那路由器上能看到就正常啊，可能手机的ip或路由本身就有问题，导致这个ip作为源地址发到了路由器上，这个ip肯定没法被路由器路由的。

jjm2473, 2024-10-24:
手机有没有使用向日葵啊，tailscale，zerotier之类的组网软件

daoyelihuowang, 2024-10-24:
> > 只有192.168.31.xx那是我内网网段ip，这些莫名出来的ip，10开头，100开头的都有
> 
> 我是说看看自己手机是不是真有这些ip地址，如果有，那路由器上能看到就正常啊，可能手机的ip或路由本身就有问题，导致这个ip作为源地址发到了路由器上，这个ip肯定没法被路由器路由的。

下次这些ip出现我去看看，不固定有时候出来好几个，不定时出现不知道下次什么时候有

daoyelihuowang, 2024-10-24:
> 手机有没有使用向日葵啊，tailscale，zerotier之类的组网软件

没有用这些软件

daoyelihuowang, 2024-10-24:
> > 只有192.168.31.xx那是我内网网段ip，这些莫名出来的ip，10开头，100开头的都有
> 
> 我是说看看自己手机是不是真有这些ip地址，如果有，那路由器上能看到就正常啊，可能手机的ip或路由本身就有问题，导致这个ip作为源地址发到了路由器上，这个ip肯定没法被路由器路由的。

我是这样看的出现那些ip的时候但这里也是只有一个正常ip
[image]

jjm2473, 2024-10-24:
手机安装termux，里面执行`ifconfig`，看看有哪些ip

daoyelihuowang, 2024-10-24:
> 手机安装termux，里面执行`ifconfig`，看看有哪些ip

输出这个：ifconfig: No file /proc/net/dev: Permission denied。好像没有足够的权限，手机没有root

jjm2473, 2024-10-24:
不需要root啊，就输出这一行吗？后面没输出网卡信息？

daoyelihuowang, 2024-10-24:
> 不需要root啊，就输出这一行吗？后面没输出网卡信息？

只输出这个软件从谷歌商店下载的
[image]

jjm2473, 2024-10-24:
手机不是只有wifi网络的，还有移动数据网络（4G，5G等），10网段有可能就是4G的网络，但是被错误发到wifi上去了。

jjm2473, 2024-10-24:
手机关掉wifi，然后打开系统设置，到关于手机-状态，到后面看看IP地址，应该就是4G网络的IP了

daoyelihuowang, 2024-10-24:
> 手机关掉wifi，然后打开系统设置，到关于手机-状态，到后面看看IP地址，应该就是4G网络的IP了

看到了用流量ip是10开头的重新开关流量以后发现也有100开头的可能就是这个问题了，谢谢大佬解答，我后续在家尽量使用wifi要是没有再出现那些ip就可以百分百确定是流量ip错误发送到wifi了

jjm2473, 2024-10-24:
@daoyelihuowang 试试关掉手机的数据网络，只连wifi，过一会儿看看路由器后台还有没有奇怪的ip

daoyelihuowang, 2024-10-24:
> @daoyelihuowang 试试关掉手机的数据网络，只连wifi，过一会儿看看路由器后台还有没有奇怪的ip

好的，主要是这些ip出现没有一点规律我以后多多观察吧，我发的图片里那ip是重启完系统就出现了，但出现这个ip以后我又重启了一次昨天到现在还没有出现那些ip，我后面尽量只用wifi看看还会不会出现那些ip，那些ip有时直接出来6，7个有时候出现只出现1个，有时候一个没有 很不固定只能后面多观察了，很可能就是流量的问题

closed 2024-10-24 by daoyelihuowang

jjm2473, 2024-10-24:
出现多个可能只是还没清除缓存数据而已，内核不是每时每刻都在扫描内网ip的，而是根据流量记录有哪些ip，某个ip一段时间不活跃才会删除掉。

daoyelihuowang, 2024-10-24:
> 出现多个可能只是还没清除缓存数据而已，内核不是每时每刻都在扫描内网ip的，而是根据流量记录有哪些ip，某个ip一段时间不活跃才会删除掉。

谢谢大佬受教了

daoyelihuowang, 2024-10-24:
> 出现多个可能只是还没清除缓存数据而已，内核不是每时每刻都在扫描内网ip的，而是根据流量记录有哪些ip，某个ip一段时间不活跃才会删除掉。

晚上又出现了3个异常ip，看mac地址还是手机（每次出现的异常ip，看mac地址都是这个手机）这些异常ip会一直存在在路由里不会随时间删除，我用电脑连手机用adb指令查看了一下输出的wlan0属性，内网ip地址只有192.168.25（本机正常ip），没有这些异常ip（10.52.199.115，100.81.140.116，100.117.129.113），还是没太搞懂那些异常ip出现的原因跟规律，但可以确定这些异常ip出现以后就不会消失了只能重启lan口或者重启系统才会消失，可能跟大佬说的数据网络的问题有关但还很不确定

daoyelihuowang, 2024-10-24:
[image]

jjm2473, 2024-10-28:
那些10或100的地址不是wlan的，而是rmnet之类的移动数据网络，手机打开数据网络就能看到这些地址了。要不你关掉手机数据和wifi一段时间，再连wifi看看。要么就换手机吧。