istoreos/istoreos #1866: 这个是什么原因呢 下载安装 Istore 增强插件报这个错

kiritomemory, 2024-10-24:
Try mirror server https://repo.istoreos.com/repo
Installing app-meta-istoreenhance (0.1.7-3) to root...
Downloading https://repo.istoreos.com/repo/all/meta/app-meta-istoreenhance_0.1.7-3_all.ipk
Installing luci-i18n-istoreenhance-zh-cn (git-24.194.36967-61083fc) to root...
Downloading https://repo.istoreos.com/repo/all/nas_luci/luci-i18n-istoreenhance-zh-cn_git-24.194.36967-61083fc_all.ipk
Collected errors:
 * opkg_conf_parse_file: Duplicate src declaration (customfeed https://op.dllkids.xyz/packages/aarch64_cortex-a53). Skipping.
 * check_data_file_clashes: Package luci-i18n-istoreenhance-zh-cn wants to install file /usr/lib/lua/luci/i18n/istoreenhance.zh-cn.lmo
        But that file is already provided by package  * luci-app-istoreenhance
 * opkg_install_cmd: Cannot install package app-meta-istoreenhance.
Try origin server https://istore.linkease.com/repo
Installing app-meta-istoreenhance (0.1.7-3) to root...
Downloading https://istore.linkease.com/repo/all/meta/app-meta-istoreenhance_0.1.7-3_all.ipk
Installing luci-i18n-istoreenhance-zh-cn (git-24.194.36967-61083fc) to root...
Downloading https://istore.linkease.com/repo/all/nas_luci/luci-i18n-istoreenhance-zh-cn_git-24.194.36967-61083fc_all.ipk
Collected errors:
 * opkg_conf_parse_file: Duplicate src declaration (customfeed https://op.dllkids.xyz/packages/aarch64_cortex-a53). Skipping.
 * check_data_file_clashes: Package luci-i18n-istoreenhance-zh-cn wants to install file /usr/lib/lua/luci/i18n/istoreenhance.zh-cn.lmo
        But that file is already provided by package  * luci-app-istoreenhance
 * opkg_install_cmd: Cannot install package app-meta-istoreenhance.

jjm2473, 2024-10-31:
`customfeed https://op.dllkids.xyz/packages/aarch64_cortex-a53` 这个软件源是不是自己加的，删掉。日志提示你，有两个相同名称的软件源了，你是不是添加了两个都叫“customfeed”的软件源？

jjm2473, 2024-10-31:
不要说下载istore增强了，你下载任何插件都会报错。

closed 2024-10-31 by jjm2473

kiritomemory, 2024-11-01:
是软件源的问题解决了