istoreos/istoreos #1873: N1刷iStoreOs出现的写入emmc以后无法启动

jswcph, 2024-10-26:
1.关于N1刷iStoreOs出现的写入emmc以后无法启动，只能再次重新刷机。但是F大的可以正常写入emmc

2. 我刷了大概20台的N1，iStoreOs出现写入emmc无法启动的发生在3台黑色上，F大的均可正常刷入

刷的是4月30的iStoreOs，查了大量资料，好像有很多人一样的问题。目前推断，应该是个别硬件的差异，感觉像是bootloader载入不正确，请求官方查一下这个问题

jjm2473, 2024-10-31:
N1的iStoreOS刷机过程跟armbian，flippy固件一样，只是制作u盘时使用的固件镜像换成istoreos就行了。分以下几种情况：

* 如果盒子是原厂固件：如果版本比较高要先降级到2.19（具体怎么降级b站很多教程），降级以后进安卓系统，联网获取ip，然后插上制作好的u盘，然后用电脑执行“adb connect 盒子ip”连接盒子，再执行“adb shell reboot update”，系统会自动重启几次，之后就会进入istoreos了。

* 如果盒子原本就是linux系统（非安卓），例如armbian或flippy的固件：那只要插上制作好的u盘即可启动到istoreos。

* 如果原本是砖，或者想完全恢复出厂状态：那么参考b站的救砖教程（可能需要拆机），救砖以后就按原厂固件刷机流程。

n1的istoreos默认网口是dhcp客户端，所以ip不是固定的，在连接显示器启动的情况下可以执行ip addr命令查看ip，否则在主路由后台看看最近分配的ip地址。

如果遇到emmc无法启动，实在不知道咋回事，就想办法把emmc刷回原厂的2.19固件，再按教程刷进emmc。

jianyuenet, 2024-12-14:
I can't use iStoreOS in my N1 box even I flash its img into my USB drive!!!!!!!!

yyehdjdbejxbskd, 2024-12-16:
大佬找到原因了吗？我也是出现相同的问题。几十个盒子都没问题，就出现了一个，对比了一下板子上的料， 没发现什么问题。

jswcph, 2024-12-16:
> N1的iStoreOS刷机过程跟armbian，flippy固件一样，只是制作u盘时使用的固件镜像换成istoreos就行了。分以下几种情况：
>
>
>
> * 如果盒子是原厂固件：如果版本比较高要先降级到2.19（具体怎么降级b站很多教程），降级以后进安卓系统，联网获取ip，然后插上制作好的u盘，然后用电脑执行“adb connect 盒子ip”连接盒子，再执行“adb shell reboot [...]

可以明确回复您，出现这种问题的 你不管是救砖刷还是从别的系统都无法刷进emmc

yyehdjdbejxbskd, 2024-12-16:
目前刷成麻雀云定制的armbian，能正常写入开机，刷istoreos 写入emmc就不行了，电流都不启动，emmc我自己换成256g的了

jswcph, 2024-12-16:
> 大佬找到原因了吗？我也是出现相同的问题。几十个盒子都没问题，就出现了一个，对比了一下板子上的料， 没发现什么问题。

没有，而且发现了新的问题，有部分pcdn的系统  可以直接U盘启动F大系统，却无法U盘启动istoreOS系统。  即便是刷进Emmc后的F大系统，也无法U盘启动istoreos  
然后这台机器拆机短接恢复到原厂系统  就可以刷istoreos而且也可以写进emmc    这是最近发现的某个pcdn系统的底层问题

总结，只要重新刷回原厂系统 还都不能正常写入istoreOS到emmc的N1盒子    就只能刷F大系统或其他安卓系统   目前无解

yyehdjdbejxbskd, 2024-12-16:
我的扩容到256g了，原厂系统能搞吗？一直都是救砖包刷系统的，没从原厂系统弄，试试你的办法

jswcph, 2024-12-18:
> 从官方的2.19的 刷入istoreos 写入EMMC 重启就凉了，估计BOOT都没了 电流都不起来，

是的，官方2.19能U盘启动，但是一写入emmc以后   整个系统崩掉  需要救砖重刷

yyehdjdbejxbskd, 2024-12-18:
放弃了 打算重新买一个

jswcph, 2024-12-18:
> 放弃了 打算重新买一个
> 
> 
> 
> 
> 
> 
> 
> ---原始邮件---
> 
> 发件人: ***@***.***&gt;
> 
> 发送时间: 2024年12月18日(周三) 中午11:42
> 
> 收件人: ***@***.***&gt;;
> 
> 抄送: ***@***.******@***.***&gt;;
> 
> 
> 
> 
> 
> 
> 
> 
> 
> 
>   
> 
> 从官方的2.19的 刷入istoreos 写入EMMC 重启就凉了，估计BOOT都没了 电流都不起来，
> 
>   
> 
> 是的，官方2.19能U盘启动，但是一写入emmc以后   整个系统崩掉  需要救砖重刷

可以刷F大  然后再安装istore商店    二合一  更强了🤣   目前刷不了的我都这样做

yyehdjdbejxbskd, 2024-12-21:
我刷了OPENWRT.AI的镜像目前没啥问题，可能是istoreos镜像的写入方式和其他不一样，其他op都是网页那样写的，istore是命令行。

yyehdjdbejxbskd, 2024-12-21:
openwrt.ai的镜像dockers的青龙面板起不来， 换了EMMC也不行，看个电视得了。没的招

jswcph, 2024-12-21:
> openwrt.ai的镜像dockers的青龙面板起不来， 换了EMMC也不行，看个电视得了。没的招

你说的是那个Kwrt   我刷过 好像N1无法写入emmc   不知道是我操作的问题吗

yyehdjdbejxbskd, 2024-12-21:
是的 kwrt  我这入了  可惜docker不行

jswcph, 2024-12-21:
> 是的 kwrt  我这入了  可惜docker不行  
> 
> 
> 
> 
> 
> 
> 
> ---原始邮件---
> 
> 发件人: ***@***.***&gt;
> 
> 发送时间: 2024年12月21日(周六) 下午3:26
> 
> 收件人: ***@***.***&gt;;
> 
> 抄送: ***@***.******@***.***&gt;;
> 
> 
> 
> 
> 
> 
> 
> 
> 
> 
>   
> 
> openwrt.ai的镜像dockers的青龙面板起不来， 换了EMMC也不行，看个电视得了。没的招
> 
>   
> 
> 你说的是那个Kwrt   我刷过 好像N1无法写入emmc   不知道是我操作的问题吗

你要用啥，F大的docker也很稳定啊  你刷F大的不就行了吗

yyehdjdbejxbskd, 2024-12-21:
刷了个joyo的双系统 改了64g的emmc准备卖了  手里还有个256g的emmc  改天再买个n1

  
是的 kwrt  我这入了  可惜docker不行

kc-bao, 2025-10-08:
此问题已在 **酷友社** 上被提及。那里可能有相关详细信息：

https://www.koolcenter.com/t/topic/2001/1

jjm2473, 2026-04-03:
> 同样的问题，u盘启动进入iStoreOS, 运行install-to-emmc.sh写入后，就进不去系统了，黑屏。

最新版本 24.10.5-2026032011 还有这个问题吗？

jjm2473, 2026-04-03:
试试这里的安装脚本：
[n1-scripts.tgz](https://github.com/user-attachments/files/26459695/n1-scripts.tgz)

解压后里面有两个脚本：
* install-to-emmc-keepdiskid.sh
* install-to-emmc-fdisk.sh

看看哪个安装以后可以启动，或者两个都不能启动，都反馈下。