istoreos/istoreos #1885: istoreOS作为主路由，无法通过IPV6地址访问管理页面的问题

ruokay1212, 2024-11-04:
1、环境说明：istoreOS作为主路由，光猫桥接，主路由拨号，lan口接了一个TPLINK二级路由，家庭大多数设备接在二级路由下
2、问题说明：wan口和lan口及接入的设备均能正确获得ipv4和ipv6地址，但是主路由即istoreOS通过ipv6地址无法访问（局域网内），ipv4地址可以正常访问，lan口下的二级路由tplink（包括二级路由下的设备）则均可以通过ipv4和ipv6地址正常访问，这是什么原因呢？？？
[image]
[image]
[image]
[image]

138161, 2024-11-06:
lan口接了一个TPLINK二级路由，家庭大多数设备接在二级路由下
这是问题所在，istore软路由和二级路由器下的设备，他们之间在ipv6环境下，类似于是直接相连的，就是说，istore软路由的lan侧ipv6与二级路由器下的设备的ipv6，在ipv6层面是直接相连，才能通讯，但实际上，他们之间隔了一个二级路由器，该路由器没有建立上下级的ipv6转发通路功能（将上级istore软路由的ipv6转发到下级PC的ipv6，反之上传也是，该路由器没有设定这样的功能，没有ipv6地址表的维护），因此，二级路由器可以访问istore，但二级路由器下的PC则不行。这是ipv4与ipv6的区别

138161, 2024-11-06:
二级路由器改为AP模式，不要使用路由器模式，就可以访问软路由了，现在的二级路由器，wan侧能得到的都是64位ipv6地址，他是没有办法自动再为其下面的PC设备，再分配ipv6地址的，因为64位ipv6是无法再分配出ipv6地址的，要小于64位，比如56位，才能再次划分出更多的64位地址

ruokay1212, 2024-11-07:
> 二级路由器改为AP模式，不要使用路由器模式，就可以访问软路由了，现在的二级路由器，wan侧能得到的都是64位ipv6地址，他是没有办法自动再为其下面的PC设备，再分配ipv6地址的，因为64位ipv6是无法再分配出ipv6地址的，要小于64位，比如56位，才能再次划分出更多的64位地址

感谢大佬您的回复，我仔细阅读了您的回复，您的意思我明白了，确实我可以尝试一下您的建议把二级路由 tplink 改成 ap 模式，在此之前我还想补充一下额外的情况，还请您帮忙看一下，除了二级路由下无法 ipv6 访问一级路由 istoreOS 以外，还存在外网无法访问一级路由的 ipv6 的地址，但是可以访问二级路由下设备的 ipv6 的地址（我二级路由下有个虚拟机，安装了群晖 nas，可以得到 ipv6 地址，且外网能够正常访问），且二级路由下的设备，无论 pc （但是 pc 在 ipv6test 这个网站测试时，提示没有 ipv6 地址） 还是虚拟机还是其他手机等设备，均能够得到 ipv6 地址，其实就是一级路由的 ipv6 地址访问均存在内外网无法访问的情况。其实事件的起因是因为一级路由 istoreos 安装 ddnsgo 使用域名解析时，提示无法通过接口获取 ipv6 地址，但是却可以通过网卡获取到 ipv6 地址，这个问题困扰我许久，现在个人的猜测是不是我的 dns 配置存在问题，目前均是设置的自动配置。再次感谢您的回复

138161, 2024-11-07:
一、但是可以访问二级路由下设备的 ipv6 的地址
是内外PC可以访问吧
二、二级路由下的设备，无论 pc （但是 pc 在 ipv6test 这个网站测试时，提示没有 ipv6 地址） 还是虚拟机还是其他手机等设备，均能够得到 ipv6 地址，
能得到地址很正常，地址是istore通过SLAAC分配的，但由于存在二级路由器，PC向上级软路由上传信息，这个通路是由二级路由器的ipv6地址表来维护保证的，除非你会人工写代码。使得二级路由器有此功能

ruokay1212, 2024-11-07:
> 一、但是可以访问二级路由下设备的 ipv6 的地址 是内外PC可以访问吧 二、二级路由下的设备，无论 pc （但是 pc 在 ipv6test 这个网站测试时，提示没有 ipv6 地址） 还是虚拟机还是其他手机等设备，均能够得到 ipv6 地址， 能得到地址很正常，地址是istore通过SLAAC分配的，但由于存在二级路由器，PC向上级软路由上传信息，这个通路是由二级路由器的ipv6地址表来维护保证的，除非你会人工写代码。使得二级路由器有此功能

确实如您所说，是内外pc都可以ipv6访问二级路由下的设备，但是访问不到一级路由下的，之前我的验证有不准确的地方。我刚测试了外网是访问不到一级路由（istoreOS）的ipv6地址，提示拒绝了连接请求，外网可以访问到一级路由器下面的二级路由（路由模式下的）TPlink的ipv6 wan口地址，但是二级路由下的设备又无法访问，确实如您所说，是二级路由在此中间的问题。感谢您的耐心解答，让我对IPV4和ipv6的不同深有感触了一回。作为小白已经在尽力去理解了。真心感谢~~

closed 2024-11-11 by ruokay1212