istoreos/istoreos #1898: opkg update 报错失败 opkg_download: Failed to download https://istore.linkease.com/repo/all/compat/Packages.gz, curl returned 35.

WvvvWv, 2024-11-12:
Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [ x ] 没有类似的issue
有类似curl的issus，但是没有解决我这边的问题

## 2. 详细叙述

### (1) 具体问题
商店手动安装passwall2提示失败，然后又试了opkg 更新发现也失败。

### (2) 路由器型号和固件版本
J4125工控机，Pve All In Boom

设备型号
QEMU Standard PC (Q35 + ICH9, 2009)
固件版本
iStoreOS 22.03.7 2024102509
内核版本
5.10.221

### (3) 详细日志和/或截图

应用商店手动安装passwall2失败报错：
[image]

SSH手动执行opkg update报错：
[image]

软件包功能里点击更新列表报错：
[image]

1JI0O, 2024-11-12:
我也遇到这个问题，环境相同，报错相同

myweihp, 2024-11-12:
遇到同样问题+1

mcbianxiao, 2024-11-12:
遇到同样问题+1

solar020, 2024-11-12:
同样的问题，任何软件都安装不进

bcseputetto, 2024-11-12:
https://github.com/bcseputetto/Are-u-ok/issues/23

Derek-Aaron, 2024-11-12:
也是同样的问题，手动上传包也不行的
[image]

WvvvWv, 2024-11-13:
> [bcseputetto/Are-u-ok#23](https://github.com/bcseputetto/Are-u-ok/issues/23)

@bcseputetto 多谢，问题已解决

closed 2024-11-13 by WvvvWv

weisonglin8, 2024-11-14:
我也是同样的问题