istoreos/istoreos #1899: Istore无法安装软件

hujunchen715, 2024-11-12:
提示
[image]
Fetch feed list for aarch64_cortex-a53
curl: (35) OpenSSL/1.1.1w: error:1408F10B:SSL routines:ssl3_get_record:wrong version number

mcbianxiao, 2024-11-12:
一样

lilu0826, 2024-11-12:
+1 我还以为是我curl出了问题，我还去下载了一个新的curl试了，也不行。

yesircl, 2024-11-12:
通知: 2024-11-12 服务器故障，暂时无法安装插件，请等待修复
https://github.com/linkease/istore/issues/22

abao-chen, 2024-11-12:
+1，害得我倒腾了半天，我也以为是手贱升级了curl的原因，到处找应该回退到哪个版本。。。

leedaaye, 2024-11-12:
无语死。。。倒腾了一天。。。。。没想到是服务器故障

majunfei, 2024-11-12:
什么原因呢

musingsz, 2024-11-13:
害我重新装个系统两次都没有成功

betsylotus, 2024-11-13:
目前修复好了吗

OrenZhang, 2024-11-13:
[image]

betsylotus, 2024-11-14:
升级一下，目前已经可以正常安装软件了

TohsakaOrz, 2024-11-14:
@betsylotus 请问如何更新呢？istore->维护 中显示没有新版本

betsylotus, 2024-11-14:
iStore [v0.1.26-3] 这个版本

> @betsylotus 请问如何更新呢？istore->维护 中显示没有新版本

TohsakaOrz, 2024-11-14:
> iStore [v0.1.26-3] 这个版本
> 
> > @betsylotus 请问如何更新呢？istore->维护 中显示没有新版本

我目前版本是 v0.1.26-0  是需要手动更新么？自动检测不到新版本

betsylotus, 2024-11-14:
> > iStore [v0.1.26-3] 这个版本
> > > @betsylotus 请问如何更新呢？istore->维护 中显示没有新版本
> 
> 我目前版本是 v0.1.26-0 是需要手动更新么？自动检测不到新版本

我是直接点了维护，他就有新版本提示的

jiayiming001, 2024-11-14:
https://github.com/bcseputetto/Are-u-ok/issues/23#issue-2651642502

用这个办法可以临时解决下，用的好点个赞，看看适不适用大多数人

yesircl, 2024-11-14:
之前的服务器修复需要些时日，执行此命令

sh -c "$(curl -sSL https://fw.koolcenter.com/iStoreOS/alpha/fix-istore.sh)"
切换到新服务器，之后都会使用新服务器

ridgepeng, 2024-11-14:
。。。。。我折腾了好久

CBY1314, 2024-11-14:
这个架构怎么安装istore？我用的lmmortalWrt？

dongcheng8088, 2024-11-15:
我是一个刚想用istoreos的小白，刚买了一个迷你主机，想试一下，几天前安装配置好了一次，昨天想重新练习一遍，没想到碰到这个问题，折腾了两天了。

YUAXI, 2024-11-15:
折腾了许久，重装了好几次各种方式重装系统，期待修复

Fred070707, 2024-11-17:
sh -c "$(curl -sSL https://fw.koolcenter.com/iStoreOS/alpha/fix-istore.sh)"

cuianbing, 2024-11-18:
还不可以吗？

keai135, 2024-11-19:
请问需要多久可以解决？手动安装插件也会提示软件源有问题不能安装：
-rw-r--r--    1 root     root      22039743 Nov 19 11:31 /tmp/is-root/tmp/OpenClash_0.46.014+x86_64_core.run
fc2c356e85f4873746216c838609da4d  /tmp/is-root/tmp/OpenClash_0.46.014+x86_64_core.run
Verifying archive integrity...  100%   MD5 checksums are OK. All good.
Uncompressing luci-app-openclash_with_clash_core  100%  
Downloading https://istore.linkease.com/repo/all/compat/Packages.gz
curl: (35) OpenSSL/1.1.1w: error:1408F10B:SSL routines:ssl3_get_record:wrong version number
*** Failed to download the package list from https://istore.linkease.com/repo/all/compat/Packages.gz

Downloading https://mirrors.tuna.tsinghua.edu.cn/openwrt/releases/22.03.7/targets/x86/64/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_core
Downloading https://mirrors.tuna.tsinghua.edu.cn/openwrt/releases/22.03.7/targets/x86/64/packages/Packages.sig
Signature check passed.
Downloading https://mirrors.tuna.tsinghua.edu.cn/openwrt/releases/22.03.7/packages/x86_64/base/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_base
Downloading https://mirrors.tuna.tsinghua.edu.cn/openwrt/releases/22.03.7/packages/x86_64/base/Packages.sig
Signature check passed.
Downloading https://mirrors.tuna.tsinghua.edu.cn/openwrt/releases/22.03.7/packages/x86_64/luci/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_luci
Downloading https://mirrors.tuna.tsinghua.edu.cn/openwrt/releases/22.03.7/packages/x86_64/luci/Packages.sig
Signature check passed.
Downloading https://mirrors.tuna.tsinghua.edu.cn/openwrt/releases/22.03.7/packages/x86_64/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_packages
Downloading https://mirrors.tuna.tsinghua.edu.cn/openwrt/releases/22.03.7/packages/x86_64/packages/Packages.sig
Signature check passed.
Downloading https://mirrors.tuna.tsinghua.edu.cn/openwrt/releases/22.03.7/packages/x86_64/routing/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_routing
Downloading https://mirrors.tuna.tsinghua.edu.cn/openwrt/releases/22.03.7/packages/x86_64/routing/Packages.sig
Signature check passed.
Downloading https://mirrors.tuna.tsinghua.edu.cn/openwrt/releases/22.03.7/packages/x86_64/telephony/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_telephony
Downloading https://mirrors.tuna.tsinghua.edu.cn/openwrt/releases/22.03.7/packages/x86_64/telephony/Packages.sig
Signature check passed.
Collected errors:
 * opkg_download: Failed to download https://istore.linkease.com/repo/all/compat/Packages.gz, curl returned 35.
更新软件源列表错误，请检查路由器自身网络连接以及是否有失效的软件源。

CBY1314, 2024-11-19:
你这软件源有Openclash吗？

hotwa, 2024-11-19:
> 你这软件源有Openclash吗？

https://github.com/AUK9527/Are-u-ok

nobody0222, 2024-11-20:
之前也遇到curl error 35的问题，已经按照说明切换到新服务器解决了。
但是我在安装https://github.com/AUK9527/Are-u-ok/raw/main/apps/all/PassWall2_1.30-3_aarch64_a53_all_sdk_22.03.7.run时遇到找不到tcping包的问题：
[image]
[image]

keai135, 2024-11-20:
使用首页终端命令运行sh -c "$(curl -sSL http://fw.koolcenter.com/iStoreOS/alpha/fix-istore.sh)"无效的更换浏览器也不行的话，可以使用ssh工具运行命令，然后再istore商店里点维护按钮，更新到[v0.1.26-3]版本就可以了
我的软件商店目前已经恢复正常，手动安装也正常

akapzg, 2024-11-20:
[image]
为啥无法连接到服务器

nobody0222, 2024-11-21:
> 之前也遇到curl error 35的问题，已经按照说明切换到新服务器解决了。 [...]

找到原因了，我是x86_64平台，然后下成[aarch64_cortex-a53平台](https://github.com/AUK9527/Are-u-ok/tree/main/apps)的包了（sry

akapzg, 2024-11-21:
> > <img alt="image" width="54.140625" height="21" src="https://private-user-images.githubusercontent.com/166852812/387989687-a0556444-566a-41d1-8786-92db943429b7.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzIxMTY2NjksIm5iZiI6MTczMjExNjM2OSwicGF0aCI6Ii8xNjY4NTI4MTIvMzg3OTg5Njg3LWEwNTU2NDQ0LTU2NmEtNDFkMS04Nzg2LTkyZGI5NDM0MjliNy5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMTIwJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTEyMFQxNTI2MDlaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT05MDVjY2FkMjE5YzEzNGRmOWY0M2MyY2U4ZGEzMTQ4ZTM0YzFkYmEyM2M4MDNhM2ZiODUzNGFmZTJiNDVlNGNkJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.99mIkDVmhWs-EIUkB8ZYCQMIcGFfel8fRnbauPoHRI8"> 为啥无法连接到服务器
> 
> 你没发现换行了么，16行,fix-isto 后面，re.sh)"跑到17行去了，url改了肯定连不上啊

不是换行的问题,这是ssh工具显示的问题,实际上没有对url做改变,不然也不会出现下面的拉清单的信息的. 应该是对他们服务器的连接出问题了.我人在海外,用浏览器也进不去那个域名

Leoli0019, 2025-02-14:
1