istoreos/istoreos #1900: istore无法安装插件，全部提示错误！！捉急

syboy1, 2024-11-12:
## 2. 详细叙述

### (1) 具体问题
点了一下出厂设置，然后再安装istore里面的插件就提示错误，用本地文件上传安装提示passwall2，提示图2错误

### (2) 路由器型号和固件版本
[image]

### (3) 详细日志和/或截图
[image]
[image]

ql4321, 2024-11-12:
同样问题。。估计是网站有被攻击了。

nbfuping, 2024-11-13:
同样的问题，有办法解决吗？

CharlesLocke, 2024-11-14:
别人已经给出解决办法了，你们怎么就是不看呢。
[image]
[image]

mmmm750519, 2024-11-16:
试了，没行

balge, 2024-11-16:
> 试了，没行

 * opkg_download: Failed to download https://mirrors.cernet.edu.cn/openwrt/releases/22.03.7/packages/x86_64/routing/Packages.gz, curl returned 6.
 * opkg_download: Check your network settings and connectivity.

 * opkg_download: Failed to download https://mirrors.cernet.edu.cn/openwrt/releases/22.03.7/packages/x86_64/telephony/Packages.gz, curl returned 6.
 * opkg_download: Check your network settings and connectivity.

更新软件源列表错误，请检查路由器自身网络连接以及是否有失效的软件源。

是的，还是不行，我还只行了sh命令，但是这个命令的也报无法连接

bcseputetto, 2024-11-16:
@balge 你这个是DNS问题，根本都还没到只有KoolCenter源报错的程度，根本就是网络没有设置清楚，上面解决KoolCenter源连接失败是只有KoolCenter源报错，你这个是所有源都报错

balge, 2024-11-16:
> @balge 你这个是DNS问题，根本都还没到只有KoolCenter源报错的程度，根本就是网络没有设置清楚，上面解决KoolCenter源连接失败是只有KoolCenter源报错，你这个是所有源都报错

谢谢提醒，仔细检查，已解决

Ray2023m, 2024-11-17:
[image]

Ray2023m, 2024-11-17:
sh -c "$(curl -sSL https://fw.koolcenter.com/iStoreOS/alpha/fix-istore.sh)"

syboy1, 2024-11-17:
感谢，切换后没问题了

QuantumLeap-us, 2024-11-17:
> sh -c "$(curl -sSL https://fw.koolcenter.com/iStoreOS/alpha/fix-istore.sh)"
已经挂了

wolf826, 2024-11-18:
给指点下…
我也是旁路由模式，在istore手动安装passwall2(软件包在本系统提供的地方下载的) 
提示：更新软件源列表错误…

bcseputetto, 2024-11-18:
> 给指点下… 我也是旁路由模式，在istore手动安装passwall2(软件包在本系统提供的地方下载的) 提示：更新软件源列表错误…

@wolf826 提供报错信息

sakkyakei, 2024-12-05:
我试过了 不行 
sh -c "$(curl -sSL https://fw.koolcenter.com/iStoreOS/alpha/fix-istore.sh)"
Fetch feed list for aarch64_cortex-a53
curl: (35) OpenSSL SSL_connect: SSL_ERROR_SYSCALL in connection to istore.istoreos.com:443

mamamii, 2025-01-20:
今天我也碰到这个问题，bcseputetto的说法正确，固件默认的DNS服务器（223.5.5.5）不能工作，换其他的如8.8.8.8或者直接用主路由的地址都行。

yaboooooo, 2025-03-29:
-rw-r--r--    1 root     root      56405887 Mar 29 21:40 /tmp/is-root/tmp/PassWall2_25.3.2_x86_64_all_sdk_22.03.7.run
64983ee19227cc35a81ad0ad79181446  /tmp/is-root/tmp/PassWall2_25.3.2_x86_64_all_sdk_22.03.7.run
Verifying archive integrity...  100%   MD5 checksums are OK. All good.
Uncompressing PassWall2_25.3.2_with_sdk_22.03.7_libopenssl_1.1  100%  
Downloading https://istore.istoreos.com/repo/all/compat/Packages.gz
Updated list of available packages in /var/opkg-lists/istore_compat
Downloading https://istore.istoreos.com/repo/all/compat/Packages.sig
Signature check passed.
Downloading https://mirrors.cernet.edu.cn/openwrt/releases/22.03.7/targets/x86/64/packages/Packages.gz
curl: (22) The requested URL returned error: 403
*** Failed to download the package list from https://mirrors.cernet.edu.cn/openwrt/releases/22.03.7/targets/x86/64/packages/Packages.gz

Downloading https://mirrors.cernet.edu.cn/openwrt/releases/22.03.7/packages/x86_64/base/Packages.gz
curl: (22) The requested URL returned error: 403
*** Failed to download the package list from https://mirrors.cernet.edu.cn/openwrt/releases/22.03.7/packages/x86_64/base/Packages.gz

Downloading https://mirrors.cernet.edu.cn/openwrt/releases/22.03.7/packages/x86_64/luci/Packages.gz
curl: (56) Failure when receiving data from the peer
*** Failed to download the package list from https://mirrors.cernet.edu.cn/openwrt/releases/22.03.7/packages/x86_64/luci/Packages.gz

Downloading https://mirrors.cernet.edu.cn/openwrt/releases/22.03.7/packages/x86_64/packages/Packages.gz
curl: (22) The requested URL returned error: 403
*** Failed to download the package list from https://mirrors.cernet.edu.cn/openwrt/releases/22.03.7/packages/x86_64/packages/Packages.gz

Downloading https://mirrors.cernet.edu.cn/openwrt/releases/22.03.7/packages/x86_64/routing/Packages.gz
curl: (22) The requested URL returned error: 403
*** Failed to download the package list from https://mirrors.cernet.edu.cn/openwrt/releases/22.03.7/packages/x86_64/routing/Packages.gz

Downloading https://mirrors.cernet.edu.cn/openwrt/releases/22.03.7/packages/x86_64/telephony/Packages.gz
curl: (22) The requested URL returned error: 403
*** Failed to download the package list from https://mirrors.cernet.edu.cn/openwrt/releases/22.03.7/packages/x86_64/telephony/Packages.gz

Collected errors:
 * opkg_download: Failed to download https://mirrors.cernet.edu.cn/openwrt/releases/22.03.7/targets/x86/64/packages/Packages.gz, curl returned 22.
 * opkg_download: Failed to download https://mirrors.cernet.edu.cn/openwrt/releases/22.03.7/packages/x86_64/base/Packages.gz, curl returned 22.
 * opkg_download: Failed to download https://mirrors.cernet.edu.cn/openwrt/releases/22.03.7/packages/x86_64/luci/Packages.gz, curl returned 56.
 * opkg_download: Failed to download https://mirrors.cernet.edu.cn/openwrt/releases/22.03.7/packages/x86_64/packages/Packages.gz, curl returned 22.
 * opkg_download: Failed to download https://mirrors.cernet.edu.cn/openwrt/releases/22.03.7/packages/x86_64/routing/Packages.gz, curl returned 22.
 * opkg_download: Failed to download https://mirrors.cernet.edu.cn/openwrt/releases/22.03.7/packages/x86_64/telephony/Packages.gz, curl returned 22.
更新软件源列表错误，请检查路由器自身网络连接以及是否有失效的软件源。

大佬们，看看我的咋回事呢。

Tsubamei, 2025-09-21:
> -rw-r--r-- 1 root root 56405887 Mar 29 21:40 /tmp/is-root/tmp/PassWall2_25.3.2_x86_64_all_sdk_22.03.7.run 64983ee19227cc35a81ad0ad79181446 [...]

【系统】【软件包】【配置OPKG…】，最下面的【/etc/opkg/distfeeds.conf】，里面内容全删了，替换为下面的：
src/gz openwrt_base https://mirrors.tuna.tsinghua.edu.cn/openwrt/releases/22.03.7/packages/aarch64_cortex-a53/base
src/gz openwrt_luci https://mirrors.tuna.tsinghua.edu.cn/openwrt/releases/22.03.7/packages/aarch64_cortex-a53/luci
src/gz openwrt_packages https://mirrors.tuna.tsinghua.edu.cn/openwrt/releases/22.03.7/packages/aarch64_cortex-a53/packages
src/gz openwrt_routing https://mirrors.tuna.tsinghua.edu.cn/openwrt/releases/22.03.7/packages/aarch64_cortex-a53/routing
src/gz openwrt_telephony https://mirrors.tuna.tsinghua.edu.cn/openwrt/releases/22.03.7/packages/aarch64_cortex-a53/telephony