istoreos/istoreos #1905: 更新 firewall4

llhdxnq, 2024-11-14:
什么时候更新 firewall4现在有的插件都安装不了[image]

bcseputetto, 2024-11-15:
你换成firewall4之后，也有一些软件也会用不了
比如ssrp，比如UU加速器，看你怎么取舍。不是非要用一些只支持firewall4的梯子客户端
openclash，passwall之类的firewall3,firewall4都支持。
[image]
下个分支版本，24.10可能会切换到firewall4。到时候需要用UU加速器的用户又要说“怎么用不了了”

llhdxnq, 2024-11-15:
> 你换成firewall4之后，也有一些软件也会用不了
> 
> 比如ssrp，比如UU加速器，看你怎么取舍。不是非要用一些只支持firewall4的梯子客户端
> 
> openclash，passwall之类的firewall3,firewall4都支持。
> 
> [image]
> 
> 下个分支版本，24.10可能会切换到firewall4。到时候需要用UU加速器的用户又要说“怎么用不了了”

你说的这些软件我完全用不上

llhdxnq, 2024-11-15:
> > 你说的这些软件我完全用不上
> 
> 
> 
> 那已经告诉你了，24.10分支可能就会切到fw4，先等待OpenWrt官方release 24.10的稳定版本，后续iStoreOS才会跟进

ok

DavideSOGG, 2024-12-28:
UU加速器在8.0.12的版本中已经支持了Openwrt23.X的nftables环境。所以想用游戏加速器的朋友们可以放心了。