istoreos/istoreos #1913: 安装 Entware 插件后软件中心面板上未检测到环境，但 SSH 端验证 opkg 可用

MuGemBot, 2024-11-18:
安装Entware插件后软件中心面板上未检测到环境，但SSH端验证opkg 可用
软件中心上其它插件的日志显示，面板上的插件全都无法检测到opkg
然而 SSH 登进路由器系统中发现 opkg install update -v 啥的都正常

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：安装Entware插件后软件中心面板上未检测到环境，但SSH端验证opkg 可用

### (2) 路由器型号和固件版本

A：华硕 梅林固件 386.10 

### (3) 详细日志和/或截图

A：
[image]
[image]

不确定是否属于域名备案连带问题

另外想查看 entware 源码排查问题，发现 https://github.com/koolshare/rogsoft/tree/master/entware/entware 里面的 .sh 脚本全都压缩加密了，不可读，经过试验还不是 gzexe 和 upx 加密的

MuGemBot, 2024-11-20:
应该是插件脚本环境变量问题

closed 2024-11-20 by MuGemBot

catalysia, 2024-12-14:
请问你解决了吗 我也是和你同样的问题 不过是在koolshare下