istoreos/istoreos #1918: 安装好之后，无法安装软件

liqiang945, 2024-11-21:
安装了最新的版本22.03.7 2024102509
 在安装软件的时候，无论是应用商店还是手动安装run包，都不行。
-rw-r--r--    1 root     root      45471828 Nov 21 17:18 /tmp/is-root/tmp/SSR-Plus_188_x86_64_all_sdk_22.03.7.run
bc2c50552e0183681b344022ae4f5ada  /tmp/is-root/tmp/SSR-Plus_188_x86_64_all_sdk_22.03.7.run
Verifying archive integrity...  100%   MD5 checksums are OK. All good.
Uncompressing ssrp_188_with_sdk_22.03.7_libopenssl1.1  100%  
Downloading https://istore.istoreos.com/repo/all/compat/Packages.gz
Updated list of available packages in /var/opkg-lists/istore_compat
Downloading https://istore.istoreos.com/repo/all/compat/Packages.sig
Signature check passed.
Downloading https://mirrors.cernet.edu.cn/openwrt/releases/22.03.7/targets/x86/64/packages/Packages.gz
curl: (22) The requested URL returned error: 403
*** Failed to download the package list from https://mirrors.cernet.edu.cn/openwrt/releases/22.03.7/targets/x86/64/packages/Packages.gz

Downloading https://mirrors.cernet.edu.cn/openwrt/releases/22.03.7/packages/x86_64/base/Packages.gz
curl: (22) The requested URL returned error: 403
*** Failed to download the package list from https://mirrors.cernet.edu.cn/openwrt/releases/22.03.7/packages/x86_64/base/Packages.gz

Downloading https://mirrors.cernet.edu.cn/openwrt/releases/22.03.7/packages/x86_64/luci/Packages.gz
curl: (22) The requested URL returned error: 403
*** Failed to download the package list from https://mirrors.cernet.edu.cn/openwrt/releases/22.03.7/packages/x86_64/luci/Packages.gz

Downloading https://mirrors.cernet.edu.cn/openwrt/releases/22.03.7/packages/x86_64/packages/Packages.gz
curl: (22) The requested URL returned error: 403
*** Failed to download the package list from https://mirrors.cernet.edu.cn/openwrt/releases/22.03.7/packages/x86_64/packages/Packages.gz

Downloading https://mirrors.cernet.edu.cn/openwrt/releases/22.03.7/packages/x86_64/routing/Packages.gz
curl: (22) The requested URL returned error: 403
*** Failed to download the package list from https://mirrors.cernet.edu.cn/openwrt/releases/22.03.7/packages/x86_64/routing/Packages.gz

Downloading https://mirrors.cernet.edu.cn/openwrt/releases/22.03.7/packages/x86_64/telephony/Packages.gz
curl: (22) The requested URL returned error: 403
*** Failed to download the package list from https://mirrors.cernet.edu.cn/openwrt/releases/22.03.7/packages/x86_64/telephony/Packages.gz

Collected errors:
 * opkg_download: Failed to download https://mirrors.cernet.edu.cn/openwrt/releases/22.03.7/targets/x86/64/packages/Packages.gz, curl returned 22.
 * opkg_download: Failed to download https://mirrors.cernet.edu.cn/openwrt/releases/22.03.7/packages/x86_64/base/Packages.gz, curl returned 22.
 * opkg_download: Failed to download https://mirrors.cernet.edu.cn/openwrt/releases/22.03.7/packages/x86_64/luci/Packages.gz, curl returned 22.
 * opkg_download: Failed to download https://mirrors.cernet.edu.cn/openwrt/releases/22.03.7/packages/x86_64/packages/Packages.gz, curl returned 22.
 * opkg_download: Failed to download https://mirrors.cernet.edu.cn/openwrt/releases/22.03.7/packages/x86_64/routing/Packages.gz, curl returned 22.
 * opkg_download: Failed to download https://mirrors.cernet.edu.cn/openwrt/releases/22.03.7/packages/x86_64/telephony/Packages.gz, curl returned 22.
更新软件源列表错误，请检查路由器自身网络连接以及是否有失效的软件源。

harleylg, 2024-11-22:
改软件源那不叫问题解决啊……
默认的软件源域名是istor.linkease.com，没有备案被机房禁用了。
[image]

mrkoalaz, 2024-11-24:
官方在BiliBili上更新了视频，原因是7月份原服务器托管商倒闭了，他们迁移机房的时候忘记了多个网址的备案，结果备案到期，被gov给封了，最近正在加紧备案，但是快两周过去了......还没进展。
有人一直刷新不出来iStore，有人刷出来了安装不了应用，都是因为这个原因。