istoreos/istoreos #1919: 无法刷新出来软件商店

newsharpstar, 2024-11-23:
反馈bug/问题模板，提建议请删除

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [X] 没有类似的issue

## 2. 详细叙述
无法显示iStore商店
### (1) 具体问题

A：点击iStore->全部软件，无法显示iStore商店。停留在正”在努力的获取数据...“。尝试过重启路由器，依然无法正常显示iStore商店。

### (2) 路由器型号和固件版本

A：X86
固件版本：
iStoreOS 22.03.7 2024102509
软件商店版本：[v0.1.25-0]

### (3) 详细日志和/或截图

A：
[image]

yunxuo, 2024-11-24:
一样的问题 
[image]

mrkoalaz, 2024-11-24:
官方在BiliBili上更新了视频，原因是7月份原服务器托管商倒闭了，他们迁移机房的时候忘记了多个网址的备案，结果备案到期，被gov给封了，最近正在加紧备案，但是快两周过去了......还没进展。

newsharpstar, 2024-11-24:
> 官方在BiliBili上更新了视频，原因是7月份原服务器托管商倒闭了，他们迁移机房的时候忘记了多个网址的备案，结果备案到期，被gov给封了，最近正在加紧备案，但是快两周过去了......还没进展。 有人一直刷新不出来iStore，有人刷出来了安装不了应用，都是因为这个原因。

感谢您的恢复。等着备案吧。

newsharpstar, 2024-11-24:
> 一样的问题 <img alt="image" width="783" src="https://private-user-images.githubusercontent.com/34650575/389246695-ec0c7df9-1c51-4666-baa4-e1bbd4c4a9af.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzI0NDc5MzMsIm5iZiI6MTczMjQ0NzYzMywicGF0aCI6Ii8zNDY1MDU3NS8zODkyNDY2OTUtZWMwYzdkZjktMWM1MS00NjY2LWJhYTQtZTFiYmQ0YzRhOWFmLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDExMjQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMTI0VDExMjcxM1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTg3YzM2MTkyNmRiY2E0NGU5MDhiNjVhNDJiNzRhNGExYzRhMmEwYzY0ZWEzNjUyZGMyYzZjMTIxZTBiYmM3MDUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.reiv3A1gKigeEXEYlXsBPj7bIzUS-eSilPOpXaPmz2Q">

我看你的iStore商店版本比我的稍微高一些，我平时没有注意到商店版本怎么更新。您知道怎么更新商店版本么？

newsharpstar, 2024-11-24:
> > > 一样的问题
> > > 我看你的iStore商店版本比我的稍微高一些，我平时没有注意到商店版本怎么更新。您知道怎么更新商店版本么？
> 
> 先把软件中心修复了 然后istore-维护-更新 ![i](https://private-user-images.githubusercontent.com/184812586/389270340-71ca16f9-5677-4d3e-94fb-54f6dab15b5c.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzI0NTcyMzYsIm5iZiI6MTczMjQ1NjkzNiwicGF0aCI6Ii8xODQ4MTI1ODYvMzg5MjcwMzQwLTcxY2ExNmY5LTU2NzctNGQzZS05NGZiLTU0ZjZkYWIxNWI1Yy5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMTI0JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTEyNFQxNDAyMTZaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT02YTk3ZmM4OTA0MzAzNWMwMTc4ZTIxM2I4MDA1YzRmODdhNzA5ODJiYjE3MmM2MWVlNDA0YTE1YzQxMjkzMTBlJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9._9hmfC6Ets5-QxbdFCiK3NxkMw33shhORGHGmeWFIE0)
> 
> 或者自己编译上传安装 https://github.com/linkease/istore/tree/main/luci/luci-app-store

感谢！

yunxuo, 2024-11-25:
> > 官方在BiliBili上更新了视频，原因是7月份原服务器托管商倒闭了，他们迁移机房的时候忘记了多个网址的备案，结果备案到期，被gov给封了，最近正在加紧备案，但是快两周过去了......还没进展。 有人一直刷新不出来iStore，有人刷出来了安装不了应用，都是因为这个原因。

感谢

yunxuo, 2024-11-25:
> ![i](https://private-user-images.githubusercontent.com/184812586/389268308-d78426d0-50a3-4c64-9bf7-042d3e7ca6f1.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3Lmd [...]

有用！！

jjm2473, 2024-11-26:
点一下istore界面上的版本号，有公告怎么修复的

newsharpstar, 2024-11-27:
> 点一下istore界面上的版本号，有公告怎么修复的

牛！！！

Gfvtff, 2025-02-07:
sh -c "$(curl -sSL https://fw.koolcenter.com/iStoreOS/alpha/fix-istore.sh)"

a447948814, 2025-05-21:
> sh -c "$(curl -sSL https://fw.koolcenter.com/iStoreOS/alpha/fix-istore.sh)"

这个sh文件已经被删除了，，现在怎么办呢 我还是经常的打不开