istoreos/istoreos #1940: 近期iStore加载软件列表失败/安装失败问题

jjm2473, 2024-11-26:
点击iStore界面的版本号，会跳转到更新公告，其中有故障说明和修复方案。
或者点击此链接 https://github.com/linkease/istore/issues/22

sakkyakei, 2024-12-05:
我还是无法列出软件列表 
错误： 
sh -c "$(curl -sSL https://fw.koolcenter.com/iStoreOS/alpha/fix-istore.sh)"
Fetch feed list for aarch64_cortex-a53
curl: (35) OpenSSL SSL_connect: SSL_ERROR_SYSCALL in connection to istore.istoreos.com:443

Vash90, 2024-12-05:
一样的，海外还是不能加载出 istore商店

whj505, 2024-12-09:
我是海外用戶,依提示安裝服務器路徑,目前仍無法看到軟件列表.
root@iSOS-R2S:~# sh -c "$(curl -sSL https://fw.koolcenter.com/iStoreOS/alpha/fix-istore.sh)"
Fetch feed list for aarch64_generic
curl: (28) Failed to connect to istore.istoreos.com port 443 after 132403 ms: Couldn't connect to server

ZPC00, 2024-12-09:
海外用户：
root@GL-MT3000:~# sh -c "$(curl -sSL https://fw.koolcenter.com/iStoreOS/alpha/fix-istore.sh)"
Fetch feed list for aarch64_cortex-a53
curl: (28) Failed to connect to istore.istoreos.com port 443 after 129729 ms: Operation timed out

jjm2473, 2024-12-16:
现在海外用户会DNS解析到不同的服务器上，已经完全恢复正常了。

closed 2024-12-16 by jjm2473