istoreos/istoreos #1941: 请问 S.M.A.R.T. 是不是检测不到 外挂的USB硬盘？

TMODER, 2024-11-28:
求教一个问题，希望大佬解答下。

N1 盒子刷的 iStoreOS ，挂载一个 USB 硬盘。

S.M.A.R.T. 选项开启  
日志显示：
Thu Nov 28 09:31:58 2024 daemon.info smartd[15351]: smartd 7.2 2020-12-30 r5155 [aarch64-linux-5.10.201] (localbuild)
Thu Nov 28 09:31:58 2024 daemon.info smartd[15351]: Copyright (C) 2002-20, Bruce Allen, Christian Franke, www.smartmontools.org
Thu Nov 28 09:31:58 2024 daemon.info smartd[15351]: Opened configuration file /etc/smartd.conf
Thu Nov 28 09:31:58 2024 daemon.crit smartd[15351]: Configuration file /etc/smartd.conf parsed but has no entries
Thu Nov 28 09:31:58 2024 daemon.info smartd[15351]: Monitoring 0 ATA/SATA, 0 SCSI/SAS and 0 NVMe devices

新建计划任务也选不到磁盘。

TMODER, 2024-11-28:
网上找了下资料，应该是默认只能看到内置的  S.M.A.R.T  硬盘信息，比如PCIE/SATA 这种。
USB虚拟的硬盘 应该需要 驱动/设备 支持，外挂的USB硬盘，需要发对应控制帧命令读取解析 S.M.A.R.T 信息

TMODER, 2024-11-28:
查了下，看 Linux 下应该要相关驱动才能支持 USB Driver 的 S.M.A.R.T. 信息。Windows下应该都是支持的
不知道当前系统有没有解决方案！

WMZMW, 2024-12-04:
我也遇到相同问题，已经解决，方法如下：
1.去https://www.smartmontools.org/browser/trunk/smartmontools/drivedb.h下载drivedb.h
2.将其复制到/etc/，并改名为smart_drivedb.h
3.终端执行smartd -B +/etc/smart_drivedb.h
4.重启iStoreOS，此时进入S.M.A.R.T.界面并稍等片刻，应该可以识别到USB硬盘盒中的硬盘

原因：smartd内置的数据库没有记录你的硬盘盒主控芯片型号，因此无法得知设备类型，导致无法识别，此时若想成功识别，需要加上参数“-d sat”手动指定设备类型为sat，但显然webui没有做到这一点。而由于drivedb.h包含了硬盘盒主控芯片型号，并且记录了其设备类型为sat，因此通过上述步骤导入后，smartd可以正常识别，无需手动指定设备类型

若导入数据库后仍然无法识别，可能说明数据库中仍然不包含你的硬盘盒主控，，可在终端执行smartctl --all /dev/sda（根据实际情况填写）查看报错信息，获得主控芯片的VID与PID，并尝试smartctl --all /dev/sda -d sat（若失败可考虑将sat更改为其它类型，详见smartctl --help）。若成功识别，则可在smart_drivedb.h中添加相关条目（填写时可参照开头部分注释与已有条目）

WMZMW, 2024-12-04:
我也遇到相同问题，已经解决，方法如下：
1.去https://www.smartmontools.org/browser/trunk/smartmontools/drivedb.h 下载drivedb.h
2.将其复制到/etc/，并改名为smart_drivedb.h
3.终端执行smartd -B +/etc/smart_drivedb.h
4.重启iStoreOS，此时进入S.M.A.R.T.界面并稍等片刻，应该可以识别到USB硬盘盒中的硬盘

原因：smartd内置的数据库没有记录你的硬盘盒主控芯片型号，因此无法得知设备类型，导致无法识别，此时若想成功识别，需要加上参数“-d sat”手动指定设备类型为sat，但显然webui没有做到这一点。而由于drivedb.h包含了硬盘盒主控芯片型号，并且记录了其设备类型为sat，因此通过上述步骤导入后，smartd可以正常识别，无需手动指定设备类型

若导入数据库后仍然无法识别，可能说明数据库中仍然不包含你的硬盘盒主控，，可在终端执行smartctl --all /dev/sda（根据实际情况填写）查看报错信息，获得主控芯片的VID与PID，并尝试smartctl --all /dev/sda -d sat（若失败可考虑将sat更改为其它类型，详见smartctl --help）。若成功识别，则可在smart_drivedb.h中添加相关条目（填写时可参照开头部分注释与已有条目）

TMODER, 2024-12-09:
> 我也遇到相同问题，已经解决，方法如下： 1.去https://www.smartmontools.org/browser/trunk/smartmontools/drivedb.h 下载drivedb.h 2.将其复制到/etc/，并改名为smart_drivedb.h 3.终端执行smartd -B +/etc/smart_drivedb.h [...]

大佬，执行 smartctl -B +/etc/smart_drivedb.h 报错了，文件确定上传上去了

smartctl 7.2 2020-12-30 r5155 [aarch64-linux-5.10.201] (localbuild)
Copyright (C) 2002-20, Bruce Allen, Christian Franke, www.smartmontools.org

ERROR: smartctl requires a device name as the final command-line argument.

Use smartctl -h to get a usage summary

是这个命令还有其他参数？

TMODER, 2024-12-09:
> 大佬，执行 smartctl -B +/etc/smart_drivedb.h 报错了，文件确定上传上去了
>
> smartctl 7.2 2020-12-30 r5155 [aarch64-linux-5.10.201] (localbuild) Copyright (C) 2002-20, Bruce Allen, Christian Franke, [...]

执行虽然报错了，但 在管理页面 看到了  硬盘信息！
牛批！！

youyou015, 2025-01-03:
> 执行虽然报错了，但 在管理页面 看到了 硬盘信息！ 牛批！！

命令用错了，是`smartd -B ...` 不是 `smartctl -B ...`。当然了其实不需要执行这个命令就可以生效了，查看smartd help文档就可以知道smart tool默认读取的drivedb就是/etc/smart_drivedb.h，所以只需要第一和第二步就可以了，这也是为啥执行错误也能生效的原因。

TMODER, 2025-01-07:
> smartd

closed 2025-01-07 by TMODER

TMODER, 2025-01-07:
是的！ 现在学到了