istoreos/istoreos #1953: J4125无规律异常断网

JiangJamm, 2024-12-01:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题
J4125(x86)主机，在非特定场景下突然掉线断网，可以进入iStoreOS管理后台，但主页不显示联网状态，经重启可恢复。
经查看系统日志，似乎在DHCP阶段出现`Out of memory`问题导致。
[image]

### (2) 路由器型号和固件版本
- 路由器配置：J4125(x86)+i226*4
- 固件版本：iStoreOS 22.03.7 2024102509
- 内核版本：5.10.221
- 网络架构：软路由作为主路由拨号+DHCP，小米路由器桥接实现无线wifi

### (3) 详细日志和/或截图
[系统log20241201.log](https://github.com/user-attachments/files/17968793/log20241201.log)

[内核log20241201.log](https://github.com/user-attachments/files/17968794/log20241201.log)

JiangJamm, 2024-12-16:
目前把WAN6的接口删掉了，并采用了@Suct41 的[方法](https://github.com/istoreos/istoreos/issues/1953#issuecomment-2517470779)，OOM问题解决了，但是仍然有无规律断网发生，报错提示**Network device 'eth0' link is down**
[image]

[image]

[系统log20241216.log](https://github.com/user-attachments/files/18148137/log20241216.log)

Suct41, 2024-12-18:
LCP terminated by peer ---> 这个才是运营商的问题，比如心跳检测
Terminating on signal 15 ---> 用户主动断开链接（比如你点了断开）