istoreos/istoreos #1955: N1的iStoreOS如何写进emmc不执行恢复出厂？

jswcph, 2024-12-02:
iStoreOS写进U盘  
N1执行U盘启动可以正常配置安装插件在U盘里的iStoreOS系统，配置好插件以后执行写入emmc操作以后
发现，U盘里所有后来安装的插件一个也不在了，也就是说不管后来安装了什么，还是恢复到了刚开始写盘的时候。

求一个方法，就是N1写入emmc保留U盘上面配置好的插件。

求大神指点

jjm2473, 2024-12-16:
从u盘启动，然后在系统-升级/备份里面备份一下，然后在emmc启动以后导入备份文件就行了。这样可以恢复配置文件，插件还要自己安装。

jswcph, 2024-12-17:
> 从u盘启动，然后在系统-升级/备份里面备份一下，然后在emmc启动以后导入备份文件就行了。这样可以恢复配置文件，插件还要自己安装。

我不需要你这样的操作，用restore.run命令就行了    把原系统的插件配置直接恢复   

我需要的是直接写入emmc就啥也不用动了   什么都写入了   要不然还需要二次开机进入恢复插件和配置之类的    你要知道  大批量操作很浪费时间   全部都在做重复性工作

jjm2473, 2024-12-18:
那你把一台机器的emmc dump出来，然后再写入另一个机器的emmc不就行了。或者你改下install-to-emmc.sh，将restore.run执行下。

jswcph, 2024-12-18:
> 那你把一台机器的emmc dump出来，然后再写入另一个机器的emmc不就行了。或者你改下install-to-emmc.sh，将restore.run执行下。

这个操作我的水平还无法理解，大神，我可以提供两台机器  您进行远程操作  如何    一台正常啥都能刷  一台刷不了istoreos

另外如何更改install-to-emmc.sh   我还有一个问题就是N1U盘里面配置好插件的固件  执行命令写入emmc后  会还原出厂    如何更改不让它还原出厂   保修插件配置

jjm2473, 2024-12-18:
我这只有一台测试机，出问题都是刷老版本官方固件（甚至有时要先刷T1固件，反正很麻烦），然后再刷istoreos。

jjm2473, 2024-12-18:
如果安装到emmc会挂掉，那么在这台机器执行 install-to-emmc.sh 的时候，把执行过程的日志发出来看看。

jjm2473, 2024-12-18:
有ttl线吗？刷istoreos启动不了的机器，把ttl的输出发出来看看。

jswcph, 2024-12-21:
> 有ttl线吗？刷istoreos启动不了的机器，把ttl的输出发出来看看。

没有啊  哥，写入一切都正常  跟能成功的机器没什么两样，就是写入完成以后  启动不了

jswcph, 2024-12-21:
> 有ttl线吗？刷istoreos启动不了的机器，把ttl的输出发出来看看。

这两个问题你是不是回复混了 哥

写入emmc都跟能正常启动的N1一模一样  就是写完了启动后没有任何反应  成砖

jjm2473, 2024-12-23:
> 这两个问题你是不是回复混了 哥

你不是说有些机器启动不了吗

> 写入emmc都跟能正常启动的N1一模一样 就是写完了启动后没有任何反应 成砖

屏幕没有任何反应，也应该有ttl输出的，又不是烧坏了。看看ttl输出什么