istoreos/istoreos #1964: n150设备istoreos下安装jellyfin硬解报错

Herryhan15, 2024-12-08:
硬件如下
[image]

在系统杂项-驱动里面的那个核显驱动里面打钩重启或者不打勾重启都无法硬解
[image]

之后我联系了jellyfin的官方开发者NyanMisaka
聊天截图如下
[image]
[image]
[image]
附上NyanMisaka要我转达的连接https://www.bilibili.com/opus/1008609478196592663
（上传了好多图片不知道成功没有）

jjm2473, 2024-12-16:
好，我打个补丁编译个模块给你试试

jjm2473, 2024-12-16:
下载解压这个zip包，里面有ipk文件，然后在istore的手动安装那边上传安装，安装完重启下系统看看： 
[kmod-drm-i915-coex_5.10.221+23.6.37-3_x86_64.zip](https://github.com/user-attachments/files/18147420/kmod-drm-i915-coex_5.10.221%2B23.6.37-3_x86_64.zip)

（驱动那边要保持选择i915-oot）

jjm2473, 2024-12-16:
如果没问题，就先用着，下次发布固件时会带上这个。

Umfs, 2024-12-19:
试了下安装ipk，Jellyfin还是不能硬解。请问大佬为什么找不到/etc/modprobe.d/这个目录？

jjm2473, 2024-12-19:
安装以后重启下，然后看看有没有`/dev/dri`目录，如果没有，那么把内核日志提取出来看看

jjm2473, 2024-12-19:
> 请问大佬为什么找不到/etc/modprobe.d/这个目录？

openwrt不是用这个目录配置模块参数的。你不用管这个，NyanMisaka说的修改这个文件，前提是内核要在6.9以上

Umfs, 2024-12-19:
那我已经打了你给的补丁，emby和jellyfin还是不行，请问是因为emby和jellyfin代码还没更新吗？

jjm2473, 2024-12-19:
都说了安装以后重启下，然后看看有没有`/dev/dri`目录，如果没有，那么把内核日志提取出来看看。能不能把内核日志发出来？

Umfs, 2024-12-19:
谢谢，我的确有重启了，还是不行，才发的帖子。我后来装了RR黑裙，现在一切正常了。非常感谢你的回复。

jjm2473, 2024-12-20:
今天发的固件将支持N150/N250/N355的核显，其中N150核显（8086:46d4）已经测试过。

closed 2024-12-20 by jjm2473