istoreos/istoreos #1983: UNRAID 虚拟机直通8125B 能识别到网卡，但是网卡状态是未连接

MelanPrince, 2024-12-15:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
有8125B旧版本显示已经解决，实际测试只能识别无法实用，显示未连接。

## 2. 详细叙述
UNRAID 虚拟机直通8125B 能识别到网卡，但是网卡状态是未连接，删掉了kmod-8125 充新安装，依旧是这样。测试同样安装环境其他版本内核5.15.167的ImmortalWrt 23.05.4 网口使用正常。
内核版本6.6.64的LEDE R2024.12.13 也是实用正常的
A：

### (2) 路由器型号和固件版本

A：主板是 TANK 定制INTEL D-1581 网卡型号 8125B，主系统是UNRAID 虚拟机直通一个网卡
istoreos-22.03.7-2024120615 依旧这样，上个版本也有此问题。

### (3) 详细日志和/或截图

A：
[image]

如图
eth0是vmnet3 eth1是8125B可以识别但是显示已断开。同样环境下测试的其他版本网卡直通驱动正常。其他版本的可以只直通一个网卡正常使用，不实用虚拟网卡。只实用虚拟网卡做旁路由的话，外网速度会掉非常多千兆就只有从几十兆到二三百M 之间。但是内外速度正常。

MelanPrince, 2024-12-16:
原版OpenWrt OpenWrt 23.05.5 r24106 安装也是只能识别到网口但是无法实用，软件源里安装 kmod-r8125 | 5.15.167+9.013.02-4 即可正常。iStoreOS 里有R8125驱动但是不管用。

jjm2473, 2024-12-16:
openwrt应该已经升级到9.014了。你看看出问题的版本是不是9.014.

GhostChild, 2025-03-17:
一样的问题 主板asrock Z390I