istoreos/istoreos #1986: 关于首页网络接口状态空白，网络-网口配置丢失问题的描述

changandeyue, 2024-12-17:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [X] 没有类似的issue

我已搜索过类似的内容，在已关闭issue中发现了类似问题，但解决方法都未奏效。

## 2. 详细叙述
### (1) 具体问题

A：IstoreOS首页-网络接口状态-显示空白
B：IstoreOS网络-网口配置-选项空白

**造成问提操作(可能是由于以下操作引起并不确定，只用作问题排查)**
1.设置动态DNS时显示：
[image]
之后安装了bind-host
2.之后安装过ddns-go和lucky插件。

**尝试过的解决方法：**
[https://github.com/istoreos/istoreos/discussions/1869#discussioncomment-11590176](url)
`ip -pretty -detail -json link show`结果：
[image]

使用 系统便利工具 修复未奏效
卸载上述插件后修复依旧未奏效。

### (2) 路由器型号和固件版本

A：R5S

### (3) 详细日志和/或截图

jjm2473, 2024-12-17:
可以确认你的系统里的ip命令不是完整版，执行 `opkg update ; opkg install --force-reinstall ip-full`就可以修复了。

changandeyue, 2024-12-17:
> 可以确认你的系统里的ip命令不是完整版，执行 `opkg update ; opkg install --force-reinstall ip-full`就可以修复了。

非常感谢您，问题解决了

closed 2024-12-17 by changandeyue

changandeyue, 2024-12-18:
> 但是为什么ip-full会被删除呢，quickstart可是依赖了ip-full

我推测在-软件包-下载bind-host操作可能删除了某些依赖(或者由于版本问题，可能与busybox有关)导致。
出于彻底解决问题不留故障的考虑，我最终尝试了恢复出厂设置，但还是会出现一些问题。

1.在此之前我对IstoreOS整个系统进行了备份。在干净的系统上恢复备份后，虽然接口状态已恢复，但是-网口接口-页面会出现不能同时将两个网口设置为LAN的错误。
2.我还在-Istore-维护-本地备份-备份了之前安装的插件。在尝试恢复插件备份时，会出现-500的错误代码。提示与busybox-ipk有关。

我还保存了初始有问题版本的IstoreOS备份文件，如需要继续深入分析问题可留邮箱。本人能力有限只能提供信息和配合测试。

jjm2473, 2024-12-19:
> 我推测在-软件包-下载bind-host操作可能删除了某些依赖(或者由于版本问题，可能与busybox有关)导致

bind-host应该只是dns工具，不太可能影响ip-full。

> 我还保存了初始有问题版本的IstoreOS备份文件，如需要继续深入分析问题可留邮箱。本人能力有限只能提供信息和配合测试。

如果有保存整个系统的文件，那么把 `/usr/lib/opkg` 文件夹分享出来看看。

jjm2473, 2024-12-19:
> 1.在此之前我对IstoreOS整个系统进行了备份。在干净的系统上恢复备份后，虽然接口状态已恢复，但是-网口接口-页面会出现不能同时将两个网口设置为LAN的错误。

你确定这是istoreos的备份？而不是老版本openwrt（21.02之前）的备份？

changandeyue, 2024-12-19:
> > 我推测在-软件包-下载bind-host操作可能删除了某些依赖(或者由于版本问题，可能与busybox有关)导致
> 
> bind-host应该只是dns工具，不太可能影响ip-full。
> 
> > 我还保存了初始有问题版本的IstoreOS备份文件，如需要继续深入分析问题可留邮箱。本人能力有限只能提供信息和配合测试。
> 
> 如果有保存整个系统的文件，那么把 `/usr/lib/opkg` 文件夹分享出来看看。

 `/usr/lib/opkg` 没有找到这个路径，
[image]

/etc/config/network 
[network.txt](https://github.com/user-attachments/files/18193866/network.txt)

jjm2473, 2024-12-26:
/etc/config/network 看起来没啥问题

> 网口接口-页面会出现不能同时将两个网口设置为LAN的错误。

这个有截图吗

changandeyue, 2024-12-26:
> /etc/config/network 看起来没啥问题
> 
> > 网口接口-页面会出现不能同时将两个网口设置为LAN的错误。
> 
> 这个有截图吗

不好意思这个没有截图，问题可以描述为：
有三个网口
第一个网口设置为WAN
第二个网口为LAN，再点击第三个网口也设置为LAN，第二个设置的网口LAN的对勾就会消失
再次选中消失的第二个网口设置为LAN，第三个网口对勾消失。
只能选中一个网口作为LAN。

jjm2473, 2024-12-26:
现在还有这个问题吗？

changandeyue, 2024-12-26:
> 现在还有这个问题吗？

上次恢复出厂后，系统没有出现过任何问题。
之前就有个网络小问题：桥接拨号上网会不定时断网，重启光猫后会恢复，时间大概两到三天。
目前的解决办法时每天按时重启WAN就不会出现断网。
请教一下，和路由器wan口需要特殊设置有关吗，目前默认（除了配置的IPV6，一切正常）。

jjm2473, 2024-12-26:
断网以后，wan口点“重启”看能不能重新拨号联网，如果可以，wan口pppoe高级设置，将“LCP 响应故障阈值”和“LCP 响应间隔”都设置为5，应该就会在断网后自动重拨了。如果重拨没用的话，那就没办法了。

changandeyue, 2024-12-26:
> 断网以后，wan口点“重启”看能不能重新拨号联网，如果可以，wan口pppoe高级设置，将“LCP 响应故障阈值”和“LCP 响应间隔”都设置为5，应该就会在断网后自动重拨了。如果重拨没用的话，那就没办法了。

好的，我尝试一下

uxfion, 2025-03-07:
> 断网以后，wan口点“重启”看能不能重新拨号联网，如果可以，wan口pppoe高级设置，将“LCP 响应故障阈值”和“LCP 响应间隔”都设置为5，应该就会在断网后自动重拨了。如果重拨没用的话，那就没办法了。

我是遇到每5-7天运营商会换一次IP，即使我设置了“LCP 响应故障阈值 10”和“LCP 响应间隔 5”也不会自动重播。除非手动重启光猫。是固件的bug吗？