istoreos/istoreos #1988: 日志里面频繁报错，导致内存占用异常，路由器卡顿

smhjw, 2024-12-18:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：系统日志里频繁出现以下提示，导致内存占用很高
[13488.337723] ntfs3: sdb1: ino=9979, "log" Looks like your dir is corrupt
[13488.337725] ntfs3: sdb1: ino=9979, "log" Looks like your dir is corrupt
[13488.337727] ntfs3: sdb1: ino=9979, "log" Looks like your dir is corrupt
[13488.337730] ntfs3: sdb1: ino=9979, "log" Looks like your dir is corrupt
[13488.337732] ntfs3: sdb1: ino=9979, "log" Looks like your dir is corrupt
[13488.337735] ntfs3: sdb1: ino=9979, "log" Looks like your dir is corrupt
[13488.337737] ntfs3: sdb1: ino=9979, "log" Looks like your dir is corrupt
[13488.337740] ntfs3: sdb1: ino=9979, "log" Looks like your dir is corrupt
[13488.337742] ntfs3: sdb1: ino=9979, "log" Looks like your dir is corrupt
[13490.026464] ntfs3: sda1: failed to read volume at offset 0x2c000
[13490.049748] ntfs3: sda1: failed to read volume at offset 0x2c000
[13492.659188] ntfs3: sda1: failed to read volume at offset 0x2c000
[13493.347702] ntfs3: 7055966 callbacks suppressed
[13493.347704] ntfs3: sdb1: ino=9979, "log" Looks like your dir is corrupt
[13493.347708] ntfs3: sdb1: ino=9979, "log" Looks like your dir is corrupt
[13493.347710] ntfs3: sdb1: ino=9979, "log" Looks like your dir is corrupt
[13493.347713] ntfs3: sdb1: ino=9979, "log" Looks like your dir is corrupt
[13493.347715] ntfs3: sdb1: ino=9979, "log" Looks like your dir is corrupt
[13493.347718] ntfs3: sdb1: ino=9979, "log" Looks like your dir is corrupt
[13493.347720] ntfs3: sdb1: ino=9979, "log" Looks like your dir is corrupt
[13493.347723] ntfs3: sdb1: ino=9979, "log" Looks like your dir is corrupt
[13493.347725] ntfs3: sdb1: ino=9979, "log" Looks like your dir is corrupt
[13493.347728] ntfs3: sdb1: ino=9979, "log" Looks like your dir is corrupt
[13495.334342] ntfs3: sda1: failed to read volume at offset 0x2c000
[13495.362462] ntfs3: sda1: failed to read volume at offset 0x2c000
[13498.357686] ntfs3: 7056800 callbacks suppressed
[13498.357688] ntfs3: sdb1: ino=9979, "log" Looks like your dir is corrupt
[13498.357692] ntfs3: sdb1: ino=9979, "log" Looks like your dir is corrupt
[13498.357694] ntfs3: sdb1: ino=9979, "log" Looks like your dir is corrupt
[13498.357697] ntfs3: sdb1: ino=9979, "log" Looks like your dir is corrupt
[13498.357699] ntfs3: sdb1: ino=9979, "log" Looks like your dir is corrupt
[13498.357701] ntfs3: sdb1: ino=9979, "log" Looks like your dir is corrupt
[13498.357704] ntfs3: sdb1: ino=9979, "log" Looks like your dir is corrupt
[13498.357706] ntfs3: sdb1: ino=9979, "log" Looks like your dir is corrupt
[13498.357708] ntfs3: sdb1: ino=9979, "log" Looks like your dir is corrupt
[13498.357711] ntfs3: sdb1: ino=9979, "log" Looks like your dir is corrupt

### (2) 路由器型号和固件版本

A：设备型号  AZW EQ（领刻n100）

      固件版本   iStoreOS 22.03.7 2024120615

### (3) 详细日志和/或截图

A：Wed Dec 18 09:54:02 2024 daemon.err dockerd[16481]: time="2024-12-18T01:54:02.805461131Z" level=debug msg=event module=libcontainerd namespace=moby topic=/tasks/exec-added
Wed Dec 18 09:54:02 2024 daemon.err dockerd[16481]: time="2024-12-18T01:54:02.805787423Z" level=debug msg=event module=libcontainerd namespace=moby topic=/tasks/exec-added
Wed Dec 18 09:54:02 2024 daemon.err dockerd[16481]: time="2024-12-18T01:54:02.872366320Z" level=debug msg="event forwarded" ns=moby topic=/tasks/exec-started type=containerd.events.TaskExecStarted
Wed Dec 18 09:54:02 2024 daemon.err dockerd[16481]: time="2024-12-18T01:54:02.872554069Z" level=debug msg="event forwarded" ns=moby topic=/tasks/exec-started type=containerd.events.TaskExecStarted
Wed Dec 18 09:54:02 2024 daemon.err dockerd[16481]: time="2024-12-18T01:54:02.872572783Z" level=debug msg=event module=libcontainerd namespace=moby topic=/tasks/exec-started
Wed Dec 18 09:54:02 2024 daemon.err dockerd[16481]: time="2024-12-18T01:54:02.872718161Z" level=debug msg=event module=libcontainerd namespace=moby topic=/tasks/exec-started
Wed Dec 18 09:54:02 2024 daemon.err dockerd[16481]: time="2024-12-18T01:54:02.872923459Z" level=debug msg="event forwarded" ns=moby topic=/tasks/exit type=containerd.events.TaskExit
Wed Dec 18 09:54:02 2024 daemon.err dockerd[16481]: time="2024-12-18T01:54:02.873048550Z" level=debug msg=event module=libcontainerd namespace=moby topic=/tasks/exit
Wed Dec 18 09:54:02 2024 daemon.err dockerd[16481]: time="2024-12-18T01:54:02.873101587Z" level=debug msg="attach: stdout: end"
Wed Dec 18 09:54:02 2024 daemon.err dockerd[16481]: time="2024-12-18T01:54:02.873114738Z" level=debug msg="attach: stderr: end"
Wed Dec 18 09:54:02 2024 daemon.err dockerd[16481]: time="2024-12-18T01:54:02.873124182Z" level=debug msg="attach done"
Wed Dec 18 09:54:02 2024 daemon.err dockerd[16481]: time="2024-12-18T01:54:02.873151383Z" level=debug msg="Health check for container 6cd09cb0630edbad93317c90787c38bd439c1c883170d1e4cb4a6b0d06b82c6e done (exitCode=0)"
Wed Dec 18 09:54:02 2024 daemon.err dockerd[16481]: time="2024-12-18T01:54:02.877463361Z" level=debug msg="event forwarded" ns=moby topic=/tasks/exit type=containerd.events.TaskExit
Wed Dec 18 09:54:02 2024 daemon.err dockerd[16481]: time="2024-12-18T01:54:02.877622424Z" level=debug msg=event module=libcontainerd namespace=moby topic=/tasks/exit
Wed Dec 18 09:54:02 2024 daemon.err dockerd[16481]: time="2024-12-18T01:54:02.877685229Z" level=debug msg="attach: stdout: end"
Wed Dec 18 09:54:02 2024 daemon.err dockerd[16481]: time="2024-12-18T01:54:02.877689750Z" level=debug msg="attach: stderr: end"
Wed Dec 18 09:54:02 2024 daemon.err dockerd[16481]: time="2024-12-18T01:54:02.877715825Z" level=debug msg="attach done"
Wed Dec 18 09:54:02 2024 daemon.err dockerd[16481]: time="2024-12-18T01:54:02.877744531Z" level=debug msg="Health check for container 36cd7c39585c9401c7bae29b63c0ea11dca912b3277587c4e9bb5a27cfc7f361 done (exitCode=0)"
Wed Dec 18 09:54:08 2024 daemon.notice ddnsto[28226]: main accept error count=0 wait_cnt=0
Wed Dec 18 09:54:13 2024 daemon.notice ddnsto[16783]: check reboot failed, ret=-1
Wed Dec 18 09:54:32 2024 daemon.err dockerd[16481]: time="2024-12-18T01:54:32.874697199Z" level=debug msg="Running health check for container 6cd09cb0630edbad93317c90787c38bd439c1c883170d1e4cb4a6b0d06b82c6e ..."
Wed Dec 18 09:54:32 2024 daemon.err dockerd[16481]: time="2024-12-18T01:54:32.875003624Z" level=debug msg="starting exec command 22a0c3d00bae4c24f87c7ca73dd19c71cce036e294848699c43cef8fddb4c3f7 in container 6cd09cb0630edbad93317c90787c38bd439c1c883170d1e4cb4a6b0d06b82c6e"
Wed Dec 18 09:54:32 2024 daemon.err dockerd[16481]: time="2024-12-18T01:54:32.877481497Z" level=debug msg="attach: stdout: begin"
Wed Dec 18 09:54:32 2024 daemon.err dockerd[16481]: time="2024-12-18T01:54:32.877553577Z" level=debug msg="attach: stderr: begin"

jjm2473, 2024-12-22:
内存占用跟ntfs估计没啥关系。报错可能只是因为你的硬盘坏了，建议关机拔掉硬盘再启动，看看还有没有问题

smhjw, 2024-12-24:
> 内存占用跟ntfs估计没啥关系。报错可能只是因为你的硬盘坏了，建议关机拔掉硬盘再启动，看看还有没有问题

ok,重新插拔硬盘以后好了