istoreos/istoreos #1990: Asia/Shanghai 时间并不是东八区时间。

wcbspp, 2024-12-19:
目前系统里设置的时间Asia/Shanghai并没有正确显示东八区时间，好像opwenwrt老问题了吧？
[image]
[image]

wcbspp, 2024-12-20:
opkg install zoneinfo-asia 重启之后也没有效果 好像之前版本没有这问题

wcbspp, 2024-12-20:
[image]

jjm2473, 2024-12-22:
点一下“与NTP服务器同步”试试

wcbspp, 2024-12-22:
> 点一下“与NTP服务器同步”试试

[image]
没什么效果

wcbspp, 2024-12-22:
> 点一下“与NTP服务器同步”试试

[image]
没效果~

bcseputetto, 2024-12-22:
@wcbspp 修改系统，时间同步，里面的候选ntp服务器
`ntp.aliyun.com`
`ntp1.aliyun.com`
`ntp2.aliyun.com`
`ntp3.aliyun.com`
`ntp4.aliyun.com`
`ntp5.aliyun.com`
然后再试试

wcbspp, 2024-12-22:
> ntp1.aliyun.com
[image]
[image]
换了阿里的也不行呢.....

bcseputetto, 2024-12-22:
@wcbspp 终端里面执行
```console
date
```
截图看下

wcbspp, 2024-12-22:
> @wcbspp 终端里面执行
> 
> ```
> date
> ```
> 
> 截图看下

[image]
看起来走的是utc时间，设置里换成etc/GMT-8可以正常显示时间，但是各种插件的日志时间会有问题。

bcseputetto, 2024-12-22:
@wcbspp 执行下面命令，截图执行结果
```
ls -l /etc
ls /usr/share/zoneinfo/Asia/
cat /etc/localtime
```

wcbspp, 2024-12-22:
> @wcbspp 执行下面命令，截图执行结果
> 
> ```
> cat /etc/config/system
> ls -l /etc
> ls /usr/share/zoneinfo/Asia/
> cat /etc/localtime
> ```

[image]
[image]
[image]
/etc/localtime这个不存在

bcseputetto, 2024-12-22:
@wcbspp 执行
```
ln -sf /usr/share/zoneinfo/Asia/Shanghai /etc/localtime
```
然后重启一下系统，再看下某些插件的时间

bcseputetto, 2024-12-22:
你这是重启后时间又变回去了？

bcseputetto, 2024-12-22:
@wcbspp 执行
```
uci set system.@system[0].zonename='Asia/Shanghai'
uci set system.@system[0].timezone='CST-8'
uci commit system
ln -sf /usr/share/zoneinfo/Asia/Shanghai /etc/localtime
/etc/init.d/system reload
```
执行完这些不重启，看一下系统时间和插件时间

wcbspp, 2024-12-22:
> @wcbspp 执行
> 
> ```
> uci set system.@system[0].zonename='Asia/Shanghai'
> uci set system.@system[0].timezone='CST-8'
> uci commit system
> ln -sf /usr/share/zoneinfo/Asia/Shanghai /etc/localtime
> /etc/init.d/system reload
> ```
> 
> 执行完这些不重启，看一下系统时间和插件时间
额
[image]

bcseputetto, 2024-12-22:
@wcbspp 你有升级过软件包吗？去用系统便利工具先修复一下系统软件看看

wcbspp, 2024-12-22:
> @wcbspp 你有升级过软件包吗？去用系统便利工具先修复一下系统软件看看

系统是上周重新装的最新版本，刚修复了一下，重启后还是没有变化

138161, 2024-12-22:
我装了chrony
#增加的源
pool 2.centos.pool.ntp.org iburst
pool ntp.aliyun.com iburst
pool 0.cn.pool.ntp.org iburst
pool 1.cn.pool.ntp.org iburst
pool 2.cn.pool.ntp.org iburst
pool 3.cn.pool.ntp.org iburst
pool 0.europe.pool.ntp.org iburst
再也没有烦恼过

wcbspp, 2024-12-24:
> ntp.aliyun.com
应该不是一个问题
我这里时间服务器本身没问题 时区不正常

wcbspp, 2024-12-24:
> @wcbspp 执行下面命令，截图执行结果
> 
> ```
> cat /etc/config/system
> ls -l /etc
> ls /usr/share/zoneinfo/Asia/
> cat /etc/localtime
> ```

问题已经解决。
/etc/localtime文件问题  rm后重新关联 ln -sf /usr/share/zoneinfo/Asia/Shanghai即可。
也非常感谢您了。

closed 2024-12-24 by wcbspp