istoreos/istoreos #1999: 开启ipv6后，浮动网关功能存在问题

KevinZjYang, 2024-12-22:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：前提：
1、主路由和旁路由均开启IPv6。旁路由设置科学上网。
2、主路由IP:192.168.31.1 
      旁路由IP:192.168.31.3 
      浮动网关IP:192.168.31.6
3、主路由DHCP选项设置 3,192.168.31.6  6,192.168.31.6
出现的问题：
这样设备获取到的IPv4网关就是192.168.31.6，但是因为开启了IPv6，设备访问网站的时候，比如youtube，会获取到IPv6的地址。
这时候问题出现了，设置只有v4的网关指向浮动网关（这时候也就是旁路由）。v6的流量并没有通过旁路由，网站就会无法访问。
请问，有没有什么办法可以解决这个问题，毕竟IPv6打开对于有些情况还是很有用的。

### (2) 路由器型号和固件版本

A：R1+R68S 均为最新版本

### (3) 详细日志和/或截图

A：

138161, 2024-12-22:
passwall里过滤了ipv6吗

KevinZjYang, 2024-12-22:
> passwall里过滤了ipv6吗

我passwall2里面开了ipv6劫持。但这不是重点，重点是v6的流量不经过旁路由，设备获取的v6网关是主路由的。

138161, 2024-12-22:
重点是v6的流量不经过旁路由，设备获取的v6网关是主路由的。
对啊，ipv6分配的PD前缀是64位的吧（指第一个路由器），那就是直接到设备的，不会绕到旁路由，只有下面两种情况可以绕
一，小于64位的PD前缀，这样，第一个路由器才能为旁路由或者二级路由再分配ipv6的二级子网，并管理ipv6的上下级通讯
二、如果是64位的PD前缀，那意味着第一个路由器下面接的都是终端，二级路由器也被认为是ipv6终端，二级路由器下的设备是无法使用ipv6功能的，这时候的旁路由需要有此功能：将其他终端ipv4到旁路由，旁路由转换该ipv4到ipv6，再由旁路由发出和接收ipv6，旁路由管理其下的ipv6通讯。
总之，ipv6与ipv4不同，ipv4中间有路由器，可以由路由器管理转发ipv4内容，但是，ipv6就跟其得到的PD前缀有很大的关系，小于64位PD前缀，中间的路由器就能实现ipv6的再分配和管理，类似ipv4，但64位的前缀就不行，中间的二级路由器无法管理ipv6的通讯，只有将第一个路由器的ipv6改成NAT模式，就是由第一个路由器强行管理ipv6，效果是电信给的ipv6只到第一个路由器的wan侧，该路由器自己设定ipv6内网，fd00开头的内网，其下的设备都是fd00开头的内网ipv6地址，由第一个路由器将下面的fd00内网改为240e开头的外网地址发出去，等于硬改地址功能。这样才有可能实现旁路由转发ipv6流量

138161, 2024-12-22:
为何64位的前缀不行？因为ipv6协议规定，凡是获得的是64位前缀，就代表该设备是终端，哪怕路由器也被认为终端，终端下面再接的设备是无法使用ipv6，因为64位前缀是能分配的最小地址，类似于192.168.1.1，不会产生192.168.1.1.xxx的

KevinZjYang, 2024-12-23:
> 重点是v6的流量不经过旁路由，设备获取的v6网关是主路由的。 对啊，ipv6分配的PD前缀是64位的吧（指第一个路由器），那就是直接到设备的，不会绕到旁路由，只有下面两种情况可以绕 一，小于64位的PD前缀，这样，第一个路由器才能为旁路由或者二级路由再分配ipv6的二级子网，并管理ipv6的上下级通讯 [...]

谢谢答疑，受益匪浅。

138161, 2024-12-23:
IPv6-PD: 240e:348:XXX:XXXX::/60
第一个路由器的wan6侧，设定中，PD长度改60，不要用64

jjm2473, 2024-12-23:
跟浮动网关无关。不管什么情况下，都不应该使用ipv6访问youtube对吧？参考这个教程，在旁路由配置一个纯ipv4的DNS，然后让科学插件对于国外域名使用这个DNS解析 https://github.com/istoreos/istoreos/discussions/1875

KevinZjYang, 2024-12-23:
> 跟浮动网关无关。不管什么情况下，都不应该使用ipv6访问youtube对吧？参考这个教程，在旁路由配置一个纯ipv4的DNS，然后让科学插件对于国外域名使用这个DNS解析 #1875

好办法，我去试试，谢谢。

jjm2473, 2024-12-23:
@KevinZjYang DNS服务弄好以后。如果是passwall，取消勾选“过滤代理域名 IPv6”，并将“远程DNS”修改成“127.0.0.53”就能实现代理域名只解析IPv4了（“过滤模式”必须是“通过 UDP 请求 DNS”）。如果是其他插件，那自己研究下

KevinZjYang, 2024-12-23:
> @KevinZjYang DNS服务弄好以后。如果是passwall，取消勾选“过滤代理域名 IPv6”，并将“远程DNS”修改成“127.0.0.53”就能实现代理域名只解析IPv4了（“过滤模式”必须是“通过 UDP 请求 DNS”）。如果是其他插件，那自己研究下

好的，谢谢

KevinZjYang, 2024-12-23:
> 跟浮动网关无关。不管什么情况下，都不应该使用ipv6访问youtube对吧？参考这个教程，在旁路由配置一个纯ipv4的DNS，然后让科学插件对于国外域名使用这个DNS解析 #1875

大佬，我又发现一个问题，电脑会有两个dns服务器，ipv4和ipv6各一个，并且一般都是ipv6优先。ipv4的是旁路由的，ipv6的是主路由的。两个都会返回dns解析结果，那这样还是会存在返回ipv6的情况。这样就又绕开旁路由了。
电脑可以修改成ipv4优先，那其他不方便修改的设备岂不是还是不行。
也不知道我理解的对不对~~~

KevinZjYang, 2024-12-23:
又折腾一晚上我放弃了。只要主路由能解析v6地址，这个问题就没办法。

jjm2473, 2024-12-26:
不要使用ipv6的dns就行了，反正ipv4的dns同样能解析ipv6地址。主路由是openwrt或者istoreos的话，网络-接口，wan6-高级设置里面，把“自动获取 DNS 服务器”取消掉，就没问题了。

jjm2473, 2024-12-26:
还有一种方式，就是防火墙把ipv6的53端口转发和入站都reject掉，客户端也就只能使用ipv4的dns了，也就是旁路由的dns服务

KevinZjYang, 2024-12-26:
成功了！！！感谢大佬 @jjm2473 @138161 的耐心解释。
简单说一下我实现的效果，全屋所有设备均能正常使用ipv6，需要代理的流量也会自动走旁路由。
原理：主路由和旁路由都安装smartdns，通过smartdns的域名规则列表实现dns解析分流，需要代理的域名在“域名规则列表”里面“停用IPV6地址解析”（两个路由器都需要这样做）。这样就实现了所有设备访问代理列表里面的域名时，无论使用v4的dns服务器，还是v6的dns服务器，都只能解析到ipv4的地址，除此之外其他都是正常解析v6。smartdns的教程我就不贴了，下面时关键的设置截图。**除了smartdns设置外其他地方不需要任何设置。**
希望对后面朋友遇到类似问题有帮助。
### **主路由：**
[image]
[image]
[image]
[image]
**这里gfw和proxy-list都需要设置，下图只是举例。**
[image]

里面有个小问题，主路由没有passwall，使用DNS服务器`https://dns.google/dns-query`可能解析不到，这时候需要设置smartdns的代理服务器，然后在设置的国外dns服务器都勾选“使用代理”。（代理地址就是旁路由passwall2里面服务器端设置）
[image]

### **旁路由**设置基本一致，除了不需要设置代理。

### 域名列表

**@felixonmars/dnsmasq-china-list 仓库收集的在中国大陆可直连的域名列表**
```
https://fastly.jsdelivr.net/https://raw.githubusercontent.com/Loyalsoldier/v2ray-rules-dat/release/china-list.txt
```
**直连列表**
```
https://fastly.jsdelivr.net/https://raw.githubusercontent.com/Loyalsoldier/v2ray-rules-dat/release/direct-list.txt
```
**苹果中国直连列表**
```
https://fastly.jsdelivr.net/https://raw.githubusercontent.com/Loyalsoldier/v2ray-rules-dat/release/apple-cn.txt
```
**谷歌中国直连列表**
```
https://fastly.jsdelivr.net/https://raw.githubusercontent.com/Loyalsoldier/v2ray-rules-dat/release/google-cn.txt
```
**代理列表**
```
https://fastly.jsdelivr.net/https://raw.githubusercontent.com/Loyalsoldier/v2ray-rules-dat/release/proxy-list.txt
```
**GFWList 列表**
```
https://fastly.jsdelivr.net/https://raw.githubusercontent.com/Loyalsoldier/v2ray-rules-dat/release/gfw.txt
```

jjm2473, 2024-12-27:
@KevinZjYang 用我说的纯ipv4 dns https://github.com/istoreos/istoreos/discussions/1875 ，根本就不需要使用smartdns来分流。
这是我的passwall的DNS配置
[image]