istoreos/istoreos #2000: 更新系统便利工具，不能打开luci登录界面了

ShumT27, 2024-12-22:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：没记错应该是在iStore商店更新系统便利工具后，就打不开luci登录界面了，但是系统还是正常在运行的。

### (2) 路由器型号和固件版本

A：R4s

### (3) 详细日志和/或截图

A：/usr/lib/lua/luci/controller/bypass.lua:5: module 'luci.model.cbi.bypass.trojan_go' not found:
	no field package.preload['luci.model.cbi.bypass.trojan_go']
	no file './luci/model/cbi/bypass/trojan_go.lua'
	no file '/usr/share/lua/luci/model/cbi/bypass/trojan_go.lua'
	no file '/usr/share/lua/luci/model/cbi/bypass/trojan_go/init.lua'
	no file '/usr/lib/lua/luci/model/cbi/bypass/trojan_go.lua'
	no file '/usr/lib/lua/luci/model/cbi/bypass/trojan_go/init.lua'
	no file './luci/model/cbi/bypass/trojan_go.so'
	no file '/usr/lib/lua/luci/model/cbi/bypass/trojan_go.so'
	no file '/usr/lib/lua/loadall.so'
	no file './luci.so'
	no file '/usr/lib/lua/luci.so'
	no file '/usr/lib/lua/loadall.so'
stack traceback:
	[C]: in function 'require'
	/usr/lib/lua/luci/controller/bypass.lua:5: in main chunk
	[C]: in function 'require'
	/usr/lib/lua/luci/dispatcher.lua:1135: in function 'createindex'
	/usr/lib/lua/luci/dispatcher.lua:1236: in function 'createtree'
	/usr/lib/lua/luci/dispatcher.lua:638: in function 'menu_json'
	/usr/lib/lua/luci/dispatcher.lua:887: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:482: in function </usr/lib/lua/luci/dispatcher.lua:481>

jjm2473, 2024-12-23:
看起来是你的bypass插件有问题

ShumT27, 2024-12-25:
> 看起来是你的bypass插件有问题

好的，谢谢老哥，我尝试用ssh卸载插件看看😆