istoreos/istoreos #2006: 最新固件的istore os 死活连不上网

WeskerC, 2024-12-26:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题
A：
更新了最新的固件istoreos-22.03.7-2024122015-x86-64-squashfs-combined-efi.img.gz后，istore os 怎么都连不上网。
现象：
1. 主页显示未联网， 固件更新一直转圈
2. 不管是中继模式还是旁路模式，都不行网线直连的机子网络会显示未识别的网络，
3. 已排除了主网络的问题，ping都能通

### (2) 路由器型号和固件版本
istoreos-22.03.7-2024122015-x86-64
A：

### (3) 详细日志和/或截图

A：
[image]
[image]
[image]

curl: (28) Failed to connect to istore.istoreos.com port 443 after 10002 ms: Timeout was reached
curl: (28) Failed to connect to mirrors.cernet.edu.cn port 443 after 10002 ms: Timeout was reached
curl: (28) Failed to connect to mirrors.cernet.edu.cn port 443 after 10002 ms: Timeout was reached
curl: (28) Failed to connect to mirrors.cernet.edu.cn port 443 after 10002 ms: Timeout was reached
curl: (28) Failed to connect to mirrors.cernet.edu.cn port 443 after 10002 ms: Timeout was reached
curl: (28) Failed to connect to mirrors.cernet.edu.cn port 443 after 10002 ms: Timeout was reached
curl: (28) Failed to connect to mirrors.cernet.edu.cn port 443 after 10002 ms: Timeout was reached
Collected errors:
 * opkg_download: Failed to download https://istore.istoreos.com/repo/all/compat/Packages.gz, curl returned 28.
 * opkg_download: Check your network settings and connectivity.

 * opkg_download: Failed to download https://mirrors.cernet.edu.cn/openwrt/releases/22.03.7/targets/x86/64/packages/Packages.gz, curl returned 28.
 * opkg_download: Check your network settings and connectivity.

 * opkg_download: Failed to download https://mirrors.cernet.edu.cn/openwrt/releases/22.03.7/packages/x86_64/base/Packages.gz, curl returned 28.
 * opkg_download: Check your network settings and connectivity.

 * opkg_download: Failed to download https://mirrors.cernet.edu.cn/openwrt/releases/22.03.7/packages/x86_64/luci/Packages.gz, curl returned 28.
 * opkg_download: Check your network settings and connectivity.

 * opkg_download: Failed to download https://mirrors.cernet.edu.cn/openwrt/releases/22.03.7/packages/x86_64/packages/Packages.gz, curl returned 28.
 * opkg_download: Check your network settings and connectivity.

 * opkg_download: Failed to download https://mirrors.cernet.edu.cn/openwrt/releases/22.03.7/packages/x86_64/routing/Packages.gz, curl returned 28.
 * opkg_download: Check your network settings and connectivity.

 * opkg_download: Failed to download https://mirrors.cernet.edu.cn/openwrt/releases/22.03.7/packages/x86_64/telephony/Packages.gz, curl returned 28.
 * opkg_download: Check your network settings and connectivity.

jjm2473, 2024-12-27:
能ping通但是不能下载文件一般都是上游防火墙拦截了端口，要么就是翻墙插件配置错误。

WeskerC, 2024-12-31:
问题已解决，是pass插件未完全关闭导致的

Kindly Regards,

closed 2025-01-01 by jjm2473