istoreos/istoreos #2018: istore无法安装插件，手动安装也不行了。

cysfireman, 2024-12-29:
istore无法安装插件，手动安装也不行了。

这是软件商店安装的提示：
Fetch feed list for x86_64
Update feeds index
Downloading https://istore.istoreos.com/repo/all/nas_luci/Packages.gz
Updated list of available packages in /tmp/is-root//tmp/opkg-lists/is_nas_luci
Downloading https://istore.istoreos.com/repo/all/store/Packages.gz
Updated list of available packages in /tmp/is-root//tmp/opkg-lists/is_store
Downloading https://istore.istoreos.com/repo/x86_64/nas/Packages.gz
Updated list of available packages in /tmp/is-root//tmp/opkg-lists/is_nas
Downloading https://istore.istoreos.com/repo/all/compat/Packages.gz
Updated list of available packages in /tmp/is-root//tmp/opkg-lists/istore_compat
Downloading https://mirrors.cernet.edu.cn/openwrt/releases/22.03.7/targets/x86/64/packages/Packages.gz
curl: (22) The requested URL returned error: 403
*** Failed to download the package list from https://mirrors.cernet.edu.cn/openwrt/releases/22.03.7/targets/x86/64/packages/Packages.gz

Downloading https://mirrors.cernet.edu.cn/openwrt/releases/22.03.7/packages/x86_64/base/Packages.gz
curl: (22) The requested URL returned error: 403
*** Failed to download the package list from https://mirrors.cernet.edu.cn/openwrt/releases/22.03.7/packages/x86_64/base/Packages.gz

Downloading https://mirrors.cernet.edu.cn/openwrt/releases/22.03.7/packages/x86_64/luci/Packages.gz
curl: (22) The requested URL returned error: 403
*** Failed to download the package list from https://mirrors.cernet.edu.cn/openwrt/releases/22.03.7/packages/x86_64/luci/Packages.gz

Downloading https://mirrors.cernet.edu.cn/openwrt/releases/22.03.7/packages/x86_64/packages/Packages.gz
curl: (22) The requested URL returned error: 403
*** Failed to download the package list from https://mirrors.cernet.edu.cn/openwrt/releases/22.03.7/packages/x86_64/packages/Packages.gz

Downloading https://mirrors.cernet.edu.cn/openwrt/releases/22.03.7/packages/x86_64/routing/Packages.gz
curl: (22) The requested URL returned error: 403
*** Failed to download the package list from https://mirrors.cernet.edu.cn/openwrt/releases/22.03.7/packages/x86_64/routing/Packages.gz

Downloading https://mirrors.cernet.edu.cn/openwrt/releases/22.03.7/packages/x86_64/telephony/Packages.gz
curl: (22) The requested URL returned error: 403
*** Failed to download the package list from https://mirrors.cernet.edu.cn/openwrt/releases/22.03.7/packages/x86_64/telephony/Packages.gz

Downloading https://istore.istoreos.com/repo/all/meta/Packages.gz
Updated list of available packages in /tmp/is-root//tmp/opkg-lists/is_meta
Collected errors:
 * opkg_download: Failed to download https://mirrors.cernet.edu.cn/openwrt/releases/22.03.7/targets/x86/64/packages/Packages.gz, curl returned 22.
 * opkg_download: Failed to download https://mirrors.cernet.edu.cn/openwrt/releases/22.03.7/packages/x86_64/base/Packages.gz, curl returned 22.
 * opkg_download: Failed to download https://mirrors.cernet.edu.cn/openwrt/releases/22.03.7/packages/x86_64/luci/Packages.gz, curl returned 22.
 * opkg_download: Failed to download https://mirrors.cernet.edu.cn/openwrt/releases/22.03.7/packages/x86_64/packages/Packages.gz, curl returned 22.
 * opkg_download: Failed to download https://mirrors.cernet.edu.cn/openwrt/releases/22.03.7/packages/x86_64/routing/Packages.gz, curl returned 22.
 * opkg_download: Failed to download https://mirrors.cernet.edu.cn/openwrt/releases/22.03.7/packages/x86_64/telephony/Packages.gz, curl returned 22.
Try mirror server https://repo.istoreos.com/repo
Unknown package 'app-meta-nlbwmon'.
Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency nlbwmon for app-meta-nlbwmon
 * pkg_hash_check_unresolved: cannot find dependency luci-i18n-nlbwmon-zh-cn for app-meta-nlbwmon
 * pkg_hash_check_unresolved: cannot find dependency luci-app-nlbwmon for app-meta-nlbwmon
 * pkg_hash_fetch_best_installation_candidate: Packages for app-meta-nlbwmon found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package app-meta-nlbwmon.
Try origin server https://istore.istoreos.com/repo
Unknown package 'app-meta-nlbwmon'.
Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency nlbwmon for app-meta-nlbwmon
 * pkg_hash_check_unresolved: cannot find dependency luci-i18n-nlbwmon-zh-cn for app-meta-nlbwmon
 * pkg_hash_check_unresolved: cannot find dependency luci-app-nlbwmon for app-meta-nlbwmon
 * pkg_hash_fetch_best_installation_candidate: Packages for app-meta-nlbwmon found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package app-meta-nlbwmon.
Fetch feed list for x86_64
Update feeds index
Downloading https://istore.istoreos.com/repo/all/nas_luci/Packages.gz
Updated list of available packages in /tmp/is-root//tmp/opkg-lists/is_nas_luci
Downloading https://istore.istoreos.com/repo/all/store/Packages.gz
Updated list of available packages in /tmp/is-root//tmp/opkg-lists/is_store
Downloading https://istore.istoreos.com/repo/x86_64/nas/Packages.gz
Updated list of available packages in /tmp/is-root//tmp/opkg-lists/is_nas
[... 58 lines cut ...]

Downloading https://mirrors.cernet.edu.cn/openwrt/releases/22.03.7/packages/x86_64/luci/Packages.gz
curl: (22) The requested URL returned error: 403
*** Failed to download the package list from https://mirrors.cernet.edu.cn/openwrt/releases/22.03.7/packages/x86_64/luci/Packages.gz

Downloading https://mirrors.cernet.edu.cn/openwrt/releases/22.03.7/packages/x86_64/packages/Packages.gz
curl: (22) The requested URL returned error: 403
*** Failed to download the package list from https://mirrors.cernet.edu.cn/openwrt/releases/22.03.7/packages/x86_64/packages/Packages.gz

Downloading https://mirrors.cernet.edu.cn/openwrt/releases/22.03.7/packages/x86_64/routing/Packages.gz
curl: (22) The requested URL returned error: 403
*** Failed to download the package list from https://mirrors.cernet.edu.cn/openwrt/releases/22.03.7/packages/x86_64/routing/Packages.gz

Downloading https://mirrors.cernet.edu.cn/openwrt/releases/22.03.7/packages/x86_64/telephony/Packages.gz
curl: (22) The requested URL returned error: 403
*** Failed to download the package list from https://mirrors.cernet.edu.cn/openwrt/releases/22.03.7/packages/x86_64/telephony/Packages.gz

Collected errors:
 * opkg_download: Failed to download https://mirrors.cernet.edu.cn/openwrt/releases/22.03.7/targets/x86/64/packages/Packages.gz, curl returned 22.
 * opkg_download: Failed to download https://mirrors.cernet.edu.cn/openwrt/releases/22.03.7/packages/x86_64/base/Packages.gz, curl returned 22.
 * opkg_download: Failed to download https://mirrors.cernet.edu.cn/openwrt/releases/22.03.7/packages/x86_64/luci/Packages.gz, curl returned 22.
 * opkg_download: Failed to download https://mirrors.cernet.edu.cn/openwrt/releases/22.03.7/packages/x86_64/packages/Packages.gz, curl returned 22.
 * opkg_download: Failed to download https://mirrors.cernet.edu.cn/openwrt/releases/22.03.7/packages/x86_64/routing/Packages.gz, curl returned 22.
 * opkg_download: Failed to download https://mirrors.cernet.edu.cn/openwrt/releases/22.03.7/packages/x86_64/telephony/Packages.gz, curl returned 22.
更新软件源列表错误，请检查路由器自身网络连接以及是否有失效的软件源。

jjm2473, 2024-12-30:
换个软件源，例如其他大学的，或者腾讯阿里云都行

jjm2473, 2025-01-08:
这跟istoreos无关，是国内的软件源mirror自己抽风，使用openwrt官方源的话，那国内用户更加用不了。

jjm2473, 2025-01-08:
这是其中一个乱封UA的案例 https://github.com/SUSTech-CRA/issues/issues/23

closed 2025-01-08 by jjm2473