istoreos/istoreos #2031: n1固件的系统分区大小能否调整一下

its-lis, 2025-01-03:
现在n1设备的固件，刷机之后默认的系统分区利用的感觉太不合理了。
开始空余68MB，再用64MB作为/boot分区以及256MB作为/rom分区；
然后又空余376MB，再用2GB作为/overlay分区；
剩余了4.5GB空间，迁移不了docker，只能装一下OpenClash之类的小应用，想装多一些其他的docker应用都不可以。
所以能不能优化一下n1的固件包，把空余的空间都利用起来呢？
比如空余的空间都调整到1MB以内，把其他空余的空间都给到/overlay分区呢
[image]
当然加个外置硬盘可以解决这个问题，合理优化一下分区，就完全用不到外置硬盘了
麻烦看看能不能优化一下，谢谢

Mioscout, 2025-01-03:
用diskman手动挂载不行吗？

its-lis, 2025-01-03:
> 用diskman手动挂载不行吗？

不行的，iStoreOS限制了，只能有四个分区
[image]
新建就会报Error：Can't create any more partitions
而且我最想解决的问题是，不挂在外置硬盘的情况下，把/overlay扩容大一些，安装一些docker应用

Mioscout, 2025-01-03:
[image]
那看来还有些区别，我这个是R2S的

its-lis, 2025-01-06:
这个样子啊，但是这个分区利用率着实有的太低了，要是其他主机可能不太在乎这些空间，主要是n1设备只有8G的硬盘

MACFORNAME, 2025-01-08:
https://www.jianshu.com/p/013910b27c45
按此链接说的方法操作，没有问题的。

jjm2473, 2025-01-14:
n1的bootloader不是开源的，那些跳过的位置实际上可能呗booloader使用，我们不能乱写数据，不然可能会变砖无法启动。

dj56959566, 2025-02-22:
> 这个样子啊，但是这个分区利用率着实有的太低了，要是其他主机可能不太在乎这些空间，主要是n1设备只有8G的硬盘

要是有自动在shh 有全自动脚本就好了。一复进去。自动调整然后就利用完本机空间。这个固件用哪个脚本没用。我也没弄好。

jjm2473, 2025-06-06:
千万别使用前面留出来的空间，除非你想变砖。N1的bootloader是厂商的，不是开源的，它会把一些数据存在这些位置。

jjm2473, 2025-06-06:
就算用起来了又怎样，就为了多400MB空间？docker迁移就缺这几百M吗？

closed 2025-06-06 by jjm2473