istoreos/istoreos #2040: iStore商店插件突然无法正常升级

ocaicaio, 2025-01-07:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题
[image]

A：iStore 显示当前版本 v0.1.27-2，有最新 iStore v0.1.27-3 可以升级，点击升级按钮后如上图，显示升级成功，但刷新页面后实际上并没有升级！？嗯，本身是有梯子，请问这里是什么情况。

### (2) 路由器型号和固件版本

A：iStoreOS 22.03.7 2024080210 / LuCI istoreos-22.03 branch git-24.207.28853-3b34de8

### (3) 详细日志和/或截图

A：[image]

jjm2473, 2025-01-08:
istore里面安装“系统便利工具”，里面有完全重装istore的功能

jjm2473, 2025-01-08:
还有，你这个istore不是官方版本。看截图，官方版本没有“更多插件。。。”文案。肯定是因为你使用了第三方软件源导致istore安装到了第三方的版本，第三方版本有问题也不需要再反馈了。

[image]

closed 2025-01-08 by jjm2473

ocaicaio, 2025-01-08:
固件就是官方版本的！可能我在软件包源那里设置了自定义第三方源吧！
我自己强制安装你家的istore 应用商店包文件就搞定了，还是谢谢大家！