istoreos/istoreos #2042: 开启ipv6 无法拉取Docker项目

1216371236, 2025-01-08:
开启ipv6 无法拉取Docker项目
我拉的网心云提示:
docker pull onething1/wxedge
Using default tag: latest
Error response from daemon: Get "https://registry-1.docker.io/v2/": read tcp [240e:3b5:32a7:5080::1]:47730->[2600:1f18:2148:bc00:8d61:9b62:40aa:8bb8]:443: read: connection reset by peer
Error: No such container: wxedge
docker run --restart=unless-stopped -d     --privileged     --network=host     --dns=127.0.0.1     --tmpfs /run     --tmpfs /tmp     -v "/mnt/sata1-4/wxedge1:/storage"     -v "/mnt/sata1-4/wxedge1/containerd:/var/lib/containerd"     -e PLACE=CTKS -e TZ=Asia/Shanghai --name wxedge "onething1/wxedge"
WARNING: Localhost DNS setting (--dns=127.0.0.1) may fail in containers.
Unable to find image 'onething1/wxedge:latest' locally
docker: Error response from daemon: Head "https://registry-1.docker.io/v2/onething1/wxedge/manifests/latest": Get "https://auth.docker.io/token?scope=repository%3Aonething1%2Fwxedge%3Apull&service=registry.docker.io": read tcp [240e:3b5:32a7:5080::1]:46372->[2600:1f18:2148:bc02:445d:9ace:d20b:c303]:443: read: connection reset by peer.
See 'docker run --help'

关闭IPV6 之后才能拉取 ,  我之后又开启了IPV6  进到网心云绑定界面  二维码加载不出来

haobinnan, 2025-01-09:
更换上有dns服务器地址，运营商的的dns解析ipv6有问题，推荐用阿里云。

1216371236, 2025-01-10:
> 更换上游dns服务器地址，运营商的dns解析ipv6有问题，推荐用阿里云。

我试一下 [强]