istoreos/istoreos #2044: istoreos作为主路由，设置二级路由。实现主路由下设备访问二级路由下设备

kongweiteng, 2025-01-09:
反馈bug/问题模板，提建议请删除
[image]

## 1.关于你要提交的问题
istoreos 作为主路由，华为Q6套装作为二级路由，分别管理自己的DHCP
主路由lan网段 ： 192.168.100.2--192.168.100.254
二级路由wan ip地址：192.168.100.221
二级路由lan 网段： 192.168.3.2--192.168.3.254

从istoreos 终端ping 192.168.100.221 网络不通，二级路由的防火墙已经完全关闭，手机端也关闭了。
主路由下多个设备可以dhcp获取到ip地址，并且两个电脑之间能ping通，就是和二级路由器无法ping通。

我想实现的是能让主路由ping通二级路由的wan口地址（192.168.100.221），进而通过路由表实现主路由下设备可以访问二级路由下的设备。但是现在卡在了无法ping通 192.168.100.221，请问是哪里的配置不对吗？
Q：是否用关键词搜索了issue? (使用 "x" 选择)
没有类似的issue

### (2) 路由器型号和固件版本
X 86 安装的最新版本的istoreos固件，20241227

kongweiteng, 2025-01-09:
[image]
[image]
[image]
[image]
[image]

paulchan11, 2025-06-13:
跟你一样问题，