istoreos/istoreos #2045: iStoreOS 22.03.7 不稳定经常断线

Amoystyle, 2025-01-09:
### (1) 具体问题

不稳定，经常断线，差不多9个小时就会断线重连

### (2) 路由器型号和固件版本

```txt
设备型号    FriendlyElec NanoPi R5S
固件版本    iStoreOS 22.03.7 2024122712
内核版本    5.10.221
已启动      9小时38分52秒
系统时间    2025-01-09 20:16:36
```

### (3) 详细日志和/或截图

```sh
Wed Jan  8 22:36:20 2025 daemon.err usbmuxd[8750]: [22:36:20.530][3] usbmuxd v1.1.1 starting up
Wed Jan  8 22:36:20 2025 daemon.err usbmuxd[8750]: [22:36:20.532][3] Using libusb 1.0.24
Wed Jan  8 22:36:20 2025 daemon.err usbmuxd[8750]: [22:36:20.535][3] Initialization complete
Wed Jan  8 22:36:20 2025 daemon.err usbmuxd[8750]: [22:36:20.535][3] Enabled exit on SIGUSR1 if no devices are attached. Start a new instance with "--exit" to trigger.
Wed Jan  8 22:36:20 2025 daemon.err ModemManager[7088]: hotplug: parent device sysfspath not found
Wed Jan  8 22:36:20 2025 daemon.debug ModemManager[7088]: hotplug: cached event found: action=add, name=tty14, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty14
Wed Jan  8 22:36:20 2025 daemon.debug ModemManager[7088]: hotplug: event reported: action=add, name=tty14, subsystem=tty
Wed Jan  8 22:36:20 2025 daemon.err ModemManager[7088]: hotplug: parent device sysfspath not found
Wed Jan  8 22:36:20 2025 daemon.debug ModemManager[7088]: hotplug: cached event found: action=add, name=tty42, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty42
Wed Jan  8 22:36:20 2025 daemon.debug ModemManager[7088]: hotplug: event reported: action=add, name=tty42, subsystem=tty
Wed Jan  8 22:36:20 2025 daemon.err ModemManager[7088]: hotplug: parent device sysfspath not found
Wed Jan  8 22:36:20 2025 daemon.debug ModemManager[7088]: hotplug: cached event found: action=add, name=tty32, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty32
Wed Jan  8 22:36:20 2025 daemon.debug ModemManager[7088]: hotplug: event reported: action=add, name=tty32, subsystem=tty
Wed Jan  8 22:36:20 2025 daemon.info smartd[8954]: smartd 7.2 2020-12-30 r5155 [aarch64-linux-5.10.221] (localbuild)
Wed Jan  8 22:36:20 2025 daemon.info smartd[8954]: Copyright (C) 2002-20, Bruce Allen, Christian Franke, www.smartmontools.org
Wed Jan  8 22:36:20 2025 daemon.info smartd[8954]: Opened configuration file /etc/smartd.conf
Wed Jan  8 22:36:20 2025 daemon.crit smartd[8954]: File /etc/smartd.conf line 3 (drive DEFAULT): Missing argument to -n Directive
Wed Jan  8 22:36:20 2025 daemon.crit smartd[8954]: Valid arguments to -n Directive are:
Wed Jan  8 22:36:20 2025 daemon.crit smartd[8954]: never[,N][,q], sleep[,N][,q], standby[,N][,q], idle[,N][,q]
Wed Jan  8 22:36:20 2025 daemon.crit smartd[8954]: Configuration file /etc/smartd.conf has fatal syntax errors.
Wed Jan  8 22:36:20 2025 daemon.info smartd[8954]: Monitoring 0 ATA/SATA, 0 SCSI/SAS and 0 NVMe devices
Wed Jan  8 22:36:20 2025 daemon.err ModemManager[7088]: hotplug: parent device sysfspath not found
Wed Jan  8 22:36:20 2025 daemon.err rmmod: module is not loaded
Wed Jan  8 22:36:20 2025 daemon.debug ModemManager[7088]: hotplug: cached event found: action=add, name=tty60, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty60
Wed Jan  8 22:36:20 2025 user.notice cifs-mount: Reloading cifs-mount due to ifup of lan (br-lan)
Wed Jan  8 22:36:20 2025 daemon.debug ModemManager[7088]: hotplug: event reported: action=add, name=tty60, subsystem=tty
Wed Jan  8 22:36:20 2025 daemon.notice procd: /etc/rc.d/S96led: setting up led WAN
Wed Jan  8 22:36:20 2025 daemon.notice procd: /etc/rc.d/S96led: setting up led LAN1
Wed Jan  8 22:36:20 2025 daemon.notice procd: /etc/rc.d/S96led: setting up led LAN2
Wed Jan  8 22:36:20 2025 daemon.err ModemManager[7088]: hotplug: parent device sysfspath not found
Wed Jan  8 22:36:20 2025 daemon.debug ModemManager[7088]: hotplug: cached event found: action=add, name=ttyFIQ0, subsystem=tty, sysfspath=/sys/devices/platform/fiq_debugger.0/tty/ttyFIQ0
Wed Jan  8 22:36:20 2025 daemon.debug ModemManager[7088]: hotplug: event reported: action=add, name=ttyFIQ0, subsystem=tty
Wed Jan  8 22:36:20 2025 daemon.err ModemManager[7088]: hotplug: parent device sysfspath not found
Wed Jan  8 22:36:20 2025 daemon.debug ModemManager[7088]: hotplug: cached event found: action=add, name=tty22, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty22
Wed Jan  8 22:36:20 2025 daemon.debug ModemManager[7088]: hotplug: event reported: action=add, name=tty22, subsystem=tty
Wed Jan  8 22:36:20 2025 daemon.err ModemManager[7088]: hotplug: parent device sysfspath not found
Wed Jan  8 22:36:21 2025 daemon.debug ModemManager[7088]: hotplug: cached event found: action=add, name=tty6, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty6
Wed Jan  8 22:36:21 2025 daemon.debug ModemManager[7088]: hotplug: event reported: action=add, name=tty6, subsystem=tty
Wed Jan  8 22:36:21 2025 daemon.err ModemManager[7088]: hotplug: parent device sysfspath not found
Wed Jan  8 22:36:21 2025 daemon.debug ModemManager[7088]: hotplug: cached event found: action=add, name=tty50, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty50
Wed Jan  8 22:36:21 2025 daemon.debug ModemManager[7088]: hotplug: event reported: action=add, name=tty50, subsystem=tty
Wed Jan  8 22:36:21 2025 daemon.err ModemManager[7088]: hotplug: parent device sysfspath not found
Wed Jan  8 22:36:21 2025 daemon.debug ModemManager[7088]: hotplug: cached event found: action=add, name=tty12, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty12
Wed Jan  8 22:36:21 2025 daemon.debug ModemManager[7088]: hotplug: event reported: action=add, name=tty12, subsystem=tty
Wed Jan  8 22:36:21 2025 daemon.err ModemManager[7088]: hotplug: parent device sysfspath not found
Wed Jan  8 22:36:21 2025 daemon.debug ModemManager[7088]: hotplug: cached event found: action=add, name=tty40, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty40
Wed Jan  8 22:36:21 2025 daemon.debug ModemManager[7088]: hotplug: event reported: action=add, name=tty40, subsystem=tty
Wed Jan  8 22:36:21 2025 daemon.err ModemManager[7088]: hotplug: parent device sysfspath not found
Wed Jan  8 22:36:21 2025 daemon.debug ModemManager[7088]: hotplug: cached event found: action=add, name=tty30, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty30
Wed Jan  8 22:36:21 2025 daemon.debug ModemManager[7088]: hotplug: event reported: action=add, name=tty30, subsystem=tty
Wed Jan  8 22:36:21 2025 daemon.info samba4-server: io_uring module found, enabling VFS io_uring. (also needs Kernel 5.4+ Support)
Wed Jan  8 22:36:21 2025 daemon.err ModemManager[7088]: hotplug: parent device sysfspath not found
Wed Jan  8 22:36:21 2025 daemon.debug ModemManager[7088]: hotplug: cached event found: action=add, name=tty59, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty59
[... 505 lines cut ...]
Thu Jan  9 15:10:17 2025 daemon.info dnsmasq-dhcp[1]: DHCPACK(br-lan) 192.168.100.185 94:f8:27:cf:05:f0 chuangmi_camera_051a01
Thu Jan  9 15:13:46 2025 daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(br-lan) 192.168.100.176 94:24:b8:02:5f:53
Thu Jan  9 15:13:46 2025 daemon.info dnsmasq-dhcp[1]: DHCPACK(br-lan) 192.168.100.176 94:24:b8:02:5f:53
Thu Jan  9 15:16:49 2025 daemon.info dnsmasq-dhcp[1]: DHCPDISCOVER(br-lan) 60:1d:9d:78:da:00
Thu Jan  9 15:16:49 2025 daemon.info dnsmasq-dhcp[1]: DHCPOFFER(br-lan) 192.168.100.208 60:1d:9d:78:da:00
Thu Jan  9 15:16:50 2025 daemon.info dnsmasq-dhcp[1]: DHCPDISCOVER(br-lan) 60:1d:9d:78:da:00
Thu Jan  9 15:16:50 2025 daemon.info dnsmasq-dhcp[1]: DHCPOFFER(br-lan) 192.168.100.208 60:1d:9d:78:da:00
Thu Jan  9 15:16:50 2025 daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(br-lan) 192.168.100.208 60:1d:9d:78:da:00
Thu Jan  9 15:16:50 2025 daemon.info dnsmasq-dhcp[1]: DHCPACK(br-lan) 192.168.100.208 60:1d:9d:78:da:00
Thu Jan  9 15:38:27 2025 daemon.warn dnsmasq[1]: possible DNS-rebind attack detected: requality.android.shouji.sogou.com
Thu Jan  9 15:46:12 2025 daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(br-lan) 192.168.100.171 22:5e:52:c0:50:e7
Thu Jan  9 15:46:12 2025 daemon.info dnsmasq-dhcp[1]: DHCPACK(br-lan) 192.168.100.171 22:5e:52:c0:50:e7 HUAWEI_P30-c31bf51071
Thu Jan  9 15:48:10 2025 daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(br-lan) 192.168.100.191 1c:d6:be:7f:11:7f
Thu Jan  9 15:48:10 2025 daemon.info dnsmasq-dhcp[1]: DHCPACK(br-lan) 192.168.100.191 1c:d6:be:7f:11:7f
Thu Jan  9 15:59:48 2025 daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(br-lan) 192.168.100.228 72:0f:91:65:62:bb
Thu Jan  9 15:59:48 2025 daemon.info dnsmasq-dhcp[1]: DHCPACK(br-lan) 192.168.100.228 72:0f:91:65:62:bb Redmi-Note-11T-Pro
[... cut ...]
Thu Jan  9 17:23:49 2025 daemon.info dnsmasq-dhcp[1]: DHCPOFFER(br-lan) 192.168.100.210 88:71:e5:b7:25:ce
Thu Jan  9 17:23:49 2025 daemon.info dnsmasq-dhcp[1]: DHCPDISCOVER(br-lan) 88:71:e5:b7:25:ce
Thu Jan  9 17:23:49 2025 daemon.info dnsmasq-dhcp[1]: DHCPOFFER(br-lan) 192.168.100.210 88:71:e5:b7:25:ce
Thu Jan  9 17:23:49 2025 daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(br-lan) 192.168.100.210 88:71:e5:b7:25:ce
Thu Jan  9 17:23:49 2025 daemon.info dnsmasq-dhcp[1]: DHCPACK(br-lan) 192.168.100.210 88:71:e5:b7:25:ce
```

Amoystyle, 2025-01-11:
[ppp0] invalid sysfs path read for net/ppp0

jjm2473, 2025-01-13:
哪里看得出来断线？

closed 2025-01-14 by Amoystyle