istoreos/istoreos #2050: 问题反馈，istoreos最新测试版24.10.0-rc4，内核6.67，安装emby无法识别核显，求解决

liweilin999, 2025-01-12:
[image]

jjm2473, 2025-01-13:
什么cpu

jjm2473, 2025-01-13:
开机以后执行下 `ls -l /dev/dri/`，还有执行`lspci -nnk`

liweilin999, 2025-01-13:
> 什么CPU

n305

liweilin999, 2025-01-13:
> 什么CPU

n305

liweilin999, 2025-01-13:
> 开机以后执行下 ，还有执行，把输出都发到这里。`ls -l /dev/dri/``lspci -nnk`

[image]

问题反馈，istoreos最新测试版24.10.0-rc4，内核6.67，istoreos-24.10.0-rc4-2025011019-x86-64-squashfs-combined-efi.img，安装emby无法识别核显，求解决，机器是n305，安装istoreos商店里emby插件也是不行，无法识别核显，然后自己创建容器也映射了核显驱动也不行，以下是创建命令
docker run -d
--privileged
--network=host
-v /opt/emby2/config:/config
-v /opt/media/:/media
-e TZ="Asia/Shanghai"
--device=/dev/dri:/dev/dri
-e UID=0
-e GID=0
-e GIDLIST=022
--restart=unless-stopped
--name emby
amilys/embyserver:latest

[image]

liweilin999, 2025-01-13:
> > 什么CPU
> 
> n305和i9-12900h

liweilin999, 2025-01-13:
> > 开机以后执行下 ，还有执行，把输出都发到这里。`` ls -l /dev/dri/``lspci -nnk ``
> > ![屏幕截图 2025-01-13 114524](https://private-user-images.githubusercontent.com/59835333/402417015-e314929b-5697-4467-ba2e-f3f15024f939.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzY3NDA3MjksIm5iZiI6MTczNjc0MDQyOSwicGF0aCI6Ii81OTgzNTMzMy80MDI0MTcwMTUtZTMxNDkyOWItNTY5Ny00NDY3LWJhMmUtZjNmMTUwMjRmOTM5LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTAxMTMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwMTEzVDAzNTM0OVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWE3NzY1MjBkOTJhMzZmZjU5YjI2YTViMDRhZWZiMTViOGI3YzJmMTcxZjg5YmJmYmI5OGI1NzE4MDM0MmIyZmEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.Gsmc2UdI1vqmTtB8ImSqHV2lGII2NUexsrBz1PbmuCM)
> 
> 问题反馈，istoreos最新测试版24.10.0-rc4，内核6.67，istoreos-24.10.0-rc4-2025011019-x86-64-squashfs-combined-efi.img，安装emby无法识别核显，求解决，机器是n305和i9-12900h，两个CPU试过都是一样，安装istoreos商店里emby插件也是不行，无法识别核显，然后自己创建容器也映射了核显驱动也不行，以下是创建命令 docker run -d --privileged --network=host -v /opt/emby2/config:/config -v /opt/media/:/media -e TZ="Asia/Shanghai" --device=/dev/dri:/dev/dri -e UID=0 -e GID=0 -e GIDLIST=022 --restart=unless-stopped --name emby amilys/embyserver:latest
> 
> ![屏幕截图 2025-01-12 143124](https://private-user-images.githubusercontent.com/59835333/402416736-c2654875-b96c-45de-9288-9f9cc532d5a5.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzY3NDA3MjksIm5iZiI6MTczNjc0MDQyOSwicGF0aCI6Ii81OTgzNTMzMy80MDI0MTY3MzYtYzI2NTQ4NzUtYjk2Yy00NWRlLTkyODgtOWY5Y2M1MzJkNWE1LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTAxMTMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwMTEzVDAzNTM0OVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWNjZmQ4NzRiNmUzMjA2MzBhNzEzNGZjNWQzN2ZmZTVlNjkyMjBhMjZhYzgxMmUxZGFjMjQ5ODc0MjYwYmRmNzkmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.s-DWSODS_r7nsIlXgFb2ymqthIdo7sFEvKZcV76L-NU)

liweilin999, 2025-01-13:
> 不要截图，把输出粘贴过来。你发的图都裂了看不到
root@iStoreOS:~# ls -l /dev/dri/
crw-------    1 root     root      226,   0 Jan 12 17:15 card0
crw-------    1 root     root      226, 128 Jan 12 17:15 renderD128
root@iStoreOS:~# lspci -nnk
00:00.0 Host bridge [0600]: Intel Corporation 440FX - 82441FX PMC [Natoma] [8086:1237] (rev 02)
        Subsystem: Red Hat, Inc. Qemu virtual machine [1af4:1100]
lspci: Unable to load libkmod resources: error -2
00:01.0 ISA bridge [0601]: Intel Corporation 82371SB PIIX3 ISA [Natoma/Triton II] [8086:7000]
        Subsystem: Red Hat, Inc. Qemu virtual machine [1af4:1100]
00:01.1 IDE interface [0101]: Intel Corporation 82371SB PIIX3 IDE [Natoma/Triton II] [8086:7010]
        Subsystem: Red Hat, Inc. Qemu virtual machine [1af4:1100]
        Kernel driver in use: ata_piix
00:01.2 USB controller [0c03]: Intel Corporation 82371SB PIIX3 USB [Natoma/Triton II] [8086:7020] (rev 01)
        Subsystem: Red Hat, Inc. QEMU Virtual Machine [1af4:1100]
        Kernel driver in use: uhci_hcd
00:01.3 Bridge [0680]: Intel Corporation 82371AB/EB/MB PIIX4 ACPI [8086:7113] (rev 03)
        Subsystem: Red Hat, Inc. Qemu virtual machine [1af4:1100]
00:02.0 VGA compatible controller [0300]: Red Hat, Inc. Virtio 1.0 GPU [1af4:1050] (rev 01)
        Subsystem: Red Hat, Inc. Device [1af4:1100]
        Kernel driver in use: virtio-pci
00:03.0 Unclassified device [00ff]: Red Hat, Inc. Virtio memory balloon [1af4:1002]
        Subsystem: Red Hat, Inc. Device [1af4:0005]
        Kernel driver in use: virtio-pci
00:05.0 PCI bridge [0604]: Red Hat, Inc. QEMU PCI-PCI bridge [1b36:0001]
00:09.0 Communication controller [0780]: Red Hat, Inc. Virtio console [1af4:1003]
        Subsystem: Red Hat, Inc. Device [1af4:0003]
        Kernel driver in use: virtio-pci
00:10.0 VGA compatible controller [0300]: Intel Corporation Alder Lake-P GT2 [Iris Xe Graphics] [8086:46a6] (rev 0c)
        Subsystem: Device [0301:02f3]
00:12.0 Ethernet controller [0200]: Red Hat, Inc. Virtio network device [1af4:1000]
        Subsystem: Red Hat, Inc. Device [1af4:0001]
        Kernel driver in use: virtio-pci
00:1e.0 PCI bridge [0604]: Red Hat, Inc. QEMU PCI-PCI bridge [1b36:0001]
00:1f.0 PCI bridge [0604]: Red Hat, Inc. QEMU PCI-PCI bridge [1b36:0001]
01:01.0 SCSI storage controller [0100]: Red Hat, Inc. Virtio SCSI [1af4:1004]
        Subsystem: Red Hat, Inc. Device [1af4:0008]
        Kernel driver in use: virtio-pci
01:02.0 SCSI storage controller [0100]: Red Hat, Inc. Virtio SCSI [1af4:1004]
        Subsystem: Red Hat, Inc. Device [1af4:0008]
        Kernel driver in use: virtio-pci
root@iStoreOS:~#

liweilin999, 2025-01-13:
现在复制粘贴了，查看下

jjm2473, 2025-01-14:
你这个是虚拟机，你确定配置好了？这个虚拟机刷istoreos 22.03能硬解吗

liweilin999, 2025-01-14:
虚拟机直通好了，虚拟机和物理机安装都是一样，emby无法识别核显，我是物理机试过不行然后才试试虚拟机，结果还是不行，刷22.03能硬解的

jjm2473, 2025-01-14:
OK，那应该只是没开GuC，之后会加上

jjm2473, 2025-01-14:
不过GuC只会影响编码，不会影响解码

liweilin999, 2025-01-14:
转码播放会影响吗？

jjm2473, 2025-01-14:
看设置，jellyfin支持软件编码。固件更新了，应该没问题了。

https://fw0.koolcenter.com/iStoreOS/alpha/x86_64/istoreos-24.10.0-rc4-2025011418-x86-64-squashfs-combined.img.gz

https://fw0.koolcenter.com/iStoreOS/alpha/x86_64_efi/istoreos-24.10.0-rc4-2025011418-x86-64-squashfs-combined-efi.img.gz

liweilin999, 2025-01-14:
好的大佬，我试试看

 

 
https://fw0.koolcenter.com/iStoreOS/alpha/x86_64/istoreos-24.10.0-rc4-2025011418-x86-64-squashfs-combined.img.gz
 
https://fw0.koolcenter.com/iStoreOS/alpha/x86_64_efi/istoreos-24.10.0-rc4-2025011418-x86-64-squashfs-combined-efi.img.gz

closed 2025-01-16 by jjm2473