istoreos/istoreos #2052: 关于 N1 接USB3.0 的 有硬盘盒，USB驱动 异常复位问题！

TMODER, 2025-01-12:
接了个USB3.0 硬盘盒，但发现USB 驱动经常异常：
[276225.475729] sd 0:0:0:0: [sda] tag#24 uas_eh_abort_handler 0 uas-tag 7 inflight: CMD 
[276225.475734] sd 0:0:0:0: [sda] tag#24 CDB: opcode=0x8a 8a 00 00 00 00 00 02 81 31 70 00 00 00 08 00 00
[276225.475743] sd 0:0:0:0: [sda] tag#23 uas_eh_abort_handler 0 uas-tag 6 inflight: CMD 
[276225.475749] sd 0:0:0:0: [sda] tag#23 CDB: opcode=0x8a 8a 00 00 00 00 00 02 80 31 70 00 00 00 08 00 00
[276225.475758] sd 0:0:0:0: [sda] tag#22 uas_eh_abort_handler 0 uas-tag 5 inflight: CMD 
[276225.475764] sd 0:0:0:0: [sda] tag#22 CDB: opcode=0x8a 8a 00 00 00 00 00 02 80 30 f0 00 00 00 08 00 00
[276225.475773] sd 0:0:0:0: [sda] tag#21 uas_eh_abort_handler 0 uas-tag 4 inflight: CMD 
[276225.475779] sd 0:0:0:0: [sda] tag#21 CDB: opcode=0x8a 8a 00 00 00 00 00 02 80 10 08 00 00 00 08 00 00
[276225.475788] sd 0:0:0:0: [sda] tag#20 uas_eh_abort_handler 0 uas-tag 3 inflight: CMD 
[276225.475793] sd 0:0:0:0: [sda] tag#20 CDB: opcode=0x8a 8a 00 00 00 00 00 02 80 10 00 00 00 00 08 00 00
[276225.475803] sd 0:0:0:0: [sda] tag#0 uas_eh_abort_handler 0 uas-tag 12 inflight: CMD 
[276225.475809] sd 0:0:0:0: [sda] tag#0 CDB: opcode=0x8a 8a 00 00 00 00 00 0f 01 47 20 00 00 00 08 00 00
[276225.535950] scsi host0: uas_eh_device_reset_handler start
[276225.536564] usb 1-1: cmd cmplt err -2
[276225.697303] usb 1-1: reset high-speed USB device number 2 using xhci-hcd
[276225.884404] scsi host0: uas_eh_device_reset_handler success

找到一个遇到相同问题的 issue，链接：https://github.com/istoreos/istoreos/issues/1006，
提供了 修改 /boot/grub/grub.cfg 方案，但N1 下boot 没这些配置。

使用 echo '0080:a001:u' > /sys/module/usb_storage/parameters/quirks
将 驱动类型 由 usa  修改为 Driver=usb-storage 后，重新插拔USB后正常，USB 驱动不再复位！

但上述修改在重启系统后失效，有无大佬知道，如何将这个修改 写入启动配置中？

TMODER, 2025-01-12:
尝试   
1.opkg remove kmod-usb-storage-uas 卸载  ,重启设备默认还是 uas
2.在 /etc/modules.conf  中添加 blacklist uas 重启无效

当前只能在启动项中加入 echo '0080:a001:u' > /sys/module/usb_storage/parameters/quirks 
然后插拔一次 硬盘，有点孬， 还没找到正确姿势...

研究中.........  <苦涩>！！！

yipush, 2025-01-14:
我也遇到了，不知道怎么解决

TMODER, 2025-01-14:
> 我也遇到了，不知道怎么解决

尝试了 usbreset 命令也无效！ 感觉是 uas 和 3.0 usb 驱动产生了某种兼容性冲突!
<苦涩>

yipush, 2025-01-14:
> > 我也遇到了，不知道怎么解决
> 
> 尝试了 usbreset 命令也无效！感觉是 uas 和 3.0 usb 驱动产生了某种兼容性冲突！ <苦涩>

反正就是uas这个鬼东西引起的，不知道要怎么彻底禁用uas，一点用都没

TMODER, 2025-01-14:
> > > 我也遇到了，不知道怎么解决
> > 
> > 
> > 尝试了 usbreset 命令也无效！感觉是 uas 和 3.0 usb 驱动产生了某种兼容性冲突！ <苦涩>
> 
> 反正就是uas这个鬼东西引起的，不知道要怎么彻底禁用uas，一点用都没
我还在找方法，要有进展 我再更新 ^_^

TMODER, 2025-01-14:
> > > 我也遇到了，不知道怎么解决
> > 
> > 
> > 尝试了 usbreset 命令也无效！感觉是 uas 和 3.0 usb 驱动产生了某种兼容性冲突！ <苦涩>
> 
> 反正就是uas这个鬼东西引起的，不知道要怎么彻底禁用uas，一点用都没

还在找方法，有进展，再更新 ^_^

TMODER, 2025-01-15:
> 我也遇到了，不知道怎么解决

找到了个 帖子：https://forum.openwrt.org/t/solved-xiaomi-mi-r-3g-uasp-external-usb-hard-disk-drive/30811
但按照方法没用！ MMP 你可以试试，我这边不行

yipush, 2025-01-15:
> > 我也遇到了，不知道怎么解决
> 
> 找到了个 帖子：https://forum.openwrt.org/t/solved-xiaomi-mi-r-3g-uasp-external-usb-hard-disk-drive/30811 但按照方法没用！ MMP 你可以试试，我这边不行

我试了也不行

TMODER, 2025-01-15:
> > > 我也遇到了，不知道怎么解决
> > 
> > 
> > 找到了个 帖子：https://forum.openwrt.org/t/solved-xiaomi-mi-r-3g-uasp-external-usb-hard-disk-drive/30811 但按照方法没用！ MMP 你可以试试，我这边不行
> 
> 我试了也不行

废了废了，目前应该只有2条路能走：
1.自定义固件，编译时不带 UAS模块
2.如果硬盘盒带电源控制，买个智能插座控制下
 
-.-

TMODER, 2025-02-13:
@yipush   老铁 找到一个可能 行得通的方案，我还没折腾，我放下链接，有时间折腾的话反馈下：
https://github.com/ffuqiangg/build_openwrt/releases/tag/2024.12.29
固件：N1-IStoreOS-2024.12.29-sfs.zip
没有内联集成UAS！ 
等有时间我也刷下看看！

lsprite, 2025-02-17:
> [@yipush](https://github.com/yipush) 老铁 找到一个可能 行得通的方案，我还没折腾，我放下链接，有时间折腾的话反馈下： https://github.com/ffuqiangg/build_openwrt/releases/tag/2024.12.29 固件：N1-IStoreOS-2024.12.29-sfs.zip 没有内联集成UAS！ 等有时间我也刷下看看！

怎么样？能用不？

TMODER, 2025-02-18:
> > [@yipush](https://github.com/yipush) 老铁 找到一个可能 行得通的方案，我还没折腾，我放下链接，有时间折腾的话反馈下： https://github.com/ffuqiangg/build_openwrt/releases/tag/2024.12.29 固件：N1-IStoreOS-2024.12.29-sfs.zip 没有内联集成UAS！ 等有时间我也刷下看看！
> 
> 怎么样？能用不？

大兄弟，还没试，最近没来得及折腾！等稍微闲点。。。~.~

yipush, 2025-02-19:
> [@yipush](https://github.com/yipush) 老铁 找到一个可能 行得通的方案，我还没折腾，我放下链接，有时间折腾的话反馈下： https://github.com/ffuqiangg/build_openwrt/releases/tag/2024.12.29 固件：N1-IStoreOS-2024.12.29-sfs.zip 没有内联集成UAS！ 等有时间我也刷下看看！

这个感觉就是openwrt加了istore的首页和商店，那还不如直接刷filppy的openwrt固件，我之前用F大的固件都是正常的，N1我现在就用来做旁路由挂个梯子了，没用到USB。最近折腾飞牛OS，文件存储都弄到飞牛上了

TMODER, 2025-02-19:
> 这个感觉就是openwrt加了istore的首页和商店，那还不如直接刷filppy的openwrt固件，我之前用F大的固件都是正常的，N1我现在就用来做旁路由挂个梯子了，没用到USB。最近折腾飞牛OS，文件存储都弄到飞牛上了

还没有深入研究，我这边就当轻NAS玩玩，万万没想到 官方出的 这个固件，集成UAS坑死个人

owl6660sky, 2025-03-26:
如果是最新版的固件，可以在系统-杂项设置-内核启动参数-自定义内核参数，这里填写参数即可。类似：usb-storage.quirks=0080:a001:u

TMODER, 2025-03-26:
> 如果是最新版的固件，可以在系统-杂项设置-内核启动参数-自定义内核参数，这里填写参数即可。类似：usb-storage.quirks=0080:a001:u

有最新的固件吗？ N1 好久没更新了额 0.0

owl6660sky, 2025-03-26:
> > 如果是最新版的固件，可以在系统-杂项设置-内核启动参数-自定义内核参数，这里填写参数即可。类似：usb-storage.quirks=0080:a001:u
> 
> 有最新的固件吗？ N1 好久没更新了额 0.0

我用的是X86，大概更新频繁一点。不太清楚N1 的情况。

yipush, 2025-03-26:
> 如果是最新版的固件，可以在系统-杂项设置-内核启动参数-自定义内核参数，这里填写参数即可。类似：usb-storage.quirks=0080:a001:u

你说的只能是x86的这么干，散了散了

owl6660sky, 2025-03-26:
> > 如果是最新版的固件，可以在系统-杂项设置-内核启动参数-自定义内核参数，这里填写参数即可。类似：usb-storage.quirks=0080:a001:u
> 
> 有最新的固件吗？ N1 好久没更新了额 0.0

好像引导方式不一样。不知道是不是/boot/uEnv.txt。如果你愿意冒险的话，但我没用过N1，不能保证。

owl6660sky, 2025-03-26:
> > 如果是最新版的固件，可以在系统-杂项设置-内核启动参数-自定义内核参数，这里填写参数即可。类似：usb-storage.quirks=0080:a001:u
> 
> 有最新的固件吗？ N1 好久没更新了额 0.0

这是deepseek给的建议，参考第二条，
在 N1（ARM） 或 X86 设备上，可以通过修改 内核启动参数 来设置 usb-storage.quirks。

（1）N1（ARM，使用 uEnv.txt）
编辑 /boot/uEnv.txt，在 APPEND= 行添加 usb-storage.quirks 参数：

APPEND=root=UUID=xxxx ... usb-storage.quirks=vid:pid:u
例如，如果 USB 设备的 vid=152d、pid=1561（常见于某些 JMicron 桥接的硬盘盒），并希望禁用 UAS（u 表示 no_uas）：

APPEND=root=UUID=xxxx ... usb-storage.quirks=152d:1561:u

xrixing, 2025-05-29:
我的掉盘问题好像解决了，不知道掉盘原因是不是这个问题