istoreos/istoreos #2056: 22.03.7版本Dnsmasq运行后域名无法解析出ipv6地址

SwordRays, 2025-01-13:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：升级到22.03.7 2024122712版本后，局域网(包括路由器自身)dns域名解析无法解析出IPv6地址。前一个版本没问题。

### (2) 路由器型号和固件版本

A：	FastRhino R66s + 22.03.7 2024122712

### (3) 详细日志和/或截图

A：在没有修改任何配置的情况下，从前一个版本升级到该版本就会出现该问题。停止Dnsmasq后即可正常解析出IPv6，手动运行dnsmasq --log-debug也可以正常解析出IPv6。

PS1：路由器的IPv6分配是正常的，从外网可以正常通过IPv6地址访问局域网中设备。

PS2：观察到以下现象1：
    ## 系统自带的执行用户为dnsmasq：
    # ps | grep dnsmasq
    16659 root      2576 S    {dnsmasq} /sbin/ujail -t 5 -n dnsmasq -u -l -r /bin/ubus -r /etc/TZ
    16662 dnsmasq  58028 S    /usr/sbin/dnsmasq -C /var/etc/dnsmasq.conf.cfg01411c -k -x /var/run
    
    ## 手动启动时，执行用户为nobody：
    # dnsmasq --log-debug
    # ps | grep dnsmasq
    19658 nobody    2988 S    dnsmasq --log-debug

PS3：观察到以下现象2：
    ## 使用系统方式启动dnsmasq时，dns服务器是127.0.0.1：
    # nslookup unraid.net
    Server:		127.0.0.1
    Address:	127.0.0.1:53
    
    Name:	unraid.net
    Address: ::
    
    Non-authoritative answer:
    Name:	unraid.net
    Address: 104.26.3.117
    Name:	unraid.net
    Address: 172.67.69.176
    Name:	unraid.net
    Address: 104.26.2.117
    
    ## 使用手动启动时，dns服务器是配置的223.5.5.5：
    # nslookup unraid.net
    Server:		223.5.5.5
    Address:	223.5.5.5:53
    
    Non-authoritative answer:
    Name:	unraid.net
    Address: 104.26.3.117
    Name:	unraid.net
    Address: 104.26.2.117
    Name:	unraid.net
    Address: 172.67.69.176
    
    Non-authoritative answer:
    Name:	unraid.net
    Address: 2606:4700:20::681a:275
    Name:	unraid.net
    Address: 2606:4700:20::ac43:45b0
    Name:	unraid.net
    Address: 2606:4700:20::681a:375

SwordRays, 2025-01-14:
以下设置是默认的，未修改过：
基本设置 - DNS - DNS分流：Dnsmasq
基本设置 - DNS - 过滤代理域名 IPv6：勾选【请大佬帮忙解释下这个的具体含义】
高级设置 - IPv6 透明代理(TProxy)：未勾选【请大佬帮忙解释下这个的具体含义】

在以前的旧固件中是正常的，无论解析出ipv6还是v4，域名在列表中的都正常代理。

升级到最新固件后就不正常，出现了解析出的ipv6是::的情况。(如果是从解析结果中彻底删除v6的解析还好，用ipv4也行，但返回了这个错误的ipv6就出问题了)

现在测试出可以解决问题的方法：
1) 手动停止Dnsmasq进程，手动启动
2) 关闭passwall
3) 在passwall中将 基本设置 - DNS - DNS分流 从 Dnsmasq 改为 Dnsmasq+ChinaDNS-NG

bcseputetto, 2025-01-14:
> 基本设置 - DNS - 过滤代理域名 IPv6：勾选【请大佬帮忙解释下这个的具体含义】

主要就是这个，跟固件新旧版本无关。
dnsmasq可以丢弃IPv6解析记录，但是不能做到不代理域名可以解析，代理域名单独丢弃IPv6解析记录
所以xiaorouji大佬在dnsmasq分流时，代理域名过滤IPv6时，只能把它全部解析为 :: 这个错误的IPv6
所以推荐你用chinadns-ng的情况下，使用`过滤代理域名 IPv6`功能

jjm2473, 2025-01-14:
istoreos默认关掉了ipv6的dns，因为ipv6太多问题，一般人也用不上。

参考这个可以开启ipv6 https://github.com/istoreos/istoreos/issues/1172 。

如果你想让passwall代理的域名只解析ipv4，正常域名可以解析ipv6，可以参考这个 https://github.com/istoreos/istoreos/discussions/1875

Linchill, 2025-01-19:
我也出现了同样的问题  不知道降级是否会有帮助