istoreos/istoreos #2067: libcomposite 模块缺失

meisekiyou, 2025-01-17:
在r3s上部署 pikvm 时遇到 /lib/modules 下缺失 libcomposite 模块，缺失这个模块无法使用usb otg模拟键鼠，自己尝试编译在 make menuconfig 中没有找到相关选项。

meisekiyou, 2025-01-18:
这边在 .config 中启用了相关kmod后得到了 kmod-usb-lib-composite.ipk 及其依赖，但使用 opkg 安装后 lsmod 找不到 libcomposite 相关模块。

.config 配置
```
meiseki@ubuntu2410:~$ cat istoreos/.config | grep kmod-usb
# CONFIG_PACKAGE_kmod-usb-acm is not set
# CONFIG_PACKAGE_kmod-usb-atm is not set
# CONFIG_PACKAGE_kmod-usb-cm109 is not set
CONFIG_PACKAGE_kmod-usb-core=m
CONFIG_PACKAGE_kmod-usb-gadget=m
CONFIG_PACKAGE_kmod-usb-gadget-cdc-composite=m
CONFIG_PACKAGE_kmod-usb-gadget-ehci-debug=m
CONFIG_PACKAGE_kmod-usb-gadget-eth=m
CONFIG_PACKAGE_kmod-usb-gadget-hid=m
CONFIG_PACKAGE_kmod-usb-gadget-mass-storage=m
CONFIG_PACKAGE_kmod-usb-gadget-ncm=m
CONFIG_PACKAGE_kmod-usb-gadget-serial=m
```

系统终端
```
root@iStoreOS:~# opkg info kmod-usb-lib-composite
Package: kmod-usb-lib-composite
Version: 5.10.221-1
Depends: kernel (= 5.10.221-1-3e7433398bd8a0461c2e9336e17b74e3), kmod-usb-gadget, kmod-fs-configfs, kmod-usb-core
Status: install hold installed
Architecture: aarch64_cortex-a53
Installed-Time: 1735044283

root@iStoreOS:~# cat /etc/modules.d/50-usb-lib-composite 
libcomposite
```

jjm2473, 2025-04-30:
如果你增加了内核模块，但是没有出现相应的ko文件，大概率是因为这个模块已经被集成到内核中了，不再需要加载ko。

而且我已经确认，rk35x系列的22.03固件中，usb-lib-composite是集成到内核中的，在系统里执行`zcat /proc/config.gz |grep USB_LIBCOMPOSITE`就能看到：
```
root@ARS4:~# zcat /proc/config.gz |grep USB_LIBCOMPOSITE
CONFIG_USB_LIBCOMPOSITE=y
```

如果pikvm只是判断ko文件有没有存在，那绝对是个bug。

closed 2025-05-01 by meisekiyou