istoreos/istoreos #2075: zerotier插件能更新一下吗？

liuwen0828, 2025-01-22:
最新测试版安装了zerotier插件后无法启动，辛苦更新一下版本，谢谢

levi882, 2025-01-25:
同意，不管怎么设置都是未启动 #

[image]

Mayuyu007, 2025-02-17:
旧版的zerotier系统内核只有1.6.6，做这个luci界面的gitub作者 几百年没更新了，你想用界面控制只能用老版本
新版本没有luci，只能命令行控制，我在最新版的Openwrt 编译库能没找到最新版的luci

Mayuyu007, 2025-02-17:
> 同意，不管怎么设置都是未启动 #
> 
> [image]

我遇到过 版本太老了，我删了这个旧版的luci，换成新版的就好了，正常连接。

levi882, 2025-02-22:
> > 同意，不管怎么设置都是未启动 #
> > [image]
> 
> 我遇到过 版本太老了，我删了这个旧版的luci，换成新版的就好了，正常连接。

新版的要自己编译吗，我直接用的immortalwrt的软件源安装luci界面不管用，只能按照openwrt官方教程用命令行创建接口来用

closed 2025-06-05 by jjm2473