istoreos/istoreos #2079: 防火墙放行指定端口不成功

ssupssup, 2025-01-24:
用的是X86 iStoreOS 22.03.7 2024122712，像这样设置：防火墙中常规设置WAN入站数据丢弃，通信规则放行ssh访问、FTP服务器的监听端口和数据端口、rustdesk端口，也做了端口转发，都不能访问，此时ARIA2也不能连接，登录终端用iptables-legacy -L -v -n查看端口也已经放行，而放行的QBITTORRENT和NAVIDROME，终端，都可以访问。但把防火墙中常规设置WAN入站数据接受，上述服务都正常，但放行所有端口又觉得有风险，想请教该怎么办？

ssupssup, 2025-01-24:
另外还有个Qbittorrent通过绑定的openwrt域名访问问题，因为只能转发端口才能访问OPENWRT的管理页面，在域名后加个端口，假如为99，那能过域名：99就能访问和登录OPENWRT的管理页面，但在Qbittorrent插件页面中，打开“管理页面”没反应，假设QB管理页面的商品是100，通过浏览器的开发工具，看到要打开的地址是：域名：99：100，可能是这个原因导致远程没法打开管理页面，其他服务，如NAVIDROME，JELLYFIN，Jackett等，都不会带上端口99，不知道这是QB的问题还是什么问题，是否有办法解决

w9315273, 2025-01-25:
到达路由器本机监听端口 用通信规则
到达其他设备 才用端口转发

放行 也是个模棱两可的词 注意表达准确

ssupssup, 2025-01-25:
> 到达路由器本机监听端口 用通信规则 到达其他设备 才用端口转发
> 
> 放行 也是个模棱两可的词 注意表达准确

谢谢提醒，重新修改了一下

w9315273, 2025-01-25:
你的表达方式还是让我难以理解…… 索性把问题丢给GPT了 你看是这意思么

你的问题可以分为两个部分：防火墙规则导致服务无法正常访问 和 Qbittorrent 管理页面重定向问题。我们分别分析和解决这两个问题。

问题一：防火墙规则导致服务无法正常访问

从描述来看，你在防火墙中设置了 WAN 入站数据丢弃，但针对需要的服务添加了通信规则，仍然无法访问特定服务，例如 RustDesk 和 Aria2。

1. 检查防火墙通信规则的优先级
防火墙中的通信规则优先级可能是问题的根源：

确保你针对 WAN 入站的允许规则的优先级高于默认的丢弃规则。
检查设置时，确保规则绑定的区域是 WAN，而非 LAN 或其他错误的区域。
操作指令：

iptables-legacy -L INPUT -v -n
iptables-legacy -L FORWARD -v -n
重点检查以下内容：

是否有针对 RustDesk、Aria2 的 ACCEPT 规则；
这些规则是否位于默认的 DROP 规则之前。
2. 确保端口转发配置正确
即使通信规则设置无误，端口转发也必须正确设置到 OpenWrt LAN 的对应 IP 和端口。 检查转发规则：

iptables-legacy -t nat -L PREROUTING -v -n
确认：

外部端口（WAN）是否正确映射到内网的 RustDesk 和 Aria2 的服务端口；
确保端口转发的目标是 OpenWrt Docker 容器中的正确 IP 地址和端口，而不是主机 IP。
3. 添加相关模块的信任规则
某些服务（如 RustDesk 和 FTP）可能需要特定协议（如 UDP）支持：

确保 RustDesk 所需的 TCP 和 UDP 端口均已放行；
FTP 的数据端口范围是否正确配置并放行。
例如，假设 RustDesk 需要 21115 端口：

iptables-legacy -A INPUT -p tcp --dport 21115 -j ACCEPT
iptables-legacy -A INPUT -p udp --dport 21115 -j ACCEPT
4. 使用 防火墙区域转发
相比单独放行每个端口，更推荐通过防火墙区域设置受控的转发规则：

创建一个新的防火墙区域，例如 docker_zone，绑定 Docker 的网桥。
允许 WAN 到 docker_zone 的特定端口转发：
在防火墙中设置 docker_zone 的规则，指定开放的端口范围，而不是直接丢弃或接受所有端口。
问题二：Qbittorrent 管理页面重定向问题

从描述来看，问题是 Qbittorrent 的管理页面在使用插件时，生成了错误的 URL（如 域名:99:100），而其他服务没有类似问题。这可能与 Qbittorrent 的外部地址配置或插件跳转逻辑相关。

1. 修改 Qbittorrent 的 WebUI 外部地址
Qbittorrent 默认生成的 URL 可能无法正确适应端口转发的规则，你可以尝试通过配置文件修改外部地址：

编辑 Qbittorrent 的配置文件 qBittorrent.conf：
vi ~/.config/qBittorrent/qBittorrent.conf
确保以下参数设置正确：
WebUI\HostHeaderValidation=false
WebUI\ServerDomains=你的域名
重启 Qbittorrent 服务：
systemctl restart qbittorrent
2. 手动指定插件的重定向地址
如果 Qbittorrent 的插件无法正确生成地址，可以尝试手动配置插件的重定向 URL。

确认插件的配置文件是否支持自定义 WebUI 地址；
如果无法修改插件配置，你可以在浏览器中通过开发工具临时修改生成的 URL。
3. Nginx 或 Caddy 代理解决问题
推荐使用反向代理来隐藏端口转发逻辑：

设置 Nginx 或 Caddy 为反向代理，接管所有服务的流量。
通过反向代理，为 Qbittorrent 分配一个子路径（如 https://你的域名/qbittorrent），避免因端口混淆导致问题。
Nginx 配置示例：

server {
    listen 80;
    server_name 你的域名;

    location /qbittorrent/ {
        proxy_pass http://127.0.0.1:100/;
        proxy_set_header Host $host;
        proxy_set_header X-Real-IP $remote_addr;
        proxy_set_header X-Forwarded-For $proxy_add_x_forwarded_for;
    }
}
最终建议

优化防火墙规则：
使用防火墙区域转发替代全端口接受；
确认规则优先级和端口映射正确。
修正 Qbittorrent 的 WebUI 跳转问题：
修改 Qbittorrent 的外部地址配置；
使用 Nginx 或 Caddy 反向代理简化域名和端口配置。
如果仍然存在问题，可以提供更详细的 iptables-legacy 输出和防火墙配置文件内容，我可以帮你进一步优化规则。