istoreos/istoreos #2082: 无法拉取docker镜像，即使把好的镜像打包传到这里进行导入，也无法创建容器

flyingtt29, 2025-01-24:
1：刚刚从官网下载安装的系统，在N1盒子上。这是从服务菜单里安装navidrome的报错信息。istoreos增强已经安了，添加了镜像代理地址了
dockerpull.cn/difegue/navidrome
docker pull dockerpull.cn/difegue/navidrome
Using default tag: latest
Error response from daemon: Get "https://dockerpull.cn/v2/": dial tcp: lookup dockerpull.cn on [::1]:53: server misbehaving
Error: No such container: navidrome
docker run --restart=unless-stopped -d     -v "/etc/navidrome:/data"     -v "/mnt/samba/3T软件BAK-备份/1000mp3:/music:ro"     --dns=172.17.0.1     -p 4533:4533  -e TZ=Asia/Shanghai -v /mnt:/mnt:rslave --name navidrome "dockerpull.cn/difegue/navidrome"
Unable to find image 'dockerpull.cn/difegue/navidrome:latest' locally
docker: Error response from daemon: Get "https://dockerpull.cn/v2/": dial tcp: lookup dockerpull.cn on [::1]:53: server misbehaving.
See 'docker run --help'.

2：我从其他amlogic的S905m的电视盒子安的armbian上用docker加镜像代理地址成功下载了该镜像并成功运行。然后打包保存该镜像到计算机，然后用istoreOS的docker镜像导入功能成功导入。但是再用创建容器的页面，无论是手工设置各个参数项，还是通过粘贴命令行自动识别，都创建失败。
这行命令在我的armbian盒子正确运行，但用同样参数在页面创建容器就失败报错。
docker run  -d -v /root/navidrome:/data -v /root/music:/music -p 4534:4533  --name navidrome deluan/navidrome
我ssh到机器上命令行运行也会报如下错误信息：
root@iStoreOS:~# docker run  -d -v /root/navidrome:/data -v /root/music:/music -p 4534:4533  --name navidrome deluan/navidrome
		docker: Error response from daemon: No command specified.

King-hsw, 2025-01-24:
同问，我也出现当前问题

[image]
istoreos增强已经安装

yangshuang9889, 2025-02-05:
我的问题差不多         不过是安装1Panel

betsylotus, 2025-02-08:
问题解决了吗

xcy960815, 2025-02-08:
有同样的问题

xcy960815, 2025-02-15:
> 有同样的问题

现在升级Istore增强 拉取镜像已经没问题了