istoreos/istoreos #2113: istoreos里istore中安装immich出问题

HCQM, 2025-02-20:
istore点击安装显示成功，点击打开，显示
404 未找到
对不起，请求的目标未找到。

No page is registered at '/admin/services/immich'. If this url belongs to an extension, make sure it is properly installed. If the extension was recently installed, try removing the /tmp/luci-indexcache file.

无法调度: /cgi-bin/luci/admin/services/immich

尝试英文中的操作还是失败。

iStoreOS
型号
Default string Default string
架构
Intel(R) Pentium(R) CPU J3710 @ 1.60GHz
目标平台
x86/64
固件版本
iStoreOS 22.03.7 2024122712 / LuCI istoreos-22.03 branch git-24.339.46321-5f9267c
内核版本
5.10.221

b0070, 2025-02-24:
对不起，请求的目标未找到。

No page is registered at '/admin/services/immich'. If this url belongs to an extension, make sure it is properly installed. If the extension was recently installed, try removing the /tmp/luci-indexcache file.

无法调度: /cgi-bin/luci/admin/services/immich

出现同样的问题

jacobsenxin, 2025-03-08:
同遇到这个问题，请问知道怎么解决吗

forhalunhaku, 2025-03-08:
同样的问题

b0070, 2025-03-11:
据说已经在修复中

Fly6688Fei, 2025-03-20:
Repaired

closed 2025-03-20 by jjm2473