istoreos/istoreos #2114: istore 安装 immich相册 遇阻

sunstone2023, 2025-02-20:
Try mirror server https://repo.istoreos.com/repo
Installing app-meta-immich (1.0.0-2) to root...
.
.
.
istore (完成于： 2025/2/20 21:27:38) > is-opkg install 'app-meta-immich'

完成后，打开.......
就出现了：

404 未找到
对不起，请求的目标未找到。

No page is registered at '/admin/services/immich'. If this url belongs to an extension, make sure it is properly installed. If the extension was recently installed, try removing the /tmp/luci-indexcache file.

无法调度: /cgi-bin/luci/admin/services/immich

b0070, 2025-02-24:
对不起，请求的目标未找到。

No page is registered at '/admin/services/immich'. If this url belongs to an extension, make sure it is properly installed. If the extension was recently installed, try removing the /tmp/luci-indexcache file.

无法调度: /cgi-bin/luci/admin/services/immich

出现同样的问题

ni2hao4, 2025-03-03:
我也是这个问题

makalulaha, 2025-03-08:
我也是这样，路由器是NanoPi R4S

closed 2025-03-20 by jjm2473