istoreos/istoreos #2116: Realtek RTL8125BG大流量下会断连

Accelerator-Li, 2025-02-22:
## 描述

iStoreOS 22.03.7 20241227

1. Realtek RTL8125BG直通给iStoreOS，8125BG和虚拟网卡vmxnet3在br-lan下。
2. 两台正常的windows电脑通过iperf3测速
3. 8125BG大流量会断连，发生断连时间不定，断连恢复大概在5秒到1分钟。（br-lan里有两个口，只有8125BG断连了）
4. 如果断连时间较短，看起来像丢包，如果断连时间较长，看起来像拔网线。
5. 已排除网线质量问题。可能是驱动问题？或者还需要调整虚拟机高级设置？
6. 一开始使用`iStoreOS 22.03.7 20240802`这个版本，发现新版本`iStoreOS 22.03.7 20241206`更新了r8125驱动，于是直接在线更新到了最新版本`iStoreOS 22.03.7 20241227`，更新到了最新版本之后问题照旧。
7. 在 `系统` > `杂项设置` > `驱动` 里尝试测试了开关`Realtek r8125 驱动`两种情况，还是断连。详细日志见下文
8. iperf3执行记录
```log
F:\Program Files\iperf>iperf3 -c 192.168.1.17 -t 120
Connecting to host 192.168.1.17, port 5201
[  4] local 192.168.1.16 port 54785 connected to 192.168.1.17 port 5201
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-1.00   sec   113 MBytes   944 Mbits/sec
[  4]   1.00-2.00   sec   112 MBytes   943 Mbits/sec
[  4]   2.00-3.00   sec   106 MBytes   893 Mbits/sec
[  4]   3.00-4.00   sec   106 MBytes   894 Mbits/sec
[  4]   4.00-5.00   sec   108 MBytes   905 Mbits/sec
[  4]   5.00-6.00   sec   107 MBytes   894 Mbits/sec
[  4]   6.00-7.00   sec   106 MBytes   892 Mbits/sec
[  4]   7.00-8.00   sec   107 MBytes   895 Mbits/sec
[  4]   8.00-9.00   sec   107 MBytes   894 Mbits/sec
[  4]   9.00-10.00  sec   107 MBytes   896 Mbits/sec
[  4]  10.00-11.00  sec   106 MBytes   893 Mbits/sec
[  4]  11.00-12.00  sec   107 MBytes   895 Mbits/sec
[  4]  12.00-13.00  sec   106 MBytes   894 Mbits/sec
[  4]  13.00-14.00  sec   107 MBytes   895 Mbits/sec
[  4]  14.00-15.00  sec   106 MBytes   893 Mbits/sec
[  4]  15.00-16.00  sec   107 MBytes   894 Mbits/sec
[  4]  16.00-17.00  sec   107 MBytes   894 Mbits/sec
[  4]  17.00-18.00  sec   107 MBytes   895 Mbits/sec
[  4]  18.00-19.00  sec   107 MBytes   899 Mbits/sec
[  4]  19.00-20.00  sec   107 MBytes   895 Mbits/sec
[  4]  20.00-21.00  sec   106 MBytes   891 Mbits/sec
[  4]  21.00-22.00  sec   106 MBytes   893 Mbits/sec
[  4]  22.00-23.00  sec   107 MBytes   895 Mbits/sec
[  4]  23.00-24.00  sec   106 MBytes   890 Mbits/sec
[  4]  24.00-25.00  sec   107 MBytes   896 Mbits/sec
[  4]  25.00-26.00  sec   106 MBytes   891 Mbits/sec
[  4]  26.00-27.00  sec   107 MBytes   896 Mbits/sec
[  4]  27.00-28.00  sec  17.5 MBytes   147 Mbits/sec
[  4]  28.00-29.00  sec  0.00 Bytes  0.00 bits/sec
[  4]  29.00-30.00  sec  0.00 Bytes  0.00 bits/sec
[  4]  30.00-31.00  sec  0.00 Bytes  0.00 bits/sec
[  4]  31.00-32.00  sec  0.00 Bytes  0.00 bits/sec
[  4]  32.00-33.00  sec  0.00 Bytes  0.00 bits/sec
[  4]  33.00-34.00  sec  3.75 MBytes  31.5 Mbits/sec
[  4]  34.00-35.00  sec   108 MBytes   905 Mbits/sec
[  4]  35.00-36.00  sec   112 MBytes   936 Mbits/sec
[  4]  36.00-37.00  sec   106 MBytes   890 Mbits/sec
```

```log
F:\Program Files\iperf>iperf3 -c 192.168.1.17 -t 120
Connecting to host 192.168.1.17, port 5201
[  4] local 192.168.1.16 port 56785 connected to 192.168.1.17 port 5201
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-1.00   sec   112 MBytes   941 Mbits/sec
[  4]   1.00-2.00   sec   108 MBytes   902 Mbits/sec
[  4]   2.00-3.00   sec   106 MBytes   889 Mbits/sec
[  4]   3.00-4.00   sec   106 MBytes   889 Mbits/sec
[  4]   4.00-5.00   sec   108 MBytes   905 Mbits/sec
[  4]   5.00-6.00   sec   106 MBytes   894 Mbits/sec
[  4]   6.00-7.00   sec   106 MBytes   892 Mbits/sec
[  4]   7.00-8.00   sec   107 MBytes   895 Mbits/sec
[  4]   8.00-9.00   sec   106 MBytes   893 Mbits/sec
[... 208 lines cut ...]
[ 3312.551355]  </IRQ>
[ 3312.551356]  do_softirq_own_stack+0x3c/0x50
[ 3312.551358]  irq_exit_rcu+0x5f/0xa0
[ 3312.551358]  sysvec_apic_timer_interrupt+0x3d/0x90
[ 3312.551359]  asm_sysvec_apic_timer_interrupt+0x12/0x20
[ 3312.551360] RIP: 0010:native_safe_halt+0x13/0x20
[ 3312.551361] Code: 07 0f 00 2d df 60 4a 00 f4 c3 cc cc cc cc 66 0f 1f 84 00 00 00 00 00 8b 05 9e fd b4 00 85 c0 7e 07 0f 00 2d bf 60 4a 00 fb f4 <c3> cc cc cc cc cc cc cc cc cc cc cc cc 55 48 89 e5 53 e8 86 44 7b
[ 3312.551361] RSP: 0018:ffffc90000087e48 EFLAGS: 00000246
[ 3312.551362] RAX: 0000000000000000 RBX: ffff888001901800 RCX: ffff8880bcd26140
[ 3312.551362] RDX: ffff8880bcd00000 RSI: ffff888001b5b000 RDI: ffff888001b5b064
[ 3312.551362] RBP: ffffc90000087e50 R08: ffff888001901800 R09: 0000000000000126
[ 3312.551363] R10: 0000000000000000 R11: 000000000000011f R12: 0000000000000001
[ 3312.551363] R13: ffff888001b5b064 R14: ffffffff822cd0c0 R15: ffffffff822cd0a8
[ 3312.551364]  ? acpi_idle_do_entry+0x4e/0x70
[ 3312.551365]  acpi_idle_enter+0x8a/0xe0
[ 3312.551366]  cpuidle_enter_state+0x7e/0x2d0
[ 3312.551367]  cpuidle_enter+0x2f/0x50
[ 3312.551368]  do_idle+0x178/0x1c0
[ 3312.551369]  cpu_startup_entry+0x1b/0x20
[ 3312.551370]  start_secondary+0xf7/0x110
[ 3312.551371]  secondary_startup_64_no_verify+0xac/0xbb
[ 3312.551372] ---[ end trace 1faa20b01c3479e6 ]---
[ 3312.551374] r8125 0000:02:01.0 eth2: Transmit timeout reset Device!
[ 3312.611149] r8125 0000:02:01.0 eth2: Device reseting!
```

cnwikee, 2025-05-24:
我也遇到这个问题，我是用来接ITV口看IPTV的，看电视时，时不时的就提示网络未连接，然后等一下自动连上。然后看了系统日志，是eth2 网口掉了，不知该如何解决：

网卡信息
```
root@iStoreOS:~# ethtool -i eth2
driver: r8152
version: v2.17.1 (2023/06/13)
firmware-version: 
expansion-rom-version: 
bus-info: usb-0000:00:14.0-4
supports-statistics: yes
supports-test: no
supports-eeprom-access: no
supports-register-dump: no
supports-priv-flags: no
root@iStoreOS:~# 
```

系统日志：
```
Sat May 24 19:29:01 2025 daemon.notice pppd[5437]: Modem hangup
Sat May 24 19:29:01 2025 daemon.info avahi-daemon[8586]: Interface eth2.IPv6 no longer relevant for mDNS.
Sat May 24 19:29:01 2025 daemon.info avahi-daemon[8586]: Leaving mDNS multicast group on interface eth2.IPv6 with address fe80::2e0:4cff:fe68:b2.
Sat May 24 19:29:01 2025 daemon.notice netifd: Network device 'eth2' link is down
Sat May 24 19:29:01 2025 daemon.notice netifd: Interface 'IPTV' has link connectivity loss
Sat May 24 19:29:01 2025 daemon.notice ttyd[17421]: rops_handle_POLLIN_netlink: DELADDR
Sat May 24 19:29:01 2025 daemon.info pppd[5437]: Connect time 18.3 minutes.
Sat May 24 19:29:01 2025 daemon.info pppd[5437]: Sent 2446 bytes, received 0 bytes.
Sat May 24 19:29:01 2025 daemon.info avahi-daemon[8586]: Withdrawing address record for fe80::2e0:4cff:fe68:b2 on eth2.
Sat May 24 19:29:01 2025 daemon.info avahi-daemon[8586]: Interface pppoe-IPTV.IPv4 no longer relevant for mDNS.
Sat May 24 19:29:01 2025 daemon.info avahi-daemon[8586]: Leaving mDNS multicast group on interface pppoe-IPTV.IPv4 with address 10.168.141.109.
Sat May 24 19:29:01 2025 daemon.notice netifd: Network device 'pppoe-IPTV' link is down
Sat May 24 19:29:01 2025 kern.info kernel: [34601.379171] r8152-cfgselector 2-4: USB disconnect, device number 8
Sat May 24 19:29:01 2025 daemon.info avahi-daemon[8586]: Withdrawing address record for 10.168.141.109 on pppoe-IPTV.
Sat May 24 19:29:01 2025 daemon.notice ttyd[17421]: rops_handle_POLLIN_netlink: DELADDR
Sat May 24 19:29:01 2025 daemon.notice pppd[5437]: Connection terminated.
Sat May 24 19:29:01 2025 daemon.info ModemManager[9688]: hotplug: remove network interface eth2: event processed
Sat May 24 19:29:01 2025 daemon.debug ModemManager[9688]: hotplug: event reported: action=remove, name=eth2, subsystem=net
Sat May 24 19:29:01 2025 daemon.info ModemManager[9716]: hotplug: remove network interface pppoe-IPTV: event processed
Sat May 24 19:29:01 2025 daemon.debug ModemManager[9716]: hotplug: event reported: action=remove, name=pppoe-IPTV, subsystem=net
Sat May 24 19:29:01 2025 daemon.info pppd[5437]: Exit.
Sat May 24 19:29:01 2025 daemon.notice netifd: Interface 'IPTV' is now down
Sat May 24 19:29:01 2025 daemon.info dnsmasq[1]: reading /tmp/resolv.conf.d/resolv.conf.auto
Sat May 24 19:29:01 2025 daemon.info dnsmasq[1]: using nameserver 202.96.128.86#53
Sat May 24 19:29:01 2025 daemon.info dnsmasq[1]: using nameserver 202.96.134.33#53
Sat May 24 19:29:01 2025 daemon.info dnsmasq[1]: using nameserver 240e:1f:1::1#53
Sat May 24 19:29:01 2025 daemon.info dnsmasq[1]: using nameserver 240e:1f:1::33#53
Sat May 24 19:29:01 2025 daemon.info dnsmasq[1]: using only locally-known addresses for test
Sat May 24 19:29:01 2025 daemon.info dnsmasq[1]: using only locally-known addresses for onion
Sat May 24 19:29:01 2025 daemon.info dnsmasq[1]: using only locally-known addresses for localhost
Sat May 24 19:29:01 2025 daemon.info dnsmasq[1]: using only locally-known addresses for local
Sat May 24 19:29:01 2025 daemon.info dnsmasq[1]: using only locally-known addresses for invalid
Sat May 24 19:29:01 2025 daemon.info dnsmasq[1]: using only locally-known addresses for bind
Sat May 24 19:29:01 2025 daemon.info dnsmasq[1]: using only locally-known addresses for lan
Sat May 24 19:29:01 2025 daemon.notice netifd: Interface 'IPTV' is disabled
Sat May 24 19:29:01 2025 kern.info kernel: [34601.909309] usb 2-4: new SuperSpeed Gen 1 USB device number 9 using xhci_hcd
Sat May 24 19:29:01 2025 daemon.notice netifd: Interface 'IPTV' is enabled
Sat May 24 19:29:01 2025 kern.info kernel: [34601.960018] cdc_ncm 2-4:2.0: MAC-Address: 00:e0:4c:68:00:b2
Sat May 24 19:29:01 2025 kern.info kernel: [34601.960032] cdc_ncm 2-4:2.0: setting rx_max = 16384
Sat May 24 19:29:01 2025 kern.info kernel: [34601.960111] cdc_ncm 2-4:2.0: setting tx_max = 16384
Sat May 24 19:29:01 2025 kern.info kernel: [34601.960976] cdc_ncm 2-4:2.0 eth2: register 'cdc_ncm' at usb-0000:00:14.0-4, CDC NCM, 00:e0:4c:68:00:b2
Sat May 24 19:29:01 2025 kern.info kernel: [34601.962982] cdc_ncm 2-4:2.0 eth2: unregister 'cdc_ncm' usb-0000:00:14.0-4, CDC NCM
Sat May 24 19:29:01 2025 daemon.notice netifd: Interface 'IPTV' is disabled
Sat May 24 19:29:01 2025 kern.info kernel: [34602.179754] r8152-cfgselector 2-4: reset SuperSpeed Gen 1 USB device number 9 using xhci_hcd
Sat May 24 19:29:01 2025 daemon.info ModemManager[10216]: hotplug: add network interface eth2: event processed
Sat May 24 19:29:01 2025 daemon.debug ModemManager[10216]: hotplug: event reported: action=add, name=eth2, subsystem=net
Sat May 24 19:29:01 2025 daemon.info ModemManager[10216]: hotplug: no need to wait for modem at sysfs path /sys/devices/pci0000:00/0000:00:14.0/usb2/2-4
Sat May 24 19:29:01 2025 daemon.warn [8726]: <warn>  [eth2] invalid sysfs path read for net/eth2
Sat May 24 19:29:01 2025 daemon.warn [8726]: <warn>  [base-manager] couldn't handle kernel event: device net/eth2 not found
Sat May 24 19:29:02 2025 daemon.notice netifd: Interface 'IPTV' is enabled
[... 173 lines cut ...]
[... cut ...]
Sat May 24 19:30:27 2025 daemon.warn dnsmasq[1]: possible DNS-rebind attack detected: dns.msftncsi.com
Sat May 24 19:31:12 2025 daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(br-lan) 192.168.1.110 3c:e9:f7:e5:84:95
Sat May 24 19:31:12 2025 daemon.info dnsmasq-dhcp[1]: DHCPACK(br-lan) 192.168.1.110 3c:e9:f7:e5:84:95 DESKTOP-C8L4AQH
Sat May 24 19:32:14 2025 daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(br-lan) 192.168.1.110 3c:e9:f7:e5:84:95
Sat May 24 19:32:14 2025 daemon.info dnsmasq-dhcp[1]: DHCPACK(br-lan) 192.168.1.110 3c:e9:f7:e5:84:95 DESKTOP-C8L4AQH
```

cnwikee, 2025-05-27:
> 我又测试了一下 使用相同版本的windows、ubuntu，驱动版本也相同，分别在物理机和虚拟机测试，物理机一切正常，虚拟机还是异常断流。 感觉可能是驱动对虚拟化环境兼容性不好

我退掉了我的USB3.0 2.5G 8156b 芯片的网卡了，老是断线，掉线。然后买了一个更便宜的ax81179芯片的USB3.0网卡，一整天了，没掉过线。日志看了没有问题了。
```
root@iStoreOS:~# ethtool -i eth2
driver: ax88179_178a
version: 5.10.221
firmware-version: 
expansion-rom-version: 
bus-info: 2-3:1.0
supports-statistics: no
supports-test: no
supports-eeprom-access: yes
supports-register-dump: no
supports-priv-flags: no
```