istoreos/istoreos #2130: x86机器每过几天就会死机一次，必须断电重启才正常

xiaoruistc, 2025-02-28:
主机名 | iStoreOS
型号 | Default string Default string
架构 | Intel(R) N100
目标平台 | x86/64
固件版本 | iStoreOS 22.03.7 2024122712 / LuCI istoreos-22.03 branch git-24.339.46321-5f9267c
内核版本 | 5.10.221
本地时间 | 2025-02-28 10:12:10
运行时间 | 1h 13m 19s
平均负载 | 1.77, 1.52, 1.38

死机后显示如下：

[10319.626412] i915 0000:00:02.0: [drm] *ERROR* AUX USBC1/DDI TC1/PHY TC1: did not complete or timeout within 10ms (status 0xad4003ff)
[10319.666405] i915 0000:00:02.0: [drm] *ERROR* AUX USBC1/DDI TC1/PHY TC1: did not complete or timeout within 10ms (status 0xad4003ff)
[10319.706402] i915 0000:00:02.0: [drm] *ERROR* AUX USBC1/DDI TC1/PHY TC1: did not complete or timeout within 10ms (status 0xad4003ff)
[10319.746400] i915 0000:00:02.0: [drm] *ERROR* AUX USBC1/DDI TC1/PHY TC1: did not complete or timeout within 10ms (status 0xad4003ff)
[10319.786394] i915 0000:00:02.0: [drm] *ERROR* AUX USBC1/DDI TC1/PHY TC1: did not complete or timeout within 10ms (status 0xad4003ff)
[10319.786396] i915 0000:00:02.0: [drm] *ERROR* AUX USBC1/DDI TC1/PHY TC1: not done (status 0xad4003ff)
[10351.654007] i915 0000:00:02.0: [drm] *ERROR* AUX USBC1/DDI TC1/PHY TC1: did not complete or timeout within 10ms (status 0xad4003ff)
[10351.684000] i915 0000:00:02.0: [drm] *ERROR* AUX USBC1/DDI TC1/PHY TC1: did not complete or timeout within 10ms (status 0xad4003ff)
[10351.713996] i915 0000:00:02.0: [drm] *ERROR* AUX USBC1/DDI TC1/PHY TC1: did not complete or timeout within 10ms (status 0xad4003ff)
[10351.743994] i915 0000:00:02.0: [drm] *ERROR* AUX USBC1/DDI TC1/PHY TC1: did not complete or timeout within 10ms (status 0xad4003ff)
[10351.773991] i915 0000:00:02.0: [drm] *ERROR* AUX USBC1/DDI TC1/PHY TC1: did not complete or timeout within 10ms (status 0xad4003ff)
[10351.773993] i915 0000:00:02.0: [drm] *ERROR* AUX USBC1/DDI TC1/PHY TC1: not done (status 0xad4003ff)

jjm2473, 2025-03-17:
这个报错只是因为没接hdmi，应该不会导致死机。试试接上hdmi。

xiaoruistc, 2025-03-20:
> 这个报错只是因为没接hdmi，应该不会导致死机。试试接上hdmi。

做软路由使用  也不接显示器啊

jjm2473, 2025-06-10:
我是说那个日志跟死机无关，接显示器就没那个日志了。

现在24.10换了i915-sriov驱动，应该不接显示器也不会有那个日志。