istoreos/istoreos #2131: 被网安扫了，发现一个漏洞并判定为高危。。。请求修复，ths

braxiong, 2025-02-28:
访问URL：http://路由器IP/cgi-bin/luci//"><iframe src=javascript:alert`test`

直接访问上述漏洞URL链接即可实现漏洞弹窗效果。

被网安判断为高危漏洞，请求修复

整改建议
（1） 对用户输入进行严格的验证，确保不接受任何可能导致脚本执行的输入。具体方法包括：
数据类型检查：验证输入数据的类型，如数字、字符串等。
长度限制：限制输入数据的长度。
正则表达式：使用正则表达式过滤掉潜在的恶意字符。
（2）对输出到HTML页面的数据进行编码，以防止恶意脚本在用户浏览器中执行。以下是几种常见的编码方法：
HTML实体编码：将特殊字符转换为HTML实体，如<转换为&lt;，>转换为&gt;。
JavaScript编码：对JavaScript中的特殊字符进行编码。
（3）使用已经过安全审查的API和库来处理HTML和JavaScript，如：
DOMPurify：一个DOM-only XSS清理库，用于清理HTML和SVG文档。
OWASP AntiSamy：一个开源的Java库，用于防止XSS攻击。
（4）设置适当的HTTP响应头，如：
Content-Security-Policy (CSP)：限制资源的加载和执行，防止XSS攻击。

tjbzyl1212, 2025-03-08:
我也被扫了。还打电话来了

jjm2473, 2025-03-17:
准确的测试url是
```
http://192.168.100.1/cgi-bin/luci//"><iframe src="javascript:alert('test')
```
或者转译版本
```
http://192.168.31.1/cgi-bin/luci//%22%3E%3Ciframe%20src=%22javascript:alert('test')
```

jjm2473, 2025-03-17:
这个问题可能是主题的bug

jjm2473, 2025-03-17:
如果是要临时应付检查，可以切换到Bootstrap主题

[image]

braxiong, 2025-03-17:
> 如果是要临时应付检查，可以切换到Bootstrap主题
> 
> <img alt="Image" width="870" src="https://private-user-images.githubusercontent.com/3896703/423385764-85086cd0-cdb0-4b3f-bd9b-458d2d2890aa.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NDIyMDExMDYsIm5iZiI6MTc0MjIwMDgwNiwicGF0aCI6Ii8zODk2NzAzLzQyMzM4NTc2NC04NTA4NmNkMC1jZGIwLTRiM2YtYmQ5Yi00NThkMmQyODkwYWEucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI1MDMxNyUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNTAzMTdUMDg0MDA2WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9NTIyYWQwM2Q1MjY3NmVhMDAyN2UyMWUxMGRmZDhlNzhiMmI5N2IyZGU3NjlkZjM2ZDY4MzczNjcxZmNlZmEzOCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.Qb-MHNRxFEUDwu2lfKagz951m4Phwcs-c0HATbgfabQ">

收到，感谢

jjm2473, 2025-03-24:
这个bug已经修复，之后发布的固件就没问题了。

closed 2025-03-24 by braxiong