istoreos/istoreos #2133: 容器内部无法连接互联网，求解决

wudingjian, 2025-03-01:
型号 | FastRhino R68s
架构 | ARMv8 Processor rev 0
目标平台 | rockchip/rk35xx
固件版本 | iStoreOS 22.03.7 2024122712 / LuCI istoreos-22.03 branch git-24.339.46321-5f9267c
内核版本 | 5.10.221

拨号上网正常后，默认状态下安装docker 如下，容器起来后，容器内部无法连接外网，容器内可以ping通主机ip，但无法ping通外网。

举例容器官网：https://github.com/Guovin/iptv-api

docker-compose

version: "3.1"
  iptv-api:
    image: guovern/iptv-api
    container_name: iptv-api
    restart: unless-stopped  
    ports:
      - "8010:8000"
    volumes:
      - ./config:/iptv-api/config
      - ./output:/iptv-api/output

运行后，日志显示无法连接外网。

138161, 2025-03-04:
DDNS-GO获取外网ip失败？docker是在路由器wan后面，docker的wan侧分配的是路由器的内网ip地址，他没有直接获得路由器wan侧ip的能力

wudingjian, 2025-03-06:
> DDNS-GO获取外网ip失败？docker是在路由器lan后面，docker的wan侧分配的是路由器的内网ip地址，他没有直接获得路由器wan侧ip的能力

 DDNS-GO获取 IP 方式，采用通过接口获取（https://myip.ipip.net, https://ddns.oray.com/checkip, https://ip.3322.net, https://4.ipw.cn, https://v4.yinghualuo.cn/bejson），非网卡或网桥获取。

我只是用 DDNS-GO来举例说明，容器内部无法访问互联网

jjm2473, 2025-03-17:
默认情况下只有docker0的容器才能上网，也就是172.17.0.0/16这个网段的才是开箱即用的。docker compose可能会自己创建一些网络，一般不会使用docker0