istoreos/istoreos #2136: istoreos是否支持NVMe M2的固态硬盘，并且系统装在固态上？

shixin-679, 2025-03-02:
我现在准备上一个畅网N150的小主机，提供2个NVMe M2接口，如果我想把系统装在固态硬盘中是否支持呢？

[image]

[image]

[image]

[image]

以前我的友善R5S小主机也提供一个M2接口，但是我插上M2固态硬盘后，系统识别不到，按照大佬的说法更新后也不行，提过issue：https://github.com/istoreos/istoreos/issues/1677

担心系统没法装在M2固态上，希望大佬解答一下，谢谢！

lizhi10, 2025-03-02:
x86可以，arm的基本不可以。因为arm的大部分机器都不支持uefi，而且官方也没有发布arm uefi的版本。希望官方可以出一个带uefi的arm版，我有一个支持uefi的arm机器

shixin-679, 2025-03-03:
> x86可以，arm的基本不可以。因为arm的大部分机器都不支持uefi，而且官方也没有发布arm uefi的版本。希望官方可以出一个带uefi的arm版本，我有一个支持uefi的arm机器

哦哦 原来是这样哦，那就放心了

jjm2473, 2025-03-17:
x86一般都可以的，没遇到nvme无法启动的情况

jjm2473, 2025-03-17:
@lizhi10 有个armsr的测试固件，24.10，在mac mini m4的虚拟机上测试过，下载地址 https://fw.koolcenter.com/iStoreOS/alpha/armsr/

closed 2025-03-17 by jjm2473