istoreos/istoreos #2153: N1外挂硬盘盒插一条m.2固态 100%掉盘，插U盘就不会掉，是供电问题吗？

simendancer, 2025-03-10:
新人刚接触软路由这块，买的斐讯N1也是闲鱼上淘的，装的是原版**iStoreOS 22.03.6 2024043010**

然后想捣腾下docker和家庭共享空间，最开始插了一个闪迪32g的U盘，用了几天感觉还挺稳，就是速度慢了点，而且空间不够大

接着就买了一条致态512g的m.2固态，装在一个itgz的硬盘盒里，替换掉U盘**直接插在机子的后面**

但是发现插上去一段时间，可能一小时都不到就会掉盘，然后在磁盘管理中对这个固态进行任何操作都会显示targer is busy，读写文件也是卡死状态，最后只能硬拔（我只开启了smb和易有云服务，并没有什么读写操作）

用AI问了一下，说是可能SSD有问题，又在电脑用DiskG扫描了一次，发现有一个坏快，修复以后格式化ext4，再重复上面操作，发现还是会掉盘

接着又用DiskG扫描，这时又会出现一个坏快，但是我SSD是新的，我在电脑上长时间用时候完全是正常的

大概上面的流程重复了3、4次，一直还是掉盘的问题或者读写卡死的问题，后来感觉不是SSD问题，有去网上找，说是**供电的问题**
N1是USB2.0，固态盒子是USB3.0

1. 找了一个手机用的**5v1a**的插头，给USB扩展坞供电，固态盒子插在扩展坞上，扩展坞插在N1后面，发现还是一样问题

2. 在恩山论坛又找了一个办法，说是找个USB公对公的线一头插N1后面靠近hdmi的那个usb口，另一台插电源**5v2a**供电，然后固态盒子插在N1另一个usb口上，发现还是一样的问题

3. 又网上找了找，说电流不够，又尝试把**5v1a**的换成**5v2a**的插头，给USB扩展坞供电，固态盒子插在扩展坞上，扩展坞插在N1后面，这次没掉盘的时间比较长，但是最后还是掉了

所以，这是不是真的就是供电问题？我看恩山上还有大哥说给N1换个台达12V3A的电源，可以挂2个硬盘都不掉。。。

**还有就是，N1官方固件是不是只有现在这个2024043010版本可以刷，更新日志里那些没有标注型号的，是通用能刷的吗？**

下面是系统日志里的关于固态挂载的记录和挂载的截图，是什么问题？

Tue Mar 11 01:39:59 2025 kern.info kernel: [473689.381256] usb 1-2.3: reset high-speed USB device number 2 using xhci-hcd
Tue Mar 11 01:40:05 2025 kern.err kernel: [473694.550228] usb 1-2.3: device descriptor read/64, error -110
Tue Mar 11 01:40:20 2025 kern.err kernel: [473709.990535] usb 1-2.3: device descriptor read/64, error -110
Tue Mar 11 01:40:20 2025 kern.info kernel: [473710.230040] usb 1-2.3: reset high-speed USB device number 2 using xhci-hcd
Tue Mar 11 01:40:20 2025 kern.err kernel: [473710.350070] usb 1-2.3: device descriptor read/64, error -71
Tue Mar 11 01:40:21 2025 kern.err kernel: [473710.630102] usb 1-2.3: device descriptor read/64, error -71
Tue Mar 11 01:40:21 2025 kern.info kernel: [473710.880440] usb 1-2.3: reset high-speed USB device number 2 using xhci-hcd
Tue Mar 11 01:40:21 2025 kern.warn kernel: [473710.880690] usb 1-2.3: Device not responding to setup address.
Tue Mar 11 01:40:21 2025 kern.warn kernel: [473711.099991] usb 1-2.3: Device not responding to setup address.
Tue Mar 11 01:40:21 2025 kern.err kernel: [473711.320202] usb 1-2.3: device not accepting address 2, error -71
Tue Mar 11 01:40:22 2025 kern.info kernel: [473711.460031] usb 1-2.3: reset high-speed USB device number 2 using xhci-hcd
Tue Mar 11 01:40:22 2025 kern.warn kernel: [473711.460303] usb 1-2.3: Device not responding to setup address.
Tue Mar 11 01:40:22 2025 kern.warn kernel: [473711.680414] usb 1-2.3: Device not responding to setup address.
Tue Mar 11 01:40:22 2025 kern.err kernel: [473711.899898] usb 1-2.3: device not accepting address 2, error -71
Tue Mar 11 01:40:22 2025 kern.info kernel: [473711.909182] usb 1-2.3: USB disconnect, device number 2
Tue Mar 11 01:40:22 2025 kern.warn kernel: [473711.939943] print_req_error: 3 callbacks suppressed
Tue Mar 11 01:40:22 2025 kern.err kernel: [473711.939957] blk_update_request: I/O error, dev sda, sector 2296 op 0x1:(WRITE) flags 0x103000 phys_seg 1 prio class 0
Tue Mar 11 01:40:22 2025 kern.err kernel: [473711.954121] Buffer I/O error on dev sda1, logical block 31, lost async page write
Tue Mar 11 01:40:22 2025 kern.err kernel: [473711.961228] blk_update_request: I/O error, dev sda, sector 515901440 op 0x1:(WRITE) flags 0x103000 phys_seg 2 prio class 0
Tue Mar 11 01:40:22 2025 kern.err kernel: [473711.981412] Buffer I/O error on dev sda1, logical block 64487424, lost async page write
Tue Mar 11 01:40:22 2025 kern.err kernel: [473711.997571] Buffer I/O error on dev sda1, logical block 64487425, lost async page write
Tue Mar 11 01:40:22 2025 kern.err kernel: [473712.014879] blk_update_request: I/O error, dev sda, sector 515901568 op 0x1:(WRITE) flags 0x103000 phys_seg 1 prio class 0
Tue Mar 11 01:40:22 2025 kern.err kernel: [473712.027617] Buffer I/O error on dev sda1, logical block 64487440, lost async page write
Tue Mar 11 01:40:22 2025 kern.err kernel: [473712.038684] blk_update_request: I/O error, dev sda, sector 515901720 op 0x1:(WRITE) flags 0x103000 phys_seg 3 prio class 0
Tue Mar 11 01:40:22 2025 kern.err kernel: [473712.050339] Buffer I/O error on dev sda1, logical block 64487459, lost async page write
Tue Mar 11 01:40:22 2025 kern.err kernel: [473712.058916] Buffer I/O error on dev sda1, logical block 64487460, lost async page write
Tue Mar 11 01:40:22 2025 kern.err kernel: [473712.068766] Buffer I/O error on dev sda1, logical block 64487461, lost async page write
Tue Mar 11 01:40:22 2025 kern.err kernel: [473712.077535] blk_update_request: I/O error, dev sda, sector 515967432 op 0x1:(WRITE) flags 0x103000 phys_seg 1 prio class 0
Tue Mar 11 01:40:22 2025 kern.err kernel: [473712.093010] Buffer I/O error on dev sda1, logical block 64495673, lost async page write
Tue Mar 11 01:40:22 2025 kern.err kernel: [473712.111040] blk_update_request: I/O error, dev sda, sector 515967472 op 0x1:(WRITE) flags 0x103000 phys_seg 5 prio class 0
Tue Mar 11 01:40:22 2025 kern.err kernel: [473712.123001] Buffer I/O error on dev sda1, logical block 64495678, lost async page write
Tue Mar 11 01:40:22 2025 kern.err kernel: [473712.133272] Buffer I/O error on dev sda1, logical block 64495679, lost async page write
Tue Mar 11 01:40:22 2025 kern.err kernel: [473712.163827] JBD2: Error while async write back metadata bh 31.
Tue Mar 11 01:40:22 2025 kern.err kernel: [473712.163843] Aborting journal on device sda1-8.
Tue Mar 11 01:40:22 2025 kern.err kernel: [473712.177125] blk_update_request: I/O error, dev sda, sector 500172800 op 0x1:(WRITE) flags 0x800 phys_seg 1 prio class 0
Tue Mar 11 01:40:22 2025 kern.err kernel: [473712.192846] JBD2: Error -5 detected when updating journal superblock for sda1-8.
Tue Mar 11 01:40:22 2025 kern.err kernel: [473712.205195] JBD2: Error while async write back metadata bh 64487424.
Tue Mar 11 01:40:22 2025 kern.err kernel: [473712.205210] JBD2: Error while async write back metadata bh 64487425.
Tue Mar 11 01:40:22 2025 kern.err kernel: [473712.211887] JBD2: Error while async write back metadata bh 64487440.
Tue Mar 11 01:40:22 2025 kern.err kernel: [473712.217005] JBD2: Error while async write back metadata bh 64487459.
Tue Mar 11 01:40:22 2025 kern.err kernel: [473712.224757] JBD2: Error while async write back metadata bh 64487460.
Tue Mar 11 01:40:22 2025 kern.err kernel: [473712.229035] JBD2: Error while async write back metadata bh 64487461.
Tue Mar 11 01:40:22 2025 kern.err kernel: [473712.236535] JBD2: Error while async write back metadata bh 64495673.
[... 70 lines cut ...]
Mon Mar 10 21:07:43 2025 kern.err kernel: [457352.705170] Buffer I/O error on dev sdb, logical block 0, async page read
Mon Mar 10 21:07:43 2025 kern.err kernel: [457352.710265] blk_update_request: I/O error, dev sdb, sector 500172800 op 0x1:(WRITE) flags 0x800 phys_seg 1 prio class 0
Mon Mar 10 21:07:43 2025 kern.err kernel: [457352.725970] Buffer I/O error on dev sdb1, logical block 62521344, lost sync page write
Mon Mar 10 21:07:43 2025 kern.err kernel: [457352.732443] JBD2: Error -5 detected when updating journal superblock for sdb1-8.
Mon Mar 10 21:07:43 2025 kern.err kernel: [457352.735193] blk_update_request: I/O error, dev sdb, sector 2048 op 0x1:(WRITE) flags 0x800 phys_seg 1 prio class 0
Mon Mar 10 21:07:43 2025 kern.err kernel: [457352.753040] Buffer I/O error on dev sdb1, logical block 0, lost sync page write
Mon Mar 10 21:07:43 2025 kern.err kernel: [457352.759000] EXT4-fs (sdb1): I/O error while writing superblock
Mon Mar 10 21:07:43 2025 kern.crit kernel: [457352.764198] EXT4-fs error (device sdb1): ext4_journal_check_start:83: Detected aborted journal
Mon Mar 10 21:07:43 2025 kern.err kernel: [457352.764791] Buffer I/O error on dev sdb, logical block 0, async page read
Mon Mar 10 21:07:43 2025 kern.crit kernel: [457352.776297] EXT4-fs (sdb1): Remounting filesystem read-only
Mon Mar 10 21:07:43 2025 kern.emerg kernel: [457352.786661] EXT4-fs (sdb1): failed to convert unwritten extents to written extents -- potential data loss!  (inode 16121933, error -30)
Mon Mar 10 21:07:43 2025 kern.err kernel: [457352.793936] Buffer I/O error on dev sdb, logical block 0, async page read
Mon Mar 10 21:07:43 2025 kern.err kernel: [457352.802979] Buffer I/O error on device sdb1, logical block 64553993
Mon Mar 10 21:07:43 2025 kern.err kernel: [457352.807244] Buffer I/O error on dev sdb, logical block 0, async page read
Mon Mar 10 21:07:43 2025 kern.err kernel: [457352.813992] Buffer I/O error on device sdb1, logical block 64553994
Mon Mar 10 21:07:50 2025 daemon.err uhttpd[5598]: Error: Error opening /dev/sdb: No such device or address
[... cut ...] tempt to concatenate field 'phy_sec' (a nil value)
Mon Mar 10 21:07:50 2025 daemon.err uhttpd[5598]: stack traceback:
Mon Mar 10 21:07:50 2025 daemon.err uhttpd[5598]: 	/usr/lib/lua/luci/model/diskman.lua:316: in function 'get_disk_info'
Mon Mar 10 21:07:50 2025 daemon.err uhttpd[5598]: 	/usr/lib/lua/luci/controller/diskman.lua:68: in function </usr/lib/lua/luci/controller/diskman.lua:57>
Mon Mar 10 21:07:50 2025 daemon.err uhttpd[5598]: lsblk: /dev/mmcblk12: not a block device

[image]

[image]

simendancer, 2025-03-10:
基本上我只要通过smb读写一下，就马上开始报 usb 1-2.3: reset high-speed USB device number 66 using xhci-hcd
然后就卡死，掉盘

simendancer, 2025-03-10:
Tue Mar 11 03:43:01 2025 kern.info kernel: [481071.040986] usb 1-2.3: reset high-speed USB device number 66 using xhci-hcd
Tue Mar 11 03:43:06 2025 kern.err kernel: [481076.220773] usb 1-2.3: device descriptor read/64, error -110
Tue Mar 11 03:43:22 2025 kern.err kernel: [481091.590534] usb 1-2.3: device descriptor read/64, error -110
Tue Mar 11 03:43:22 2025 kern.info kernel: [481091.830747] usb 1-2.3: reset high-speed USB device number 66 using xhci-hcd
Tue Mar 11 03:43:22 2025 kern.err kernel: [481091.950961] usb 1-2.3: device descriptor read/64, error -71
Tue Mar 11 03:43:22 2025 kern.err kernel: [481092.211219] usb 1-2.3: device descriptor read/64, error -71
Tue Mar 11 03:43:23 2025 kern.info kernel: [481092.461072] usb 1-2.3: reset high-speed USB device number 66 using xhci-hcd
Tue Mar 11 03:43:23 2025 kern.warn kernel: [481092.461199] usb 1-2.3: Device not responding to setup address.
Tue Mar 11 03:43:23 2025 kern.warn kernel: [481092.680547] usb 1-2.3: Device not responding to setup address.
Tue Mar 11 03:43:23 2025 kern.err kernel: [481092.900711] usb 1-2.3: device not accepting address 66, error -71
Tue Mar 11 03:43:23 2025 kern.info kernel: [481093.030574] usb 1-2.3: reset high-speed USB device number 66 using xhci-hcd
Tue Mar 11 03:43:23 2025 kern.warn kernel: [481093.030673] usb 1-2.3: Device not responding to setup address.
Tue Mar 11 03:43:23 2025 kern.warn kernel: [481093.251028] usb 1-2.3: Device not responding to setup address.
Tue Mar 11 03:43:24 2025 kern.err kernel: [481093.470753] usb 1-2.3: device not accepting address 66, error -71
Tue Mar 11 03:43:24 2025 kern.info kernel: [481093.480212] usb 1-2.3: USB disconnect, device number 66
Tue Mar 11 03:43:24 2025 kern.err kernel: [481093.531039] blk_update_request: I/O error, dev sdb, sector 2048 op 0x1:(WRITE) flags 0x100000 phys_seg 1 prio class 0
Tue Mar 11 03:43:24 2025 kern.warn kernel: [481093.538982] buffer_io_error: 4 callbacks suppressed
Tue Mar 11 03:43:24 2025 kern.err kernel: [481093.538987] Buffer I/O error on dev sdb1, logical block 0, lost async page write
Tue Mar 11 03:43:24 2025 kern.err kernel: [481093.550194] blk_update_request: I/O error, dev sdb, sector 2056 op 0x1:(WRITE) flags 0x103000 phys_seg 1 prio class 0
Tue Mar 11 03:43:24 2025 kern.err kernel: [481093.608283] Buffer I/O error on dev sdb1, logical block 1, lost async page write
Tue Mar 11 03:43:24 2025 kern.err kernel: [481093.644169] blk_update_request: I/O error, dev sdb, sector 10728 op 0x0:(READ) flags 0x83700 phys_seg 3 prio class 0
Tue Mar 11 03:43:24 2025 kern.err kernel: [481093.653392] blk_update_request: I/O error, dev sdb, sector 10728 op 0x0:(READ) flags 0x3000 phys_seg 1 prio class 0
Tue Mar 11 03:43:24 2025 kern.crit kernel: [481093.665306] EXT4-fs error (device sdb1): ext4_wait_block_bitmap:570: comm kworker/u8:3: Cannot read block bitmap - block_group = 0, block_bitmap = 1085
Tue Mar 11 03:43:24 2025 kern.err kernel: [481093.665416] blk_update_request: I/O error, dev sdb, sector 10856 op 0x1:(WRITE) flags 0x103000 phys_seg 1 prio class 0
Tue Mar 11 03:43:24 2025 kern.err kernel: [481093.679628] EXT4-fs (sdb1): previous I/O error to superblock detected
Tue Mar 11 03:43:24 2025 kern.err kernel: [481093.692277] Buffer I/O error on dev sdb1, logical block 1101, lost async page write
Tue Mar 11 03:43:24 2025 kern.err kernel: [481093.702587] blk_update_request: I/O error, dev sdb, sector 10984 op 0x1:(WRITE) flags 0x103000 phys_seg 1 prio class 0
Tue Mar 11 03:43:24 2025 kern.err kernel: [481093.718043] Buffer I/O error on dev sdb1, logical block 1117, lost async page write
Tue Mar 11 03:43:24 2025 kern.err kernel: [481093.724844] blk_update_request: I/O error, dev sdb, sector 2048 op 0x1:(WRITE) flags 0x800 phys_seg 1 prio class 0
Tue Mar 11 03:43:24 2025 kern.err kernel: [481093.736257] Buffer I/O error on dev sdb1, logical block 0, lost sync page write
Tue Mar 11 03:43:24 2025 kern.err kernel: [481093.741621] EXT4-fs (sdb1): I/O error while writing superblock
Tue Mar 11 03:43:24 2025 kern.err kernel: [481093.741631] blk_update_request: I/O error, dev sdb, sector 76520 op 0x1:(WRITE) flags 0x103000 phys_seg 1 prio class 0
Tue Mar 11 03:43:24 2025 kern.err kernel: [481093.742416] Buffer I/O error on dev sdb1, logical block 9309, lost async page write
Tue Mar 11 03:43:24 2025 kern.err kernel: [481093.766141] blk_update_request: I/O error, dev sdb, sector 10736 op 0x0:(READ) flags 0x3000 phys_seg 1 prio class 0
Tue Mar 11 03:43:24 2025 kern.crit kernel: [481093.784656] EXT4-fs error (device sdb1): ext4_wait_block_bitmap:570: comm kworker/u8:3: Cannot read block bitmap - block_group = 1, block_bitmap = 1086
Tue Mar 11 03:43:24 2025 kern.err kernel: [481093.799340] blk_update_request: I/O error, dev sdb, sector 2048 op 0x1:(WRITE) flags 0x800 phys_seg 1 prio class 0
Tue Mar 11 03:43:24 2025 kern.err kernel: [481093.819741] Buffer I/O error on dev sdb1, logical block 0, lost sync page write
Tue Mar 11 03:43:24 2025 kern.err kernel: [481093.830252] EXT4-fs (sdb1): I/O error while writing superblock
Tue Mar 11 03:43:24 2025 kern.crit kernel: [481093.836092] EXT4-fs error (device sdb1): ext4_check_bdev_write_error:217: comm kworker/u8:3: Error while async write back metadata
Tue Mar 11 03:43:24 2025 kern.err kernel: [481093.852009] Buffer I/O error on dev sdb1, logical block 0, lost sync page write
Tue Mar 11 03:43:24 2025 kern.err kernel: [481093.858151] EXT4-fs (sdb1): I/O error while writing superblock
[... cut ...]
Tue Mar 11 03:43:24 2025 kern.warn kernel: [481093.944233] EXT4-fs warning (device sdb1): ext4_end_bio:347: I/O error 10 writing to inode 16121952 starting block 102632)
Tue Mar 11 03:43:24 2025 kern.warn kernel: [481093.944240] EXT4-fs warning (device sdb1): ext4_end_bio:347: I/O error 10 writing to inode 16121952 starting block 102656)
Tue Mar 11 03:43:24 2025 kern.crit kernel: [481093.944266] EXT4-fs error (device sdb1): ext4_check_bdev_write_error:217: comm touch: Error while async write back metadata
Tue Mar 11 03:43:24 2025 kern.err kernel: [481093.944371] Aborting journal on device sdb1-8.

simendancer, 2025-03-14:
闲鱼淘的一个台达12V3A的电源昨天到货了，直接把固态插在N1后面，十几个小时了很稳定没掉过。。。
目前看来可能真的是供电问题，各种外部供电的办法都是虚的，只有换电源才是真实的
再观察两天，希望可以一直稳定下去

jjm2473, 2025-03-17:
如果还会出现这些报错，那执行`lsusb -tv`看看输出什么

simendancer, 2025-03-19:
> 如果还会出现这些报错，那执行`lsusb -tv`看看输出什么
掉了
又报了大量的
Wed Mar 19 22:58:08 2025 kern.info kernel: [158425.398809] usb 1-2: new high-speed USB device number 7 using xhci-hcd
Wed Mar 19 22:58:08 2025 kern.warn kernel: [158425.398874] usb 1-2: Device not responding to setup address.
Wed Mar 19 22:58:08 2025 kern.warn kernel: [158425.618826] usb 1-2: Device not responding to setup address.
Wed Mar 19 22:58:08 2025 kern.err kernel: [158425.839036] usb 1-2: device not accepting address 7, error -71

lsusb -tv
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/0p, 5000M
    ID 1d6b:0003  
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/2p, 480M
    ID 1d6b:0002

diave971, 2025-03-23:
硬盘盒兼容问题？或者使用usb2.0的数据线连接看看

simendancer, 2025-03-25:
> 硬盘盒兼容问题？或者使用usb2.0的数据线连接看看

磁盘直接不显示了。。。

simendancer, 2025-04-07:
前段时间发现openclash经常内存暴涨，而且不管怎么设置感觉访问国内网站就是慢，我就把它卸载了，装了shellcrash。然后自那之后，目前一星期都没掉过盘， 而且用smb和webdav各种访问都不会掉盘，日志里也没有任何跟usb有关的日志了，真的很神奇