istoreos/istoreos #2161: LAN接口下联设备，DHCP获取不到IPv4地址但可以获取到IPv6地址，手动配置静态地址也无法上网

Donleo-feng, 2025-03-16:
尝试重启，断开上下连接设备，直接使用网线接上电脑也无法获取IP地址，只能重置固件恢复，已经出现几次了。

设备型号
FriendlyElec NanoPi R5C
固件版本
iStoreOS 22.03.7 2024122712
内核版本
5.10.221

[image]

日志信息如下，看是否能分析一下是什么问题
Sun Mar 16 20:22:47 2025 daemon.info dnsmasq-dhcp[1]: DHCPDISCOVER(br-lan) 00:0e:c6:7b:4e:0c
Sun Mar 16 20:22:47 2025 daemon.info dnsmasq-dhcp[1]: DHCPDISCOVER(br-lan) 00:0e:c6:7b:4e:0c
Sun Mar 16 20:22:47 2025 daemon.info dnsmasq-dhcp[1]: DHCPOFFER(br-lan) 192.168.3.2 00:0e:c6:7b:4e:0c
Sun Mar 16 20:22:47 2025 daemon.info dnsmasq-dhcp[1]: DHCPOFFER(br-lan) 192.168.3.23 00:0e:c6:7b:4e:0c
Sun Mar 16 20:22:47 2025 daemon.warn dnsmasq-dhcp[1]: Error sending DHCP packet to 192.168.3.2: Operation not permitted
Sun Mar 16 20:22:47 2025 daemon.warn dnsmasq-dhcp[1]: Error sending DHCP packet to 192.168.3.23: Operation not permitted
Sun Mar 16 20:22:51 2025 daemon.info dnsmasq-dhcp[1]: DHCPDISCOVER(br-lan) 00:0e:c6:7b:4e:0c
Sun Mar 16 20:22:51 2025 daemon.info dnsmasq-dhcp[1]: DHCPDISCOVER(br-lan) 00:0e:c6:7b:4e:0c
Sun Mar 16 20:22:51 2025 daemon.info dnsmasq-dhcp[1]: DHCPOFFER(br-lan) 192.168.3.2 00:0e:c6:7b:4e:0c
Sun Mar 16 20:22:51 2025 daemon.info dnsmasq-dhcp[1]: DHCPOFFER(br-lan) 192.168.3.23 00:0e:c6:7b:4e:0c
Sun Mar 16 20:22:51 2025 daemon.warn dnsmasq-dhcp[1]: Error sending DHCP packet to 192.168.3.23: Operation not permitted
Sun Mar 16 20:22:51 2025 daemon.warn dnsmasq-dhcp[1]: Error sending DHCP packet to 192.168.3.2: Operation not permitted
Sun Mar 16 20:22:51 2025 daemon.err odhcpd[6604]: Failed to send to fe80::6ddc:9138:2a6b:c689%lan@br-lan (Operation not permitted)
Sun Mar 16 20:22:59 2025 daemon.info dnsmasq-dhcp[1]: DHCPDISCOVER(br-lan) 00:0e:c6:7b:4e:0c
Sun Mar 16 20:22:59 2025 daemon.info dnsmasq-dhcp[1]: DHCPDISCOVER(br-lan) 00:0e:c6:7b:4e:0c
Sun Mar 16 20:22:59 2025 daemon.info dnsmasq-dhcp[1]: DHCPOFFER(br-lan) 192.168.3.23 00:0e:c6:7b:4e:0c
Sun Mar 16 20:22:59 2025 daemon.info dnsmasq-dhcp[1]: DHCPOFFER(br-lan) 192.168.3.2 00:0e:c6:7b:4e:0c
Sun Mar 16 20:22:59 2025 daemon.warn dnsmasq-dhcp[1]: Error sending DHCP packet to 192.168.3.2: Operation not permitted
Sun Mar 16 20:22:59 2025 daemon.warn dnsmasq-dhcp[1]: Error sending DHCP packet to 192.168.3.23: Operation not permitted
Sun Mar 16 20:22:59 2025 daemon.err odhcpd[6604]: Failed to send to fe80::6ddc:9138:2a6b:c689%lan@br-lan (Operation not permitted)
Sun Mar 16 20:23:15 2025 daemon.err odhcpd[6604]: Failed to send to fe80::6ddc:9138:2a6b:c689%lan@br-lan (Operation not permitted)
Sun Mar 16 20:23:17 2025 daemon.info dnsmasq-dhcp[1]: DHCPDISCOVER(br-lan) 00:0e:c6:7b:4e:0c
Sun Mar 16 20:23:17 2025 daemon.info dnsmasq-dhcp[1]: DHCPDISCOVER(br-lan) 00:0e:c6:7b:4e:0c
Sun Mar 16 20:23:17 2025 daemon.info dnsmasq-dhcp[1]: DHCPOFFER(br-lan) 192.168.3.23 00:0e:c6:7b:4e:0c
Sun Mar 16 20:23:17 2025 daemon.info dnsmasq-dhcp[1]: DHCPOFFER(br-lan) 192.168.3.2 00:0e:c6:7b:4e:0c
Sun Mar 16 20:23:17 2025 daemon.warn dnsmasq-dhcp[1]: Error sending DHCP packet to 192.168.3.2: Operation not permitted
Sun Mar 16 20:23:17 2025 daemon.warn dnsmasq-dhcp[1]: Error sending DHCP packet to 192.168.3.23: Operation not permitted
Sun Mar 16 20:23:47 2025 daemon.err odhcpd[6604]: Failed to send to fe80::6ddc:9138:2a6b:c689%lan@br-lan (Operation not permitted)
Sun Mar 16 20:23:48 2025 daemon.info dnsmasq-dhcp[1]: DHCPDISCOVER(br-lan) 00:0e:c6:7b:4e:0c
Sun Mar 16 20:23:48 2025 daemon.info dnsmasq-dhcp[1]: DHCPDISCOVER(br-lan) 00:0e:c6:7b:4e:0c
Sun Mar 16 20:23:48 2025 daemon.info dnsmasq-dhcp[1]: DHCPOFFER(br-lan) 192.168.3.2 00:0e:c6:7b:4e:0c
Sun Mar 16 20:23:48 2025 daemon.info dnsmasq-dhcp[1]: DHCPOFFER(br-lan) 192.168.3.23 00:0e:c6:7b:4e:0c
Sun Mar 16 20:23:48 2025 daemon.warn dnsmasq-dhcp[1]: Error sending DHCP packet to 255.255.255.255: Operation not permitted
Sun Mar 16 20:23:48 2025 daemon.warn dnsmasq-dhcp[1]: Error sending DHCP packet to 255.255.255.255: Operation not permitted
Sun Mar 16 20:23:50 2025 daemon.info dnsmasq-dhcp[1]: DHCPDISCOVER(br-lan) 00:0e:c6:7b:4e:0c
Sun Mar 16 20:23:50 2025 daemon.info dnsmasq-dhcp[1]: DHCPDISCOVER(br-lan) 00:0e:c6:7b:4e:0c
Sun Mar 16 20:23:50 2025 daemon.info dnsmasq-dhcp[1]: DHCPOFFER(br-lan) 192.168.3.23 00:0e:c6:7b:4e:0c
Sun Mar 16 20:23:50 2025 daemon.info dnsmasq-dhcp[1]: DHCPOFFER(br-lan) 192.168.3.2 00:0e:c6:7b:4e:0c
Sun Mar 16 20:23:50 2025 daemon.warn dnsmasq-dhcp[1]: Error sending DHCP packet to 255.255.255.255: Operation not permitted
Sun Mar 16 20:23:50 2025 daemon.warn dnsmasq-dhcp[1]: Error sending DHCP packet to 255.255.255.255: Operation not permitted
Sun Mar 16 20:23:59 2025 daemon.info dnsmasq-dhcp[1]: DHCPDISCOVER(br-lan) 00:0e:c6:7b:4e:0c
Sun Mar 16 20:23:59 2025 daemon.info dnsmasq-dhcp[1]: DHCPDISCOVER(br-lan) 00:0e:c6:7b:4e:0c
Sun Mar 16 20:23:59 2025 daemon.info dnsmasq-dhcp[1]: DHCPOFFER(br-lan) 192.168.3.23 00:0e:c6:7b:4e:0c
Sun Mar 16 20:23:59 2025 daemon.info dnsmasq-dhcp[1]: DHCPOFFER(br-lan) 192.168.3.2 00:0e:c6:7b:4e:0c
Sun Mar 16 20:23:59 2025 daemon.warn dnsmasq-dhcp[1]: Error sending DHCP packet to 255.255.255.255: Operation not permitted
Sun Mar 16 20:23:59 2025 daemon.warn dnsmasq-dhcp[1]: Error sending DHCP packet to 255.255.255.255: Operation not permitted
Sun Mar 16 20:24:14 2025 daemon.info dnsmasq-dhcp[1]: DHCPDISCOVER(br-lan) 00:0e:c6:7b:4e:0c
Sun Mar 16 20:24:14 2025 daemon.info dnsmasq-dhcp[1]: DHCPDISCOVER(br-lan) 00:0e:c6:7b:4e:0c
Sun Mar 16 20:24:14 2025 daemon.info dnsmasq-dhcp[1]: DHCPOFFER(br-lan) 192.168.3.2 00:0e:c6:7b:4e:0c
Sun Mar 16 20:24:14 2025 daemon.info dnsmasq-dhcp[1]: DHCPOFFER(br-lan) 192.168.3.23 00:0e:c6:7b:4e:0c
Sun Mar 16 20:24:14 2025 daemon.warn dnsmasq-dhcp[1]: Error sending DHCP packet to 255.255.255.255: Operation not permitted
Sun Mar 16 20:24:14 2025 daemon.warn dnsmasq-dhcp[1]: Error sending DHCP packet to 255.255.255.255: Operation not permitted
Sun Mar 16 20:24:45 2025 daemon.info dnsmasq[1]: exiting on receipt of SIGTERM
Sun Mar 16 20:24:45 2025 daemon.info dnsmasq[1]: exiting on receipt of SIGTERM
Sun Mar 16 20:24:45 2025 daemon.info dnsmasq[1]: started, version 2.89 cachesize 1000
Sun Mar 16 20:24:45 2025 daemon.info dnsmasq[1]: DNS service limited to local subnets
Sun Mar 16 20:24:45 2025 daemon.info dnsmasq[1]: compile time options: IPv6 GNU-getopt no-DBus UBus no-i18n no-IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset no-nftset auth cryptohash DNSSEC no-ID loop-detect inotify dumpfile
Sun Mar 16 20:24:45 2025 daemon.info dnsmasq[1]: UBus support enabled: connected to system bus
[... 99 lines cut ...]
Sun Mar 16 20:29:51 2025 daemon.info dnsmasq[1]: read /etc/hosts - 12 names
Sun Mar 16 20:29:51 2025 daemon.info dnsmasq[1]: read /tmp/hosts/dhcp.cfg01411c - 14 names
Sun Mar 16 20:29:51 2025 daemon.info dnsmasq[1]: read /tmp/hosts/dhcp.cfg03411c - 14 names
Sun Mar 16 20:29:51 2025 daemon.info dnsmasq[1]: read /tmp/hosts/odhcpd - 0 names
Sun Mar 16 20:29:51 2025 daemon.info dnsmasq-dhcp[1]: read /etc/ethers - 0 addresses
Sun Mar 16 20:29:52 2025 daemon.err odhcpd[23492]: Failed to send to fe80::6ddc:9138:2a6b:c689%lan@br-lan (Operation not permitted)
Sun Mar 16 20:29:54 2025 daemon.err odhcpd[23492]: Failed to send to fe80::6ddc:9138:2a6b:c689%lan@br-lan (Operation not permitted)
Sun Mar 16 20:29:58 2025 daemon.err odhcpd[23492]: Failed to send to fe80::6ddc:9138:2a6b:c689%lan@br-lan (Operation not permitted)
Sun Mar 16 20:30:06 2025 daemon.err odhcpd[23492]: Failed to send to fe80::6ddc:9138:2a6b:c689%lan@br-lan (Operation not permitted)
Sun Mar 16 20:30:22 2025 daemon.err odhcpd[23492]: Failed to send to fe80::6ddc:9138:2a6b:c689%lan@br-lan (Operation not permitted)
Sun Mar 16 20:30:32 2025 daemon.warn odhcpd[23492]: A default route is present but there is no public prefix on lan thus we don't announce a default route!
Sun Mar 16 20:30:43 2025 daemon.err odhcp6c[14699]: Failed to send SOLICIT message to ff02::1:2 (Operation not permitted)
Sun Mar 16 20:30:54 2025 daemon.err odhcpd[23492]: Failed to send to fe80::6ddc:9138:2a6b:c689%lan@br-lan (Operation not permitted)
Sun Mar 16 20:30:57 2025 daemon.err odhcp6c[10868]: Failed to send SOLICIT message to ff02::1:2 (Operation not permitted)
Sun Mar 16 20:31:06 2025 daemon.err zerotier-one[17308]: sendto: Operation not permitted
Sun Mar 16 20:31:10 2025 daemon.err nmbd[16866]: [2025/03/16 20:31:10.529359,  0] ../../source3/libsmb/nmblib.c:923(send_udp)
Sun Mar 16 20:31:10 2025 daemon.err nmbd[16866]:   Packet send failed to 192.168.3.255(138) ERRNO=Operation not permitted
[... cut ...]
Sun Mar 16 20:31:10 2025 daemon.err nmbd[16866]: [2025/03/16 20:31:10.531353,  0] ../../source3/nmbd/nmbd_namequery.c:245(query_name)
Sun Mar 16 20:31:10 2025 daemon.err nmbd[16866]:   query_name: Failed to send packet trying to query name WORKGROUP<1d>
Sun Mar 16 20:31:26 2025 user.info : luci: failed login on /admin/network/dhcp for root from fd1f:2c34:7a20:0:d4d7:c45e:b493:8845
Sun Mar 16 20:31:29 2025 user.info : luci: accepted login on /admin/network/dhcp for root from fd1f:2c34:7a20:0:d4d7:c45e:b493:8845

zgcbug, 2025-05-26:
解决了吗？我的重启下lan口可以用，不过没多久又不行了。

kc-bao, 2025-10-08:
此问题已在 **酷友社** 上被提及。那里可能有相关详细信息：

https://www.koolcenter.com/t/topic/1800/1

070822, 2026-02-08:
一样的，不过手动设置 IPv4 后，可以上网