istoreos/istoreos #2162: docker桥接网络模式无法访问互联网

a447948814, 2025-03-17:
旁路由模式，docker容器无法使用桥接网络连接互联网，内网可以。
搜到了一个issue 说把lan的ip动态伪装勾上就行了。我试了还是不行（也可能配置不对？）。
当时想彻底关闭防火墙，不知道动到啥了。怎么恢复默认防火墙的配置？

jjm2473, 2025-03-17:
先试试能不能ping通223.5.5.5

a447948814, 2025-03-17:
宝哥终于回复了，目前看，com.docker.network.bridge.enable_ ip_ masquerade是flase，容器内ping 223.5.5.5 也是不通的。

jjm2473, 2025-03-17:
不需要管docker的配置，docker已经配置成不使用iptables了，除非你自己安装了其他版本的docker

jjm2473, 2025-03-17:
要么就恢复出厂设置后，测试看docker的网络有没有问题

a447948814, 2025-03-17:
宝哥 直接远程看下可以吗 qq 447948814

jjm2473, 2025-03-17:
先备份下，再恢复出厂设置。之后恢复也只是安装下插件，配置文件都可以还原。

a447948814, 2025-03-17:
我找了大佬看了一下午，防火墙都重装了 也没弄好。你帮我看下呗  你最好了 最帅了

jjm2473, 2025-03-17:
防火墙的常规设置中，将“转发”改成“接受”试试

a447948814, 2025-03-17:
我全部都是接受，如果默认docker不是通过iptabes，我更没法查了，网上搜索出来的都是在讲这个。

a447948814, 2025-03-17:
[image]

a447948814, 2025-03-17:
突然就通了 我也不知道我搞了啥，，。
[image]

a447948814, 2025-03-17:
破案了。。。只要passwall启用 。docker容器外网就ping不通。不启用就没问题。。

a447948814, 2025-03-17:
破案了，passwall这个设置导致的：
[image]

jjm2473, 2025-03-24:
这个dns重定向应该没啥问题，除非你的passwall本身就无法访问外网

a447948814, 2025-04-01:
> 这个dns重定向应该没啥问题，除非你的passwall本身就无法访问外网

哥 ，，你说的对，，的确不是那个问题，，我现在又遇到了桥接网络无法上网的问题，，，。能不能帮我看看