istoreos/istoreos #2183: iStoreos是否有升级docker版本的计划

shiningth, 2025-03-29:
目前iStoreos的docker版本还是20.10.22,有没有计划升级到更高版本?比如26版,非常期待

w9315273, 2025-03-29:
等op 24版本就行了 iStoreos今年内应该会上吧

qingtian110, 2025-04-02:
我靠，看了一下 openwrt 的 docker 28 了

[image]

```txt
  _______                     ________        __
 |       |.-----.-----.-----.|  |  |  |.----.|  |_
 |   -   ||  _  |  -__|     ||  |  |  ||   _||   _|
 |_______||   __|_____|__|__||________||__|  |____|
          |__| W I R E L E S S   F R E E D O M
 -----------------------------------------------------
 OpenWrt 24.10.0, r28427-6df0e3d02a
 -----------------------------------------------------
[root@OpenWrt ~]# docker info
Client:
 Version:    28.0.4
 Context:    default
 Debug Mode: false

Server:
 Containers: 1
  Running: 1
  Paused: 0
  Stopped: 0
 Images: 1
 Server Version: 28.0.4
 Storage Driver: overlay2
  Backing Filesystem: xfs
  Supports d_type: true
  Using metacopy: false
  Native Overlay Diff: true
  userxattr: false
 Logging Driver: json-file
 Cgroup Driver: cgroupfs
 Cgroup Version: 1
 Plugins:
  Volume: local
  Network: bridge host ipvlan macvlan null overlay
  Log: awslogs fluentd gcplogs gelf journald json-file local splunk syslog
 Swarm: inactive
 Runtimes: runc io.containerd.runc.v2
 Default Runtime: runc
 Init Binary: docker-init
 containerd version: 
 runc version: 
 init version: de40ad0
 Security Options:
  seccomp
   Profile: builtin
 Kernel Version: 6.12.21
 Operating System: OpenWrt 24.10.0
 OSType: linux
 Architecture: aarch64
 CPUs: 4
 Total Memory: 3.834GiB
 Name: OpenWrt
 ID: 4508e978-e83b-4d40-b208-0c5223e76d94
 Docker Root Dir: /home/data/docker
 Debug Mode: false
 Experimental: true
 Insecure Registries:
  ::1/128
  127.0.0.0/8
 Registry Mirrors:
  https://docker.m.daocloud.io/
 Live Restore Enabled: false
```

qingtian110, 2025-04-02:
https://github.com/sbwml/packages_utils_docker
https://github.com/sbwml/packages_utils_dockerd
https://github.com/sbwml/packages_utils_runc

这里的 docker 和官方版本同步的 https://docs.docker.com/engine/release-notes/28/