istoreos/istoreos #2197: 关于 radxa E52c等搭载RK3582 rockchip的开核支持

ralopa673, 2025-04-04:
rk3582是rk3588的阉割版变种，原生通常包括额外的2个大核，甚至部分体质更佳的包括更多的GPU核心等。

_这里有一些必要的信息：https://patchwork.ozlabs.org/project/uboot/patch/20241210222341.3537543-2-jonas@kwiboo.se/_  

iStoreOS 22.03在默认的uboot下屏蔽了2个大核，使得coremark跑分不足6w。

而armbian、ophub-flippy openwrt的固件则默认开核，使得coremark跑分为10w，接近amd64的N100性能。

建议iStoreOS在24.10的固件中，搭载默认开核的uboot。

ralopa673, 2025-04-04:
这里以e52c的某固件为例，默认是开核的

[image]

jjm2473, 2025-04-07:
rk之所以搞出3582，不就是有些核心品质有问题要屏蔽吗？如果开核的话，不会出现不稳定的情况吗

jjm2473, 2025-04-07:
我之前测试过，开了编解码器，虽然系统没报错，但是转码后的视频是花屏的，完全没用。不知道cpu开核有没有这种问题。

boin, 2025-05-23:
看到24.1很多版本已经上了，继续蹲一手完美开核支持。苍蝇搓手

ralopa673, 2025-06-14:
> e52c和e54c已经适配完成，自动开核，两个都测试过没问题。测试固件将上传到 https://fw.koolcenter.com/iStoreOS/alpha/24.10/ ，注意只有全新刷机才会更新uboot，开核才能生效，所以建议使用tf卡刷机测试。测试版本固件不要在线升级不然会回到正式版。之后可能会提供单独升级uboot的教程。
> 
> 检查开核状态的命令`curl --silent https://cdn.haguro.top/check-rk3582.sh | bash -` （来自 https://forum.radxa.com/t/rk3582-soc-broken-ip-node-check/21562 ）

大佬辛苦啦！

我刷了一下后，暂时发现一个bug：

首先，我把e52c的istoreOS-22.03.7-20250407通过 https://www.bilibili.com/video/BV1eFJVzbEjU/?t=181 的流程，在iStore里做了”系统备份“。
然后，直接用rk工具线刷 24.10开核alpha 固件到emmc。
接着，在24.10的iStore做了”系统恢复“。然后就遇到其他机型（如r66s）的早期24.10 alpha固件的bug。例如，访问《系统-文件传输》、《docker配置页》报错 Runtime Error - luci/http，如图所示。 

[image]

[image]

最后我放弃了恢复，重新装插件来用，暂时没发现其他的问题。

jjm2473, 2025-06-14:
系统备份只能在同一个版本恢复。正确的方式是：
1. 线刷24.10
2. 在线升级到22.03，不保留配置
3. 恢复系统备份
4. 修复系统软件（systools的功能）
5. 保留配置升级24.10

简单来说，要满足两个原则：
1. 系统备份不能跨版本恢复
2. 必须有旧版本保留配置升级到新版本的过程，而且升级前执行修复系统软件

ralopa673, 2025-06-14:
> 系统备份只能在同一个版本恢复。正确的方式是：
> 
>     1. 线刷24.10
> 
>     2. 在线升级到22.03，不保留配置（这一步是为了保留24.10的uboot）
> 
>     3. 恢复22.03的系统备份
> 
>     4. 修复系统软件（systools的功能）
> 
>     5. 保留配置升级24.10
> 
> 
> 简单来说，要满足两个原则：
> 
>     1. 系统备份不能跨版本恢复
> 
>     2. 必须有旧版本保留配置升级到新版本的过程，而且升级前执行修复系统软件

原来如此。我没有做 2、4、5。

谢谢

jjm2473, 2025-06-14:
其实你遇到的错误很明显是openclash修改系统的文件导致的（只要安装过openclash就会出现），然后又没经过正常的升级流程。

如果只是针对这个问题，执行 `echo 0.0.0 > /overlay/.upgrading ； touch /usr/lib/opkg/.upgrading`，如果在沙箱环境还要执行`echo 0.0.0 > /ext_overlay/.upgrading`。最后执行`reboot`重启。重启以后系统会执行一些升级的修复操作，应该就能解决了。

boin, 2025-06-18:
E52C 直接SD卡 boot 是跑不起来的吗？我这跑起来网卡没有。没敢刷新。

附件是正常（Emmc 旧版本）和 新版本 （SD卡）的Uboot 日志

```
DDR d5483af87d cym 23/11/23-16:15:24,fwver: v1.15
LPDDR4, 2112MHz
channel[0] BW=16 Col=10 Bk=8 CS0 Row=15 CS=1 Die BW=16 Size=512MB
channel[1] BW=16 Col=10 Bk=8 CS0 Row=15 CS=1 Die BW=16 Size=512MB
channel[2] BW=16 Col=10 Bk=8 CS0 Row=15 CS=1 Die BW=16 Size=512MB
channel[3] BW=16 Col=10 Bk=8 CS0 Row=15 CS=1 Die BW=16 Size=512MB
Manufacturer ID:0x5
CH0 RX Vref:32.6%, TX Vref:13.2%,0.0%
CH1 RX Vref:32.6%, TX Vref:13.2%,0.0%
CH2 RX Vref:35.2%, TX Vref:13.2%,0.0%
CH3 RX Vref:33.4%, TX Vref:13.2%,0.0%
change to F1: 528MHz
change to F2: 1068MHz
change to F3: 1560MHz
change to F0: 2112MHz
out
U-Boot SPL board init
U-Boot SPL 2017.09-ge4e124926e-230922 #lxh (Sep 25 2023 - 10:58:38), fwver: v1.13
unknown raw ID 0 0 0
unrecognized JEDEC id bytes: 00, 00, 00
Trying to boot from MMC2
part_get_info_efi: *** ERROR: Invalid GPT ***
part_get_info_efi: ***        Using Backup GPT ***
No misc partition
spl: partition error
Trying fit image at 0x4000 sector
## Verified-boot: 0
## Checking atf-1 0x00040000 ... sha256(a7d1d8d191...) + OK
## Checking u-boot 0x00a00000 ... sha256(2022dfec85...) + OK
## Checking fdt-1 0x00a922b8 ... sha256(15f051f4ee...) + OK
## Checking atf-2 0x000f0000 ... sha256(aa71013e72...) + OK
## Checking atf-3 0xff100000 ... sha256(4b2065349b...) + OK
Jumping to U-Boot(0x00a00000) via ARM Trusted Firmware(0x00040000)
Total: 229.20/369.901 ms

INFO:    Preloader serial: 2
NOTICE:  BL31: v2.3():v2.3-682-g4ca8a8422:derrick.huang, fwver: v1.45
NOTICE:  BL31: Built : 10:11:21, Dec 27 2023
INFO:    spec: 0x1
INFO:    code: 0x82
INFO:    ext 32k is not valid
INFO:    ddr: stride-en 4CH
INFO:    GICv3 without legacy support detected.
INFO:    ARM GICv3 driver initialized in EL3
INFO:    valid_cpu_msk=0xff bcore0_rst = 0x0, bcore1_rst = 0x0
INFO:    l3 cache partition cfg-0
INFO:    system boots from cpu-hwid-0
INFO:    disable memory repair
INFO:    idle_st=0x21fff, pd_st=0x11fff9, repair_st=0xfff70001
INFO:    dfs DDR fsp_params[0].freq_mhz= 2112MHz
INFO:    dfs DDR fsp_params[1].freq_mhz= 528MHz
INFO:    dfs DDR fsp_params[2].freq_mhz= 1068MHz
INFO:    dfs DDR fsp_params[3].freq_mhz= 1560MHz
INFO:    BL31: Initialising Exception Handling Framework
INFO:    BL31: Initializing runtime services
WARNING: No OPTEE provided by BL2 boot loader, Booting device without OPTEE initialization. SMC`s destined for OPTEE will return SMC_UNK
ERROR:   Error initializing runtime service opteed_fast
INFO:    BL31: Preparing for EL3 exit to normal world
INFO:    Entry point address = 0xa00000
INFO:    SPSR = 0x3c9

U-Boot 2024.10-OpenWrt-2025061310 (Jun 13 2025 - 04:08:11 +0000)

Model: Easepi RK3588 Board
SoC:   RK3582
[... 442 lines cut ...]
Filesystem           1K-blocks      Used Available Use% Mounted on
/dev/root               145664    145664         0 100% /rom
tmpfs                  1014416     16464    997952   2% /tmp
/dev/mmcblk1p3         1992552      1100   1970972   0% /overlay
/dev/mmcblk1p4        12556960    185932  11711368   2% /ext_overlay
overlayfs:/overlay    12556960    185932  11711368   2% /
tmpfs                     4096         0      4096   0% /mnt
/dev/mmcblk1p1           64468     19836     43324  31% /boot
tmpfs                      512         0       512   0% /dev
/dev/mmcblk0p2          155392    155392         0 100% /mnt/mmc0-2
root@iStoreOS:/# ifconfig 
lo        Link encap:Local Loopback  
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:1091 errors:0 dropped:0 overruns:0 frame:0
          TX packets:1091 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:90167 (88.0 KiB)  TX bytes:90167 (88.0 KiB)

```

[old-sys.txt](https://github.com/user-attachments/files/20789475/old-sys.txt)  

[new-sys.txt](https://github.com/user-attachments/files/20789476/new-sys.txt)

boin, 2025-06-18:
还需要啥信息我给dump出来 @jjm2473 辛苦拉！

jjm2473, 2025-06-19:
@boin 这是因为emmc和sd使用了同一个uuid，导致内核从sd卡加载，但是rootfs却是emmc的。看日志内核版本是6.6.86，但是系统却是22.03.7，很明显不匹配。

关机，拔掉sd卡，然后开机，执行这个命令 `dd if=/dev/urandom of="/dev/mmcblk1" bs=1 count=4 seek=440 conv=fsync` 将emmc的分区表uuid改一下，然后插sd卡重启就正常了。

boin, 2025-06-19:
> [@boin](https://github.com/boin) 这是因为emmc和sd使用了同一个uuid，导致内核从sd卡加载，但是rootfs却是emmc的。看日志内核版本是6.6.86，但是系统却是22.03.7，很明显不匹配。
> 
> 关机，拔掉sd卡，然后开机，执行这个命令 `dd if=/dev/urandom of="/dev/mmcblk1" bs=1 count=4 seek=440 conv=fsync` 将emmc的分区表uuid改一下，然后插sd卡重启就正常了。

谢谢，我再试试

yrcscs, 2026-02-17:
单独升级Uboot后是这样的结果：
```bash
root@iStoreOS:/tmp/upload# curl --silent https://cdn.haguro.top/check-rk3582.sh
| bash -
CPU开核结果：6核
GPU开核结果：YES
解码器1开核结果：YES
解码器2开核结果：YES
编码器1开核结果：YES
编码器2开核结果：YES
```
可以看到GPU开核成功了，但是RK3582的CPU依然是原本的2大+4小核，有人遇到这样的问题了么？

jjm2473, 2026-02-24:
@yrcscs 开核本来就是看运气的，要是100%都能开核，那就不会当成 rk3582(3) 卖了。rk3582只能保证6个以上CPU核，以及NPU能用，其他都看运气。简单来说，rk3582(3)就是rk3588s的瑕疵品，屏蔽掉有问题的核心便宜卖。