istoreos/istoreos #2206: 局域网ip+端口 无法访问docker桥接网络中的容器

youngchou0928, 2025-04-06:
开了ipv6
ipv6域名+端口居然可以访问
唯独局域网+端口不可以

目前已有两个桥接网络
-默认bridge的docker0网络
-以及ha的hassio桥接网络
均不可被局域网访问
容器联网 访问广域网都正常

尝试修改/etc/sysctl.conf
net.bridge.bridge-nf-call-ip6tables=1
net.bridge.bridge-nf-call-iptables=1
net.bridge.bridge-nf-call-arptables=1
net.ipv4.ip_forward=1
仍然无效

jjm2473, 2025-04-07:
bridge-nf-call-* 不应该开，开了反而可能导致防火墙拦截。

youngchou0928, 2025-04-07:
> bridge-nf-call-* 不应该开，开了反而可能导致防火墙拦截。

单开net.ipv4.ip_forward=1也无效

> 你用的什么ip访问容器？

就是iStoreOS的局域网IP：192.168.31.10

youngchou0928, 2025-04-07:
> 你用的什么ip访问容器？

继续排查，只有docker默认网络docker0不能被局域网访问
ha桥接网络hassio以及新建桥接网络 可以访问

youngchou0928, 2025-04-07:
> 你用的什么ip访问容器？

[image]

排查是默认iptables里的DOCKER-USER链此REJECT规则阻止了外部访问
正在想办法删除

closed 2025-04-18 by jjm2473